h2zero/NimBLE-Arduino #816: onRead sequence question.

DontSerious, 2024-12-21:
Last version i used is 1.4.3, when i try to read value from esp32, the value that i get is after onRead process(i will call a setValue() inside onRead())
`void onRead(NimBLECharacteristic *characteristic, NimBLEConnInfo &connInfo)
    {
        String rxValue = getCurrentPatternName();
        characteristic->setValue(rxValue);
        Serial.println("pathName on Read: " + rxValue);
    }`
After i update to 2.1.1, the value is old value which is the value last time i call setValue().
Right now, i want to know if this sequence is change and how can i change my code to adapt this new version.

h2zero, 2024-12-21:
Not sure why you closed this, did you resolve your issue?

puzrin, 2025-01-04:
@h2zero I have the same issue when I try to migrate from 1.6.

I use the same long characteristic to write/read chunked data. Desktop client splits blob into multiple chunks and writes those to device's characteristic. Then tries to get response, reading chunks fro the same characteristic.

 In `onRead,` I make `setValue()` to fill with response chunks, but the desktop client receives the last value it's written before. It behaves like setValue() did nothing (but I don't know what happens under the hood).

Do you have ideas how to fix/workaround this?

h2zero, 2025-01-04:
@puzrin Can you please add `-D CONFIG_NIMBLE_CPP_DEBUG_ASSERT_ENABLED=1` and `-D CONFIG_NIMBLE_CPP_LOG_LEVEL=4` to your build flags and post the logs?

puzrin, 2025-01-04:
```
--- Terminal on /dev/ttyACM1 | 9600 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32c3-api1-20210207
[INFO]: State => Init
[INFO]: State => Idle
[INFO]: BLE initialized
D NimBLEServer: >> handleGapEvent: 
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: mtu update event; conn_handle=1 mtu=255
D NimBLEServer: << handleGapEvent
[INFO]: BLE: Device connected, conn_handle 1
[INFO]: BLE: MTU updated to 255, conn_handle 1
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: subscribe event; attr_handle=8, subscribed: true
D NimBLEServer: << handleGapEvent
D NimBLEServer: Gatt Write event
D NimBLEServer: Gatt Read event
D NimBLEServer: Gatt Write event
D NimBLEServer: Gatt Read event
[INFO]: BLE AUTH: Received chunk of length 28
[INFO]: BLE AUTH: Sending response chunk of length 83
[INFO]: BLE AUTH: Received chunk of length 28
[INFO]: BLE AUTH: Sending response chunk of length 83
D NimBLEServer: Gatt Write event
D NimBLEServer: Gatt Read event
D NimBLEServer: Gatt Write event
D NimBLEServer: Gatt Read event
[INFO]: BLE AUTH: Received chunk of length 28
[INFO]: BLE AUTH: Sending response chunk of length 83
[INFO]: BLE AUTH: Received chunk of length 28
[INFO]: BLE AUTH: Sending response chunk of length 83
D NimBLEServer: Gatt Write event
D NimBLEServer: Gatt Read event
D NimBLEServer: Gatt Write event
D NimBLEServer: Gatt Read event
[INFO]: BLE AUTH: Received chunk of length 28
[INFO]: BLE AUTH: Sending response chunk of length 83
[INFO]: BLE AUTH: Received chunk of length 28
[INFO]: BLE AUTH: Sending response chunk of length 83
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: subscribe event; attr_handle=8, subscribed: false
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
D NimBLEAdvertising: >> Advertising start: duration=0, dirAddr=NULL
D NimBLEAdvertising: << Advertising start
D NimBLEServer: << handleGapEvent
[INFO]: BLE: Device disconnected, conn_handle 1
[INFO]: BLE: Session destroyed
```

`[INFO]` are messages of my app logger work. It writes in background, order can be not in sync with your log messages.

h2zero, 2025-01-05:
Thanks @puzrin I have open #840 to resolve this

closed 2025-01-05 by h2zero

DontSerious, 2025-01-07:
Sry, I changed my function workflow and everything worked again because I felt like I didn't understand the workflow. So I close it...thx for your work sincerely.

PR #840 (2025-01-05): Fix characteristic value unable to be updated in callback
Fixes #816