h2zero/NimBLE-Arduino #817: There is no backward compatibility of V2

ednieuw, 2024-12-21:
The new release is not backward compatible with V1 and does not raise any warning when compiling with the new version. Functions compile without warning but simply do not work anymore.
User of my sketches complain Bluetooth is not working after an new upload because they automatically upgraded their library versions. 
I can of course fork V1.4.3 and rename the library but then I have to change all my sketches I made the last years.  That will cause a lot of unexpected work.
Please rename the V2 NimBLE library to NimBLE2 or similar and replace 1.4.3 in the NimBLE library as Version 3.0 so the original NimBLE library automatically use V1 again.

h2zero, 2024-12-21:
This is the reason for release versions, renaming the library is not necessary or warranted, Just use the release that works for you.

thorrak, 2024-12-21:
Are you using Platformio for dependency management? Might be worth locking to major versions for libraries that implement [semantic versioning](https://semver.org/) as most do these days. 

A lot of libraries implement backwards incompatible changes at major version increments. The uplift for NimBLE-Arduino 2.x was nothing compared to ArduinoJSON 7.x

ednieuw, 2024-12-21:
A main issue is that V1 functions compile without a warning with version V2 but do not function anymore. 
I recognize the problem now but many users of the sketch will not. With Arduino IDE2.0, that many people use, there is no way to stay with NimBLE V1. 
But thanks for the work that is done.

h2zero, 2024-12-21:
@ednieuw That was surprising to hear so I had to have a look for myself. It is indeed possible to change the version of the library used in Arduino IDE 2.0:
[image]

ednieuw, 2024-12-23:
@h2zero Unfortunately you can not lock a library version to a sketch in the Arduino IDE. 

An upgrade to NimBLE V2 is possible but many of my sketches, also used by other people, are using the NimBLE library. 
These old sketches will still compile without warning and with Bluetooth not functioning.
Maybe you can include a warning during compilation.

I tried to copy to 1.4.3 library to my sketch folder and "#include "NimBLE-Arduino\src\NimBLEDevice.h"  but then many header files can not be found.

`C:\Users\ednie\Documents\Files\Arduino\Arduino_Nano-ESP32-WordClock\ESP32Arduino_WordClockV060\NimBLE-Arduino\src\nimble\nimble\include\nimble\ble.h:25:10: fatal error: nimble/porting/nimble/include/syscfg/syscfg.h: No such file or directory`

h2zero, 2024-12-23:
> Unfortunately you can not lock a library version to a sketch in the Arduino IDE.

That is unfortunate indeed, but you can switch back and forth depending on the project you're working on.

>An upgrade to NimBLE V2 is possible but many of my sketches, also used by other people, are using the NimBLE library.

The changes required to use v2.x should not be too burdensome.

>These old sketches will still compile without warning and with Bluetooth not functioning.
Maybe you can include a warning during compilation.

Not sure how I would go about including a warning, and I'm sure most users would not appreciate a compiler warning every time they build their projects. 

>I tried to copy to 1.4.3 library to my sketch folder and "#include "NimBLE-Arduino\src\NimBLEDevice.h" but then many header files can not be found.

You would have to copy just the src folder to your sketch folder for this to possibly build due to the include paths.

ednieuw, 2024-12-24:
> Not sure how I would go about including a warning, and I'm sure most users would not appreciate a compiler warning every time they build their projects.

The compiler warning can just be a comment to upgrade. Then there is some kind of warning.
This morning I already had a complaint from one of my sketch users bluetooth was not working. Most users are not able to dig into these large c++ libraries. Many users hardly know that they have upgraded a library in the Arduino IDE. 
The IDE complains at every start there are boards and libraries to upgrade and it can do it for all libraries with one click.

Is it possible to overload the depreciated functions?  
Or 
Add a #define NIMBLE_VERSION 2 
Then it is possible to discriminate between the two versions in the sketch. 

I now apply the same approach to the Arduino Nano ESP32 version 2.0.18, supported by Arduino, and the Espressif core 3.0.7, which support new features.
```
#if ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(3, 0, 0)
#include <EdSoftLED.h>
                      #else
#include <Adafruit_NeoPixel.h> 
                      #endif
```

> I tried to copy to 1.4.3 library to my sketch folder and "#include "NimBLE-Arduino\src\NimBLEDevice.h" but then many header files can not be found.

>You would have to copy just the src folder to your sketch folder for this to possibly build due to the include paths.

Ah, that works. You have to place the contents of the src folder beside the ino sketch. That is too messy for me; too many tabs in the IDE and files in my sketch folder.

thorrak, 2024-12-24:
This seems like a major oversight in the Arduino IDE rather than an issue with this library. Semantic versioning is pretty standard — it’s wild that there isn’t a way to lock to a specific version - major or minor - given the possibility for any of your dependencies to make incompatible changes even between revisions. 

Have you considered switching to a better IDE such as Platformio?

ednieuw, 2024-12-24:
> Have you considered switching to a better IDE such as Platformio?

Yes, several times. But there is so much installed I loose control what's happening. 

The people I write my software for are often not very experienced programmers. More builders of machines, soldering electronics and so on.  The Arduino IDE is because of its simplicity preferable to use and to help with.
That is the reason I want to solve the problems in the software and not in installations. If also try to get rid of libraries for obvious reasons.
I am a fan now of uploading a compiled binary file with OTA and control/setup the MCU with BLE or a HTML page. (For as long as securities do not kill these options)

h2zero, 2024-12-24:
I would recommend in this case that the users and the code be updated to use v2.

That said there are some new config options in v2 that you can use to check the existence of and know what that v2 is in use.
```
#ifdef CONFIG_NIMBLE_CPP_FREERTOS_TASK_BLOCK_BIT
```