h2zero/NimBLE-Arduino #821: Guru Meditation Error (LoadProhibited) while executing NimBLEClient::discoverAttributes()

gittehupe, 2024-12-25:
Hello (and merry christmas to everybody :-) ),

with the update of NimBLE-Arduino from 1.4.3 to 2.1.2 the executing of NimBLEClient::discoverAttributes() leads now to a Guru Meditation Error (LoadProhibited):

```
D NimB
EClD NimBLEClient: >> handleGapEvent 
D NimBLEClient: >> handleGapEvent 
I NimBLEClient: mtu update: mtu=23
D NimBLEClientCallbacks: onMTUChange: default
D NimBLEClient: <C< handleGapEvent
D NimBLEClient: exchangeMTUCb: status=0, mtu=23
D NimBLEClient: >> handleGapEvent 
D NimBLEClient: << handleGapEvent
onnected to 00:1a:22:10:ac:c3
D NimBLEClient: << connect()
Discovering all attributes...
D NimBLEClient: Service Discovered >> status: 0 handle: 256
D NimBLEClient: Service Discovered >> status: 0 handle: 512
D NimBLEClient: Service Discovered >> status: 0 handle: 768
D NimBLEClient: Service Discovered >> status: 0 handle: 1024
D NimBLEClient: Service Discovered >> status: 0 handle: 65280
D NimBLEClient: Service Discovered >> status: 14 handle: -1
D NimBLEClient: << Service Discovered
D NimBLERemoteService: >> retrieveCharacteristics()
D NimBLERemoteService: Characteristic Discovery >>
D NimBLERemoteService: Characteristic Discovery >>
D NimBLERemoteService: Characteristic Discovery >>
D NimBLERemoteService: Characteristic Discovery >>
D NimBLERemoteService: Characteristic Discovery >>
D NimBLERemoteService: Characteristic Discovery >>
D NimBLERemoteService: << Characteristic Discovery
D NimBLERemoteService: << retrieveCharacteristics()
D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a00
D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 0 handle: 288
D NimBLERemoteCharacteristic: << Descriptor Discovery
D NimBLERemoteCharacteristic: << retrieveDescriptors(): found 1 descriptors.
D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 0 handle: 289
D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a01
D NimBLERemoteCharacteristic: << Descriptor Discovery
D NimBLERemoteCharacteristic: << retrieveDescriptors(): found 0 descriptors.
D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 0 handle: 304
D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a02
D NimBLERemoteCharacteristic: << Descriptor Discovery
D NimBLERemoteCharacteristic: << retrieveDescriptors(): found 0 descriptors.
D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 0 handle: 305
D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a03
D NimBLERemoteCharacteristic: << Descriptor Discovery
D NimBLERemoteCharacteristic: << retrieveDescriptors(): found 0 descriptors.
D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 0 handle: 320
D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a04
E NimBLERemoteCharacteristic: ble_gattc_disc_all_dscs: rc=3 
D NimBLERemoteCharacteristic: << Descriptor Discovery
D NimBLEClient: >> getService: uuid: 3e135142-654f-9090-134a-a6ff5bb77046
D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 10 handle: -1
D NimBLEClient: << getService: found the service with uuid: 3e135142-654f-9090-134a-a6ff5bb77046
D NimBLERemoteCharacteristic: << Descriptor Discovery
D NimBLERemoteService: >> getCharacteristic: uuid: d0e8434d-cd29-0996-af41-6c90f4e0eb2a
D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 0 handle: 305
D NimBLERemoteService: >> retrieveCharacteristics()
D NimBLERemoteCharacteristic: << Descriptor Discovery
D NimBLERemoteService: << retrieveCharacteristics()
D NimBLERemoteService: << Characteristic not found
Failed to find our characteristic UUID: D NimBLERemoteCharacteristic: >> setNotify()
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400deb19  PS      : 0x00060230  A0      : 0x800debce  A1      : 0x3ffcea30  
A2      : 0x00000000  A3      : 0x3f4052e8  A4      : 0x3ffcea8c  A5      : 0x00000001  
A6      : 0x3ffc57f8  A7      : 0xe8434dcd  A8      : 0x4019bc10  A9      : 0x00000406  
A10     : 0x4019bc10  A11     : 0x3ffcea8c  A12     : 0x3ffcea48  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000034  LBEG    : 0x4009287d  LEND    : 0x4009288d  LCOUNT  : 0xfffffff9  

Backtrace: 0x400deb16:0x3ffcea30 0x400debcb:0x3ffcea80 0x400d36f2:0x3ffceac0 0x400d3bc9:0x3ffceb00 0x400d3c5b:0x3ffceb60 0x400d4dc9:0x3ffceba0 0x400f5d7d:0x3ffcec00
```
Any idea on the reason for this behavior?

Thanks a lot in advance!

gittehupe

h2zero, 2024-12-25:
Looks like you didn't check if the characteristic returned was nullptr and then called subscribe?

gittehupe, 2024-12-26:
OK, Thanks for the hint. I will investigate this thouroughly. I will let you know the reuslt :-)

gittehupe, 2024-12-30:
OK, I stripped down the code now to:
`#include <Arduino.h>

#include "NimBLEDevice.h"

NimBLEClient *client = nullptr;
NimBLEAddress btadd = NimBLEAddress("00:1A:22:10:AC:C3", 0);

void setup()
{
  Serial.begin(115200); // Begin Serial
  while (Serial == false)
  {
  }; // Wait for the serial connection to start up

  NimBLEDevice::init(""); // prepare the NimBLE stack to be ready for commands
  Serial.println("\nStarting Arduino BLE Client application...\n");
  client = NimBLEDevice::createClient();
  /**
   *  Set initial connection parameters:
   *  These settings are safe for 3 clients to connect reliably, can go faster if you have less
   *  connections. Timeout should be a multiple of the interval, minimum is 100ms.
   *  Min interval: 12 * 1.25ms = 15, Max interval: 12 * 1.25ms = 15, 0 latency, 150 * 10ms = 1500ms timeout
   */
  client->setConnectionParams(12, 12, 0, 150);

  /** Set how long we are willing to wait for the connection to complete (milliseconds), default is 30000. */
  client->setConnectTimeout(5 * 1000);

  bool connected = client->connect(btadd, false);

  delay(5 * 1000);

  if (!connected)
  {
    Serial.println("Failed to connect to device.");
  }
  else
  {

    Serial.printf("\nFound device\n");
  }

  client->discoverAttributes();

  Serial.println("\n\nWaiting for 3 seconds so you can read this info ^^^");

  delay(3000);
}

void loop()
{

  delay(500);
}
`

This way I still get resets as follows:
`ELF file SHA256: 0640369f7aa19f50

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
[... 49 lines cut ...]
D NimBLERemoteCharacteristic: << Descriptor Discovery
D NimBLERemoteCharacteristic: << retrieveDescriptors(): found 0 descriptors.
D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 0 handle: 305
D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a03
D NimBLERemoteCharacteristic: << Descriptor Discovery
D NimBLERemoteCharacteristic: << retrieveDescriptors(): found 0 descriptors.
D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 0 handle: 320
D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a04
D NimBLERemoteCharacteristic: << Descriptor Discovery
E NimBLERemoteCharacteristic: ble_gattc_disc_all_dscs: rc=3 

D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 10 handle: -1
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d3610  PS      : 0x00060630  A0      : 0x800d8646  A1      : 0x3ffcc6d0  
A2      : 0x0000000a  A3      : 0x3ffc3564  A4      : 0x00000141  A5      : 0x3ffcc734  
A6      : 0x800e6010  A7      : 0x3ffc3638  A8      : 0x3ffc8cac  A9      : 0x0000000a  
A10     : 0x0000004c  A11     : 0x3ffbbfa4  A12     : 0x0000000a  A13     : 0xffffffff  
A14     : 0x00060c20  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800e6010  LBEG    : 0x400915c9  LEND    : 0x400915d9  LCOUNT  : 0xfffffff9  

Backtrace: 0x400d360d:0x3ffcc6d0 0x400d8643:0x3ffcc710 0x400d9901:0x3ffcc730 0x400d5081:0x3ffcc770 0x400d4dd9:0x3ffcc7c0 0x400dc39f:0x3ffcc7f0 0x400db1b4:0x3ffcc820 0x400db1bf:0x3ffcc840 0x40081339:0x3ffcc860 0x400d2d94:0x3ffcc880`

Obviously I am using discoverAttributes() wrongly?!

doudar, 2024-12-30:
I believe you need:

If (client != nullptr){
client.discoverAttributes()
}

gittehupe, 2024-12-30:
> If (client != nullptr){
> client.discoverAttributes()
> }

Thanks for your proposal. But this did not make any difference.

h2zero, 2025-01-02:
You are calling `discoverAttributes()` without first checking if the client is connected, put that inside your `if` block.

gittehupe, 2025-01-05:
I thought that in the given example a proper connection is already ensured by checking connected (= client->connect(btadd, false)) right before the discovery leading to the "Found device" output?!

...
**D NimBLEClientCallbacks: onConnect: default**
D NimBLEClient: exchangeMTUCb: status=0, mtu=23
D NimBLEClient: << connect()
D NimBLEClient: >> handleGapEvent
D NimBLEClient: << handleGapEvent

**Found device**
...

thekurtovic, 2025-01-05:
@gittehup Your example checks the result but does nothing with it beyond printing. Should do something like this.
```
if (!connected)
{
    Serial.println("Failed to connect to device.");
    return;
}
```
Anyway since your log shows the crash when it _does_ connect, this probably isn't the reason. Curious whether it still crashes when `connected` is false and `discoverAttributes` is called anyway.

Can you decode the backtrace next time to try and get some more information? The crash dumps suggest the two were slightly different.