h2zero/NimBLE-Arduino #823: Version 2.1.2 bonded device not recognized after reconnection with random address

sanastasiou, 2024-12-26:
Hi @h2zero , I just tested 2.1.2 and here's is a sample from a log output. I can reproduce this 100%.

1. Bond device
2. Disconnect device
3. Reconnect device with different random id address
4. Try to access encrypted characteristic issues bonding request again

```
D NimBLEAdvertising: >> Advertising start: duration=0, dirAddr=NULL
primary service
           uuid 0x1800
         handle 1
     end_handle 5
characteristic
           uuid 0x2a00
     def_handle 2
     val_handle 3
   min_key_size 0
          flags [READ]
characteristic
           uuid 0x2a01
     def_handle 4
     val_handle 5
   min_key_size 0
          flags [READ]
primary service
           uuid 0x1801
         handle 6
     end_handle 13
characteristic
           uuid 0x2a05
     def_handle 7
     val_handle 8
   min_key_size 0
          flags [INDICATE]
ccc descriptor
           uuid 0x2902
         handle 9
   min_key_size 0
          flags [READ|WRITE]
characteristic
           uuid 0x2b3a
     def_handle 10
     val_handle 11
   min_key_size 0
          flags [READ]
characteristic
           uuid 0x2b29
     def_handle 12
     val_handle 13
   min_key_size 0
          flags [READ|WRITE]
primary service
           uuid 00002ffa-0000-1000-8000-00805f9b34fb
         handle 14
     end_handle 20
characteristic
           uuid 82c314b2-e53a-4d5e-b021-fc5887b4373f
     def_handle 15
     val_handle 16
   min_key_size 0
          flags [READ|NOTIFY|INDICATE]
ccc descriptor
           uuid 0x2902
         handle 17
   min_key_size 0
          flags [READ|WRITE]
characteristic
           uuid 3c5577d7-d182-482a-940a-7e5062441f43
     def_handle 18
[... 300 lines cut ...]
    -D CONFIG_BT_NIMBLE_ROLE_CENTRAL_DISABLED=1    ; Disable central role
    -D CONFIG_BT_NIMBLE_ROLE_OBSERVER_DISABLED=1   ; Disable observer role
    -D CONFIG_BT_NIMBLE_MESH=0                      ; Disable mesh if not needed
    -D CONFIG_BT_NIMBLE_CRYPTO_STACK_MBEDTLS=0      ; Use internal crypto (faster)
```

Security options:

```
        NimBLEDevice::setSecurityAuth(true, true, true);
        NimBLEDevice::setSecurityPasskey(pin);
        NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);
        NimBLEDevice::setSecurityInitKey(BLE_SM_PAIR_KEY_DIST_ID | BLE_SM_PAIR_KEY_DIST_ENC);
        NimBLEDevice::setSecurityRespKey(BLE_SM_PAIR_KEY_DIST_ID | BLE_SM_PAIR_KEY_DIST_ENC);
```

```
        uint8_t adv_flags = BLE_HS_ADV_F_DISC_GEN | BLE_HS_ADV_F_BREDR_UNSUP;
        advData.setFlags(adv_flags);
        fNimbleAdvertising->setAdvertisementData(advData);
```

The peer is an android phone with Android 13. Note that after sometime, the android tries to connect with the same address it did the bonding with **f0:5c:77:f5:bb:25** which is the BLE Mac Address and then obviously everything works. 

So I am not sure whether this is some bug or not.. Shouldn't the C3 be able to identify that the device is indeed bonded when the peer  ID adress is random?

sanastasiou, 2024-12-28:
Further testing reveals following. If after the bonding is finished in ESP32 the peer is disconnected withing the first 5 seconds, the peer still thinks it is not bonded.

After a little more time ( currently testing via trial and error ) the bonding information is also present on the peer device and then everything works as it should. So the question is, how do know if the peer has also received the bonding information before we attempt a disconnect? Apparently onAuthenticationComplete is not enough for this.

sanastasiou, 2025-01-01:
Additional further testing..

1. Bonding first with iOS device and afterwards with Android device works fine all the time.
2. Bonding first with Android device leads to Android not seeing the bonded device after whitelisting is active. Bonding afterwards with iOS device also fixes miraculously the Android issue.
3. Also found this: https://issuetracker.google.com/issues/242755161?pli=1 it appears, Android 13 is pure trashware.

Will switch over to Android 14 and try again. Will keep this thread updated.

h2zero, 2025-01-02:
@sanastasiou this is also of interest: https://devzone.nordicsemi.com/f/nordic-q-a/38247/android-automatically-unbonding-on-reconnection

sanastasiou, 2025-01-02:
@h2zero yeah, this is pretty much the issue. Disconnecting after a short time somehow removes bodning info from android while ESP32 thinks it is still bonded.. leading to all kinds of mess. Thanks for the link. I hope they fixed it in Android 14..