h2zero/NimBLE-Arduino #824: Unable to connect to example NimBLE_Server  on nrf51822

goddade, 2024-12-27:
Logs:
```
nRF Connect, 2024-12-28
NimBLE-Server (E9:55:BF:DD:4B:AC)
V	00:58:32.291	Connecting to E9:55:BF:DD:4B:AC...
D	00:58:32.291	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	00:58:32.790	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	00:58:32.790	Connected to E9:55:BF:DD:4B:AC
V	00:58:32.791	Discovering services...
D	00:58:32.791	gatt.discoverServices()
D	00:58:32.817	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
W	00:58:37.821	Connection parameters update failed with status 8 (interval: 45.0ms, latency: 0, timeout: 5000ms)
E	00:58:37.821	Error 8 (0x8): GATT INSUF AUTHORIZATION
D	00:58:37.830	[Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	00:58:37.830	Disconnected
D	00:58:37.872	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED

```

goddade, 2025-01-01:
I tried android, windows and iphone, none of them can connect.
Does anyone have the same problem?

h2zero, 2025-01-05:
@goddade The example is flawed sadly, I missed adding a 0 to the end of one of the parameters in the `updateParams` call in the `onConnect` callback. Please remove this line https://github.com/h2zero/NimBLE-Arduino/blob/480ecaf1950f9d3764538081ac16a3dab60d083f/examples/NimBLE_Server/NimBLE_Server.ino#L30 and try again.

goddade, 2025-01-06:
After deleting `pServer->updateConnParams(connInfo.getConnHandle(), 24, 48, 0, 18);`, new error:
```
nRF Connect, 2025-01-06
NimBLE-Server (E9:55:BF:DD:4B:AC)
V	13:45:18.067	Connecting to E9:55:BF:DD:4B:AC...
D	13:45:18.067	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	13:45:23.278	[Callback] Connection state changed with status: 133 and new state: DISCONNECTED (0)
E	13:45:23.278	Error 133 (0x85): GATT ERROR
I	13:45:23.278	Disconnected
```

goddade, 2025-01-08:
UART output:
```
Starting NimBLE Server
Advertising Started
Client address: 6d:b6:b1:ef:11:17
Client disconnected - start advertising
Client address: 6d:b6:b1:ef:11:17
Client disconnected - start advertising

```

h2zero, 2025-01-08:
I will need to see logs with debug level logging enabled. You can do this from the Arduino IDE menu or if using platformio:
```
build_flags = -DCONFIG_NIMBLE_CPP_LOG_LEVEL=5
```

goddade, 2025-01-09:
```
Starting NimBLE Server
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
D NimBLEDevice: Setting bonding: 0, mitm: 0, sc: 1
D NimBLEService: >> start(): Starting service: UUID: 0xdead, handle: 0x0000
D NimBLEService: Adding 1 characteristics for service UUID: 0xdead, handle: 0x0000
D NimBLEService: << start()
D NimBLEService: >> start(): Starting service: UUID: 0xbaad, handle: 0x0000
D NimBLEService: Adding 1 characteristics for service UUID: 0xbaad, handle: 0x0000
D NimBLEService: << start()
D NimBLEAdvertising: >> Advertising start: duration=0, dirAddr=NULL
primary service
           uuid 0x1800
         handle 1
     end_handle 5
characteristic
                         uuid 0x2a00
     def_handle 2
     val_handle 3
   min_key_size 0
          flags [READ]
characteristic
                         uuid 0x2a01
     def_handle 4
     val_handle 5
   min_key_size 0
          flags [READ]
primary service
           uuid 0x1801
         handle 6
     end_handle 13
characteristic
                         uuid 0x2a05
     def_handle 7
     val_handle 8
   min_key_size 0
          flags [INDICATE]
ccc descriptor
                         uuid 0x2902
         handle 9
   min_key_size 0
          flags [READ|WRITE]
characteristic
                         uuid 0x2b3a
     def_handle 10
     val_handle 11
   min_key_size 0
          flags [READ]
characteristic
                         uuid 0x2b29
     def_handle 12
     val_handle 13
   min_key_size 0
          flags [READ|WRITE]
primary service
           uuid 0xdead
         handle 14
     end_handle 17
characteristic
                         uuid 0xbeef
     def_handle 15
     val_handle 16
   min_key_size 0
          flags [READ|WRITE|READ_ENC|WRITE_ENC]
descriptor
                     uuid 0x2904
         handle 17
   min_key_size 0
          flags [READ]
primary service
           uuid 0xbaad
         handle 18
     end_handle 22
characteristic
                         uuid 0xf00d
     def_handle 19
     val_handle 20
   min_key_size 0
          flags [READ|WRITE|NOTIFY]
ccc descriptor
                         uuid 0x2902
         handle 21
   min_key_size 0
          flags [READ|WRITE]
descriptor
                     uuid 0xc01d
         handle 22
   min_key_size 0
          flags [READ|WRITE|WRITE_ENC]
D NimBLEAdvertising: setAdvertisementData: 02 01 06 0e 09 4e 69 6d 42 4c 45 2d 53 65 72 76 65 72 05 03 ad de ad ba
D NimBLEAdvertising: << Advertising start
Advertising Started
D NimBLEServer: >> handleGapEvent:
Client address: 72:f2:cd:c2:bd:19
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent:
Client disconnected - start advertising
D NimBLEAdvertising: >> Advertising start: duration=0, dirAddr=NULL
D NimBLEAdvertising: << Advertising start
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent:
Client address: 72:f2:cd:c2:bd:19
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent:
Client disconnected - start advertising
D NimBLEAdvertising: >> Advertising start: duration=0, dirAddr=NULL
D NimBLEAdvertising: << Advertising start
D NimBLEServer: << handleGapEvent
```

h2zero, 2025-01-09:
Thank you @goddade, I just tested the example on my nRF51 and could not reproduce the error. I did have an issue reading the secured characteristic because the example uses BT5 security that the nRF51 does not support. 

To resolve that this line https://github.com/h2zero/NimBLE-Arduino/blob/74ac317404aebb4683b99a4e5dd4f963db829575/examples/NimBLE_Server/NimBLE_Server.ino#L148
Needs to be replaced with `NimBLEDevice::setSecurityAuth(false, false, false);`

After that, testing with my iPhone and nRFConnect I had no issues connecting and reading/writing etc..

goddade, 2025-01-10:
Not work.
My chip is QFAAG3.
platformio.ini:
```
[env:nrf51822]
platform = n-able
board = nrf51_beacon
framework = arduino

debug_tool = blackmagic
upload_protocol = blackmagic
debug_port = COM7
upload_port = COM7
monitor_port = COM6

lib_deps =
    h2zero/NimBLE-Arduino

build_flags =
    -DCONFIG_NIMBLE_CPP_LOG_LEVEL=5
```

goddade, 2025-01-10:
Thanks for your help.

I found generic_nrf51822_xxaa.json in n-able.
After changing to `board = generic_nrf51822_xxaa`, I can connect.

Does nrf51 not support bonding?
How to encrypt point-to-point communication?

h2zero, 2025-01-10:
That's odd, not sure why nrf51_beacon doesn't work, I'll look into that. Glad you have it working 😄 

> Does nrf51 not support bonding?

It supports bonding, just not BT5 "secure connections", though that can be enabled it's off by default due to the flash size increase.

> How to encrypt point-to-point communication?

`NimBLEDevice::setSecurityAuth(true, false, false);` // enables bonding
`NimBLEDevice::setSecurityAuth(true, true, false);` // enables bonding and man in the middle protection, i.e. passkey (PIN)

goddade, 2025-01-11:
The problem maybe in this line: `"extra_flags": "-DARDUINO_BEACON_PCA20006 -DDEVICE_RAM_SIZE=16 -DUSE_LFXO",`
I don't know what `ARDUINO_BEACON_PCA20006` means, but no 32k crystal on my board.

goddade, 2025-01-11:
Now it can connect but cannot bond.

Commented out in `onConnect`:
```
// pServer->updateConnParams(connInfo.getConnHandle(), 24, 48, 0, 180);
```
Security config:
```
  NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY); // use passkey
  NimBLEDevice::setSecurityPasskey(123456);
  NimBLEDevice::setSecurityAuth(true, false, false);
```
NRF connect logs:
```
nRF Connect, 2025-01-11
NimBLE-Server (E9:55:BF:DD:4B:AC)
V	17:32:33.932	Connecting to E9:55:BF:DD:4B:AC...
D	17:32:33.932	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	17:32:34.349	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	17:32:34.349	Connected to E9:55:BF:DD:4B:AC
D	17:32:34.365	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	17:32:34.366	Discovering services...
D	17:32:34.366	gatt.discoverServices()
I	17:32:34.787	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	17:32:35.002	[Callback] Services discovered with status: 0
I	17:32:35.002	Services discovered
V	17:32:35.025	Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
- Server Supported Features [R] (0x2B3A)
- Client Supported Features [R W] (0x2B29)
Unknown Service (0000dead-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [R W] (0000beef-0000-1000-8000-00805f9b34fb)
   Characteristic Presentation Format (0x2904)
Unknown Service (0000baad-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N R W] (0000f00d-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
   Unknown Descriptor (0000c01d-0000-1000-8000-00805f9b34fb)
D	17:32:35.026	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	17:32:35.029	gatt.setCharacteristicNotification(0000f00d-0000-1000-8000-00805f9b34fb, true)
I	17:32:35.086	Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
V	17:32:37.844	Starting pairing...
D	17:32:37.844	device.createBond()
D	17:32:37.923	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BONDING (11)
D	17:32:37.961	[Broadcast] Action received: android.bluetooth.device.extra.PAIRING_VARIANT, pairing variant: PIN
I	17:32:38.309	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	17:33:12.910	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: NONE (10), reason: REMOTE DEVICE DOWN (4)
I	17:33:12.910	Bonding failed, reason: REMOTE DEVICE DOWN (4)
D	17:33:18.660	[Callback] Connection state changed with status: 22 and new state: DISCONNECTED (0)
E	17:33:18.660	Error 22 (0x16): GATT CONN TERMINATE LOCAL HOST
I	17:33:18.660	Disconnected
D	17:33:18.730	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```
UART logs:
```
Starting NimBLE Server
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
D NimBLEDevice: Setting bonding: 1, mitm: 0, sc: 0
D NimBLEService: >> start(): Starting service: UUID: 0xdead, handle: 0x0000
D NimBLEService: Adding 1 characteristics for service UUID: 0xdead, handle: 0x0000
D NimBLEService: << start()
D NimBLEService: >> start(): Starting service: UUID: 0xbaad, handle: 0x0000
D NimBLEService: Adding 1 characteristics for service UUID: 0xbaad, handle: 0x0000
D NimBLEService: << start()
D NimBLEAdvertising: >> Advertising start: duration=0, dirAddr=NULL
primary service
[... 86 lines cut ...]
D NimBLEServerCallbacks: onConnParamsUpdate: default
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent:
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent:
D NimBLEServerCallbacks: onConnParamsUpdate: default
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent:
Server Passkey Display
D NimBLEServer: BLE_SM_IOACT_DISP; ble_sm_inject_io result: 0
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent:
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent:
D NimBLEServerCallbacks: onConnParamsUpdate: default
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent:
Secured connection to: 45:ab:fd:c6:a4:68
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent:
Client disconnected - start advertising
D NimBLEAdvertising: >> Advertising start: duration=0, dirAddr=NULL
D NimBLEAdvertising: << Advertising start
D NimBLEServer: << handleGapEvent
```

h2zero, 2025-01-11:
Try changing the second parameter in `NimBLEDevice::setSecurityAuth(true, false, false);` to  `true`, that is what enable MITM protection, which is what the passkey is.

goddade, 2025-01-11:
I tried `true, true, false` first. They have the same error.

h2zero, 2025-01-12:
Thank you @goddade, I have found a bug in the NimBLE stack that I will address shortly. In the meantime the workaround to resolve this for your use is to add `NimBLEDevice::setSecurityInitKey(BLE_SM_PAIR_KEY_DIST_ENC);` after `NimBLEDevice::setSecurityAuth(true, true, false);`

goddade, 2025-01-12:
It can bond now.
Thanks for your patience and help!

closed 2025-01-12 by goddade