h2zero/NimBLE-Arduino #826: Scan works on on board but fails on another one

matthias-bs, 2024-12-29:
Hi,

I'm trying to port https://github.com/theengs/decoder/blob/development/examples/ESP32/ScanAndDecode/ScanAndDecode.ino to NimBLE-Arduino 2.1.2.

You can find my code at https://github.com/matthias-bs/BresserWeatherSensorLW/tree/main/extras/BLETest.

The result depends on the board I'm using:
1. [Lilygo T3 S3 LR1121](https://lilygo.cc/products/t3-s3-lr1121) - works as expected
2. [Lilygo LoRa32](https://lilygo.cc/products/lora3?variant=42272562282677) - fails with `E NimBLEScan: Error starting scan; rc=530, `

<details><summary>Lilygo T3 S3 LR1121 logfile - o.k.</summary>
<p>
```
17:32:52:474 -> =========== Before Setup Start ===========
17:32:52:474 -> Chip Info:
17:32:52:474 -> ------------------------------------------
17:32:52:474 ->   Model             : ESP32-S3
17:32:52:477 ->   Package           : 0
17:32:52:477 ->   Revision          : 0.02
17:32:52:479 ->   Cores             : 2
17:32:52:482 ->   CPU Frequency     : 240 MHz
17:32:52:485 ->   XTAL Frequency    : 40 MHz
17:32:52:487 ->   Features Bitfield : 0x00000012
17:32:52:491 ->   Embedded Flash    : No
17:32:52:492 ->   Embedded PSRAM    : No
17:32:52:494 ->   2.4GHz WiFi       : Yes
17:32:52:496 ->   Classic BT        : No
17:32:52:498 ->   BT Low Energy     : Yes
17:32:52:501 ->   IEEE 802.15.4     : No
17:32:52:505 -> ------------------------------------------
17:32:52:508 -> INTERNAL Memory Info:
17:32:52:510 -> ------------------------------------------
17:32:52:514 ->   Total Size        :   361188 B ( 352.7 KB)
17:32:52:518 ->   Free Bytes        :   323272 B ( 315.7 KB)
17:32:52:523 ->   Allocated Bytes   :    32756 B (  32.0 KB)
17:32:52:526 ->   Minimum Free Bytes:   319468 B ( 312.0 KB)
17:32:52:530 ->   Largest Free Block:   258036 B ( 252.0 KB)
17:32:52:534 -> ------------------------------------------
17:32:52:535 -> Flash Info:
17:32:52:539 -> ------------------------------------------
17:32:52:543 ->   Chip Size         :  4194304 B (4 MB)
17:32:52:547 ->   Block Size        :    65536 B (  64.0 KB)
17:32:52:551 ->   Sector Size       :     4096 B (   4.0 KB)
17:32:52:555 ->   Page Size         :      256 B (   0.2 KB)
17:32:52:557 ->   Bus Speed         : 80 MHz
17:32:52:560 ->   Bus Mode          : QIO
17:32:52:563 -> ------------------------------------------
17:32:52:565 -> Partitions Info:
17:32:52:569 -> ------------------------------------------
17:32:52:576 ->                 nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
17:32:52:583 ->             otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
17:32:52:591 ->                app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
17:32:52:598 ->                app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
17:32:52:606 ->              spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
17:32:52:614 ->            coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
17:32:52:618 -> ------------------------------------------
17:32:52:619 -> Software Info:
17:32:52:623 -> ------------------------------------------
17:32:52:627 ->   Compile Date/Time : Dec 29 2024 17:28:41
17:32:52:629 ->   Compile Host OS   : linux
17:32:52:634 ->   ESP-IDF Version   : v5.3.2-174-g083aad99cf-dirty
17:32:52:637 ->   Arduino Version   : 3.1.0
17:32:52:641 -> ------------------------------------------
17:32:52:641 -> Board Info:
17:32:52:645 -> ------------------------------------------
17:32:52:649 ->   Arduino Board     : LILYGO_T3S3_LR1121
17:32:52:653 ->   Arduino Variant   : lilygo_t3_s3_lr1121
17:32:52:664 ->   Arduino FQBN      : esp32:esp32:lilygo_t3s3:PSRAM=disabled,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=debug,EraseFlash=none,Revision=Radio_LR1121
17:32:52:683 -> ============ Before Setup End ============
17:32:52:784 -> [  1063][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 19 already has type USB_DM (45) with bus 0x3fc9eb74
[... 137 lines cut ...]
17:23:56:292 -> D NimBLEScan: << start()
17:23:56:292 -> =========== After Setup Start ============
17:23:56:301 -> INTERNAL Memory Info:
17:23:56:301 -> ------------------------------------------
17:23:56:312 ->   Total Size        :   292252 B ( 285.4 KB)
17:23:56:312 ->   Free Bytes        :   218396 B ( 213.3 KB)
17:23:56:312 ->   Allocated Bytes   :    66604 B (  65.0 KB)
17:23:56:323 ->   Minimum Free Bytes:   218208 B ( 213.1 KB)
17:23:56:323 ->   Largest Free Block:   110580 B ( 108.0 KB)
17:23:56:323 -> ------------------------------------------
17:23:56:334 -> GPIO Info:
17:23:56:334 -> ------------------------------------------
17:23:56:334 ->   GPIO : BUS_TYPE[bus/unit][chan]
17:23:56:334 ->   --------------------------------------  
17:23:56:344 ->      1 : UART_TX[0]
17:23:56:344 ->      3 : UART_RX[0]
17:23:56:344 -> ============ After Setup End =============
```
</p>
</details>

Any ideas what might be the cause?

Best regards,
Matthias

h2zero, 2025-01-02:
Not sure exactly but the controller is not happy with the scan parameters, possibly this:   `_pBLEScan->setFilterPolicy(CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE);`

matthias-bs, 2025-01-02:
I commented `_pBLEScan->setFilterPolicy(CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE);` out, but the result remains the same.

h2zero, 2025-01-04:
@matthias-bs Could you try one of the examples and see if you still have this error. I don't really see any issues in your code.

matthias-bs, 2025-01-04:
@h2zero Both NimBLE_active_passive_scan and NimBLE_Scan_Continuous work as expected. That's weird...

matthias-bs, 2025-01-21:
It seems  `E NimBLEScan: Error starting scan; rc=530` was caused by
```
_pBLEScan->setFilterPolicy(CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE);
```

I found https://github.com/h2zero/NimBLE-Arduino/blob/3081bc1bb8db1ea6c67a1e06da63a2c4c833e4fb/src/NimBLEDevice.cpp#L262 which says
```
NimBLEDevice::setScanFilterMode(CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE)
```
has to be callled before
```
NimBLEDevice::init()
```
This works with an ESP32, but gives a linker error on ESP32-S3:
https://github.com/matthias-bs/BresserWeatherSensorLW/actions/runs/12881867796/job/35913153354?pr=125

Is the function `NimBLEDevice::setScanFilterMode()` really not available on ESP32-S3 or is it a flaw in the library code?

h2zero, 2025-01-21:
Looks like espressif changed the macro that enables this for the s3/c3. For the same effect you can add this flag to the build: `-D CONFIG_BT_CTRL_SCAN_DUPL_TYPE=2`

matthias-bs, 2025-01-21:
Is there a chance that this feature will be restored in NimBLE-Arduino?
I don't know how to apply any defines in the Arduino IDE, which is still the tool of choice for most users.

h2zero, 2025-01-21:
Yes, I will add the necessary macros to fix this.

h2zero, 2025-01-21:
@matthias-bs If you add a file called `build_opt.h` to your sketch you can put the defines in there.

matthias-bs, 2025-01-21:
@h2zero Ah, thanks for the hint!

closed 2025-01-26 by h2zero

matthias-bs, 2025-01-28:
@h2zero Thanks a lot for resolving this!

PR #877 (2025-01-26): Rename config macros to enable duplicate scan options on s3/c3
Fixes #826