h2zero/NimBLE-Arduino #829: Crash when calling NimBLEScan::clearResults()

thorrak, 2024-12-30:
Currently, my program calls the following inside `loop()`:

```c++
bool tiltScanner::scan()
{
    bool retval = false;
    if (shouldRun) {
        if (!pBLEScan->isScanning()) { // Check if scan already in progress
            //Try to start a new scan
            yield();
            if (pBLEScan->start(BLE_SCAN_TIME, false, true)) {
                retval = true; //Scan successfully started.
            } else {
                Log.verbose(F("Scan failed to start.\r\n"));
            }
        }
    }
    return retval;
}
```

Since `continue` is false, this results in `NimBLEScan::clearResults()` being called. Unfortunately, since updating to NimBLE-Arduino 2.1.2 I'm getting crashes (heap poisoning) from inside clearResults:

```
--- Terminal on /dev/cu.usbserial-2142420 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
E (720) esp_core_dump_flash: No core dump partition found!

CORRUPT HEAP: Bad head at 0x3ffdd020. Expected 0xabba1234 got 0x3ffcddfc

assert failed: multi_heap_free multi_heap_poisoning.c:279 (head != NULL)

Backtrace: 0x4008352d:0x3ffd21e0 0x40096d59:0x3ffd2200 0x4009d14a:0x3ffd2220 0x4009be53:0x3ffd2350 0x400836df:0x3ffd2370 0x4009d17d:0x3ffd2390 0x401a735d:0x3ffd23b0 0x401a6515:0x3ffd23d0 0x400ee3d7:0x3ffd23f0 0x400ee3ff:0x3ffd2410 0x400ee471:0x3ffd2440 0x400debf5:0x3ffd2460 0x400d8d86:0x3ffd2480 0x40114360:0x3ffd24a0 0x40097f26:0x3ffd24c0
  #0  0x4008352d in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:463
  #1  0x40096d59 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:92
  #2  0x4009d14a in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:80
  #3  0x4009be53 in multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:279 (discriminator 1)
  #4  0x400836df in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps_base.c:70
  #5  0x4009d17d in free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/heap.c:39
  #6  0x401a735d in operator delete(void*) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49
  #7  0x401a6515 in operator delete(void*, unsigned int) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/del_ops.cc:33
  #8  0x400ee3d7 in std::__new_allocator<NimBLEAdvertisedDevice*>::deallocate(NimBLEAdvertisedDevice**, unsigned int) at /Users/jbeeler/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/new_allocator.h:168
      (inlined by) std::allocator_traits<std::allocator<NimBLEAdvertisedDevice*> >::deallocate(std::allocator<NimBLEAdvertisedDevice*>&, NimBLEAdvertisedDevice**, unsigned int) at /Users/jbeeler/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/alloc_traits.h:516
      (inlined by) std::_Vector_base<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> >::_M_deallocate(NimBLEAdvertisedDevice**, unsigned int) at /Users/jbeeler/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_vector.h:387
      (inlined by) std::_Vector_base<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> >::~_Vector_base() at /Users/jbeeler/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_vector.h:366
  #9  0x400ee3ff in std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> >::~vector() at /Users/jbeeler/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_vector.h:735
      (inlined by) NimBLEScan::clearResults() at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/NimBLEScan.cpp:474
  #10 0x400ee471 in NimBLEScan::start(unsigned long, bool, bool) at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/NimBLEScan.cpp:310
  #11 0x400debf5 in tiltScanner::scan() at .pio/libdeps/d32_pro_tft/ArduinoJson/src/ArduinoJson/Object/ObjectImpl.hpp:39
  #12 0x400d8d86 in loop() at .pio/libdeps/d32_pro_tft/ArduinoJson/src/ArduinoJson/Object/ObjectImpl.hpp:39
  #13 0x40114360 in loopTask(void*) at /Users/jbeeler/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:74
  #14 0x40097f26 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

What's weird about this is that it seems to happen randomly. I've had it occur after ~24 hours of scanning/restarting, and I've had it occur after a few minutes.

thorrak, 2024-12-30:
Interesting, while I was typing this, I've had it occur inside `handleGapEvent` (also inside the `clearResults` call when `pScan-Mm_maxResults == 0`:

```
CORRUPT HEAP: Bad head at 0x3ffde078. Expected 0xabba1234 got 0x3ffcddfc

assert failed: multi_heap_free multi_heap_poisoning.c:279 (head != NULL)

Backtrace: 0x4008352d:0x3ffe5e60 0x40096d59:0x3ffe5e80 0x4009d14a:0x3ffe5ea0 0x4009be53:0x3ffe5fd0 0x400836df:0x3ffe5ff0 0x4009d17d:0x3ffe6010 0x401a735d:0x3ffe6030 0x401a6515:0x3ffe6050 0x400ee3d7:0x3ffe6070 0x400ee3ff:0x3ffe6090 0x400ee6e9:0x3ffe60c0 0x400f08b1:0x3ffe60f0 0x400f0951:0x3ffe6110 0x400f0d8a:0x3ffe6180 0x400f3dbf:0x3ffe61a0 0x40081465:0x3ffe61c0 0x400ee183:0x3ffe61e0 0x40097f26:0x3ffe6200
  #0  0x4008352d in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:463
  #1  0x40096d59 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:92
  #2  0x4009d14a in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:80
  #3  0x4009be53 in multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:279 (discriminator 1)
  #4  0x400836df in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps_base.c:70
  #5  0x4009d17d in free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/heap.c:39
  #6  0x401a735d in operator delete(void*) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49
  #7  0x401a6515 in operator delete(void*, unsigned int) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/del_ops.cc:33
  #8  0x400ee3d7 in std::__new_allocator<NimBLEAdvertisedDevice*>::deallocate(NimBLEAdvertisedDevice**, unsigned int) at /Users/jbeeler/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/new_allocator.h:168
      (inlined by) std::allocator_traits<std::allocator<NimBLEAdvertisedDevice*> >::deallocate(std::allocator<NimBLEAdvertisedDevice*>&, NimBLEAdvertisedDevice**, unsigned int) at /Users/jbeeler/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/alloc_traits.h:516
      (inlined by) std::_Vector_base<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> >::_M_deallocate(NimBLEAdvertisedDevice**, unsigned int) at /Users/jbeeler/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_vector.h:387
      (inlined by) std::_Vector_base<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> >::~_Vector_base() at /Users/jbeeler/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_vector.h:366
  #9  0x400ee3ff in std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> >::~vector() at /Users/jbeeler/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_vector.h:735
      (inlined by) NimBLEScan::clearResults() at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/NimBLEScan.cpp:474
  #10 0x400ee6e9 in NimBLEScan::handleGapEvent(ble_gap_event*, void*) at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/NimBLEScan.cpp:149
  #11 0x400f08b1 in ble_gap_call_event_cb at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:839
  #12 0x400f0951 in ble_gap_disc_complete at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:1204
  #13 0x400f0d8a in ble_gap_master_timer at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:2401
      (inlined by) ble_gap_timer at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:2513
  #14 0x400f3dbf in ble_hs_timer_exp at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:434
      (inlined by) ble_hs_timer_exp at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:414
  #15 0x40081465 in ble_npl_event_run at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:527
      (inlined by) nimble_port_run at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/nimble/porting/nimble/src/nimble_port.c:313
  #16 0x400ee183 in NimBLEDevice::host_task(void*) at .pio/libdeps/d32_pro_tft/NimBLE-Arduino/src/NimBLEDevice.cpp:811
  #17 0x40097f26 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

h2zero, 2025-01-02:
@thorrak could you test with this: https://github.com/h2zero/esp-nimble-cpp/pull/278 and let me know?

thorrak, 2025-01-02:
Yep -- Have two devices running this now. We'll see how they fare over the next ~24 hours!

thorrak, 2025-01-03:
One of the two crashed overnight last night, but I wasn't able to get a dump to be able to see if it was the same issue. I've changed how both devices are running to (hopefully) capture a dump if they crash again today.

h2zero, 2025-01-03:
Great, thanks for the update!

thekurtovic, 2025-01-03:
The backtrace from the crash are unlikely to be helpful. As you already observed, the crash can occur at different points, and is a symptom of the corruption opposed to being directly related to the cause. I experienced similar issues where it appeared to be random, but once I switched from `HEAP_POISONING_LIGHT` to `HEAP_POISONING_COMPREHENSIVE` it was consistent and allowed me to zero in on it.

Your first log suggests the crash occurred within one second of run-time, has it crashed without receiving any advertisements first? Could you post your scan callbacks?

thorrak, 2025-01-03:
Mostly, I'm wondering if the trace still indicates that the crash is occurring in `clearResults` as opposed to elsewhere. For both of the traces that I posted, the crash is in `clearResults` -- regardless of how `clearResults` is called.

The scan callbacks are available [here](https://github.com/thorrak/tiltbridge/blob/nimble_fix/src/tilt/tiltScanner.cpp#L20-L33), and are copied below:

```c++
void ScanCallbacks::onResult(const NimBLEAdvertisedDevice* advertisedDevice)
{
    if (advertisedDevice->getManufacturerData().length() >= 24)
    {
        if (advertisedDevice->getManufacturerData()[0] == 0x4c && advertisedDevice->getManufacturerData()[1] == 0x00 &&
            advertisedDevice->getManufacturerData()[2] == 0x02 && advertisedDevice->getManufacturerData()[3] == 0x15)
        {
#ifdef BLE_PRINT_ALL_DEVICES
            Log.verbose(F("Advertised iBeacon Device: %s \r\n"), advertisedDevice->toString().c_str());
#endif
            tilt_scanner.load_tilt_from_advert_hex(advertisedDevice->getManufacturerData(), advertisedDevice->getRSSI());
        }
    }
}
```

`tilt_scanner.load_tilt_from_advert_hex()` is available [here](https://github.com/thorrak/tiltbridge/blob/6b9b334cfb58e2512c24bd3fe9932fa1fd7b09c8/src/tilt/tiltScanner.cpp#L98-L153), but is mostly just processing the various chunks of the manufacturer data according to my application.

I think that the fact that the crash happened within a second of runtime in this one instance was a red herring, unfortunately. That was the only time I had seen crashes occur that quickly after boot -- previously, it mostly occurred after several hours.

thekurtovic, 2025-01-03:
I missed that originally, but my point still stands about the crash being a symptom of the corruption. In my case I was typically seeing the same 2-3 offenders since the code was flowing in a predictable manner. I suspect it is happening in `clearResults` because your code just happens to restart the scan after the corruption occurs, and that call to start is what is detecting the corruption when `NimBLEScan::clearResults` allocates memory with `std::vector<NimBLEAdvertisedDevice*>().swap(m_scanResults.m_deviceVec);`.

You could try adding calls to `heap_caps_check_integrity_all` in your code to try to isolate the issue, but I never had any success with it. Are you able to increase the detection to `HEAP_POISONING_COMPREHENSIVE`? I think that would help in terms of reproducibility.
More info here https://docs.espressif.com/projects/esp-idf/en/v5.3.2/esp32s3/api-reference/system/heap_debug.html

My best advice would be to simplify your project as much as possible. Comment out anything unrelated to the core loop of your BLE application, then slowly reintroduce the pieces until you find a link.

h2zero, 2025-01-04:
@thorrak Any new crashes?

thorrak, 2025-01-06:
No, so I _think_ this is good to close with merging that change into the NimBLE-Arduino library, but it's hard to say, because...

> Are you able to increase the detection to HEAP_POISONING_COMPREHENSIVE? I think that would help in terms of reproducibility.

I don't think this change can be made with the Arduino framework as most of the menuconfig options are precompiled with the framework packages themselves. To be able to do this I went ahead and converted my project to use ESP-IDF with Arduino as a component. Interestingly (expectedly?) the _only_ library that I wasn't able to directly use was NimBLE-Arduino. I had to switch from NimBLE-Arduino to [esp-nimble-cpp](https://github.com/h2zero/esp-nimble-cpp) for it to compile due to some typedef differences on Timer classes.

With the above said: After all that, thus far, I've not experienced any crashes:

[image]

thorrak, 2025-01-07:
Since https://github.com/h2zero/esp-nimble-cpp/pull/278 appears to be merged to master and I haven't experienced any more crashes since, I think this is good to close. I'll open a new issue if I experience any further issues.

Thanks again to both of you guys - @thekurtovic and @h2zero - I appreciate everything you've done!

closed 2025-01-07 by thorrak