h2zero/NimBLE-Arduino #831: Unable to Change Default Device Name in NimBLE

faizan-ali-mlk, 2025-01-01:
I attempted to change the name of the device, but it consistently shows the default name "NimBLE" during scans. I used the following code to set the device name:

```cpp
NimBLEDevice::init("ESP32 Device");
NimBLEDevice::setDeviceName("ESP32 Device");
```

However, this did not work, and the device continues to display the default name. 

I tried investigating where the default name is being set, but I could not locate the source of the issue.

**Workaround:**

The following code successfully changes the name, but I would prefer not to set the name in this manner:

```cpp
NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
pAdvertising->setName("ESP32->Device");
```

**Request for Assistance:**

Could you provide guidance on how to correctly set the device name using `NimBLEDevice::init` or `NimBLEDevice::setDeviceName` without relying on the advertising configuration? Additionally, if there is a known limitation or a different approach required, please let me know.

h2zero, 2025-01-02:
`NimBLEDevice::setDeviceName("ESP32 Device");` set's the device name that is read from the device name characteristic and is stored by the cache in your phone so it will always display this until it's read again or the cache is cleared. If you try to find the device with a different phone that has not seen it before then the new name will be shown there.

`pAdvertising->setName("ESP32->Device");` is the name that is actually advertised in the packet constantly and therefore updated by the phone each time the packet is seen.

faizan-ali-mlk, 2025-01-02:
Thanks for explaining. That is why, after writing my firmware to my device a few times, the given names appeared again.

h2zero, 2025-01-04:
@faizan-ali-mlk Can we close this issue then?

faizan-ali-mlk, 2025-01-05:
I understand what you said, but I don't know when the value will update again. I also tried calling:

```cpp
NimBLEDevice::deinit(true);
```

However, it still shows the old value. I think it would be better if you could provide a method to clear the cache so that we can update the name easily. Right now, being stuck with one name and waiting for it to update again is not very helpful in my current scenario.

Otherwise, if you still believe this is not an issue, feel free to close it.

h2zero, 2025-01-05:
This is not something the library is capable of, the cache exists on the phone and would need to be done by the phone application.

faizan-ali-mlk, 2025-01-05:
Why are you saying it is on a phone? I am using the library for an ESP32 chip, not a phone. Are you referring to an actual phone?

h2zero, 2025-01-05:
What device are you scanning the esp32 with? How are you seeming the device name not changing? I assumed with a phone as that is most common.

faizan-ali-mlk, 2025-01-05:
I have this kind of ESP32 microcontroller

https://www.amazon.de/-/en/Development-Bluetooth-Microcontroller-Processor-Antenna/dp/B07S1GGM35

h2zero, 2025-01-05:
Is that what you are using to do scan? Where are you seeing the name of the device?

faizan-ali-mlk, 2025-01-05:
I have written firmware using your library, which is installed on an ESP32. I am trying to connect this device to a mobile app. My main goal is to transfer data from the ESP32 to a mobile device. On the mobile device, I am not using your library; instead, it is based on pure Android and iOS libraries written in Kotlin. In the app, I search for nearby Bluetooth devices, and I can see the name of the ESP32 device.

h2zero, 2025-01-05:
Thank you, that is what I thought. When I am talking about the phone cache that is not controllable from the esp32, it's up to the phone app to deal with the name displayed or clear the cached data so that the name will be read by the phone from esp32 again to update it. The issue you are seeing is not anything that can be done by the esp32, it's name has changed but the application on the phone or other devices needs to read the name to update what is displayed, this doesn't happen automatically, so if this is a custom app on the phone you will need to force it to read the name from that side.

faizan-ali-mlk, 2025-01-06:
It is my custom app, and I also thought that maybe you were referring to the phone app's cache. However, even after deleting and reinstalling the app, the name still shows the same. This makes me confident that the cache does not belong to the app. So, what kind of cache are you referring to, then?

h2zero, 2025-01-07:
The cache is internal to the phone OS and the app get it's data from there, sometimes turning bluetooth off and on again in the phone settings will clear it, otherwise a full reboot. If you want your app to have this name updated regularly though you will need to read the device name characteristic from your app to have it update without clearing the cache etc.