h2zero/NimBLE-Arduino #837: Setting advertisement of custom UUID's are funky

doudar, 2025-01-04:
I have a cycling app that uses some custom generated UUID's that are not advertised after adding them to advertising.

This is using NimBLE-Arduino 2.1.2 and Platform 6.9.0

I've added and removed them (One at a time) for testing and here are my findings:

pAdvertising->addServiceUUID(NimBLEUUID((uint16_t)0x180D)); //Works

pAdvertising->addServiceUUID(NimBLEUUID("b4cc1223-bc02-4cae-adb9-1217ad2860d1")); //No work :(

pAdvertising->addServiceUUID(NimBLEUUID("77776277-7877-7774-4466-896665500000")); //No work :(

//Interestingly, If I do
NimBLEService *pService = spinBLEServer.pServer->createService(NimBLEUUID("b4cc1223-bc02-4cae-adb9-1217ad2860d1"));
spinBLEServer.pServer->getAdvertising()->addServiceUUID(pService->getUUID()); //Works

h2zero, 2025-01-04:
Happy New Year @doudar I'll have a look, strange indeed.

h2zero, 2025-01-04:
@doudar I think I see your issue, make sure you enable scan response before adding anything to the advertising and it should work.

doudar, 2025-01-04:
Thanks! Heading to dinner, I'll let you know if if that works when i get back! 

Happy to see you working on 2.0!

doudar, 2025-01-05:
@h2zero , Enabling scan response BEFORE adding advertising seemed to fix the Advertised name not being set as well as being able to add the UUID to advertising without doing the workaround :

``pServer->getAdvertising()->addServiceUUID(pService->getUUID());```

I still cannot get this UUID to add to advertising, no matter how I set it. Give it a try on your end:
pServer->getAdvertising()->addServiceUUID(NimBLEUUID("b4cc1223-bc02-4cae-adb9-1217ad2860d1"));

h2zero, 2025-01-05:
@doudar that's the code I used to test the related PR #838. Could you try your code using that branch and let me know?

doudar, 2025-01-05:
> @doudar that's the code I used to test the related PR #838. Could you try your code using that branch and let me know?

Testing now

doudar, 2025-01-05:
> @doudar that's the code I used to test the related PR #838. Could you try your code using that branch and let me know?

No dice....Sorry :(

doudar, 2025-01-05:
Btw, just so we're clear, It's that **specific** UUID. The other custom UUID's are working now.

h2zero, 2025-01-05:
Hmmm, any chance of a simplified example to reproduce this? I've run out of possibilities to explain at the moment.

doudar, 2025-01-05:
> Hmmm, any chance of a simplified example to reproduce this? I've run out of possibilities to explain at the moment.

I'll see if I can spin it up this afternoon after work.

doudar, 2025-01-05:
So, I just quickly recompiled the branch I'm working on in SmartSpin2k, and that UUID is advertising, but now a different UUID dropped. Is this possibly a NVS issue? Also, what's the max length of the advertising data? I'm currently trying to set 2 16 bit UUID's and 3 full custom UUID's

h2zero, 2025-01-05:
Ahh, there's your problem. The advertisement only has room for 31 bytes, each 128 bit uuid takes 16 of that plus 2 byes for the header. You can only have 1 of those uuids in the advertisement and another in the scan response.

PR #838 (2025-01-05): Fix advertising data not set if scan response enabled later.
Fixes #837
closed 2025-01-05 by h2zero