h2zero/NimBLE-Arduino #839: BLE OTA update library

h2zero, 2025-01-05:
For anyone interested in OTA updates over BLE on the esp32 using espressif's phone app or @gb88's [webapp](https://gb88.github.io/BLEOTA/). 
I have created a repo, sponsored by [Theengs](https://www.theengs.io).

You can find it [here](https://github.com/h2zero/NimBLEOta) and it's super simple to use, hope you enjoy it!

doudar, 2025-01-05:
Awesome! Too bad I didn't have this for https://github.com/doudar/SS2kConfigApp and https://github.com/doudar/SmartSpin2k . I had to roll my own. 

Just curious, what are your transmission times for a 2mb firmware?

gb88, 2025-01-05:
Great work! Thanks for keep it compatible with my web app!

h2zero, 2025-01-05:
>Just curious, what are your transmission times for a 2mb firmware?

I haven't done any proper profiling of it but it's pretty quick for BLE.

>Great work! Thanks for keep it compatible with my web app!

Thank you for the web app 😄