h2zero/NimBLE-Arduino #842: Can i use PHY 2M with legacy fallback for BLE 4 clients?

puzrin, 2025-01-05:
I'm doing RPC over BLE and am interested in maximizing the speed of long characteristics read/write (without caring about power).

Currently ended with this things:

1. Connection tune after been opened

```cpp
    void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc) override {
        // For BLE 5 clients with DLE extension support - set data packet size to max.
        // This boosts transfer speed to ~ 45 Kb/sec for big transfers.
        pServer->setDataLen(conn_handle, 251);

        // Update connection parameters for maximum performance and stability
        // min conn interval 7.5ms, max conn interval 7.5ms, latency 0, timeout 2s
        // Smaller intervals help with BLE 4 clients without DLE, but makes no sense
        // for BLE 5 clients with DLE.
        pServer->updateConnParams(conn_handle, 0x06, 0x06, 0, 200);
    }
```

2. Long characteristic length should be properly selected, to avoid extra fragmentation

```cpp
    // Max DLE data size is 251 bytes. Every R/W to characteristic should be
    // 244 or 495 bytes to fit into 1 or 2 DLE packets (including overheads).
    // That maximizes speed on bulk transfers.
    // In real world, 495 bytes do not give any notable benefits. So, use 244 bytes.
```

3. In RPC protocol, write confirmations should exist only in last chunks, but this layer is out of this project scope.

---

The last possible thing I see is using PHY 2M. After the above, that's not critical, but it is interesting—just for fun.

The idea was to create multiple advertisements with an extended one - pin secondary channel to 2M. Unfortunately, could not succeed - connection halted after device discovery with 2M secondary channel, and I had not enough time to dive in. So, question is:

1. Is it possible on esp32 (c3) to work in such mode? With simultaneous BLE5 2M and BLE4 1M? With several different clients connected.
2. If yes - may be you know proper way to setup?

PS. I'm closing this to avoid pushing about unfinished issue, but would like to know your opinion.

closed 2025-01-05 by puzrin

h2zero, 2025-01-05:
This should work, just need to change the PHY for the connection after connected. Should be no issue using different PHY's for each connection.

puzrin, 2025-01-05:
> This should work, just need to change the PHY for the connection after connected.

Who is responsible for that? In my case clients are Web Bluetooth-capable browsers. There are no such control (those can operate only at GATT level). And on server side (esp32 + nimble) I could not find the way to investigate client's capability and upgrade PHY. May be you have a recipe to activate 2M from server side?

h2zero, 2025-01-07:
Usually it is the client/master, the function you want is here: https://github.com/h2zero/NimBLE-Arduino/blob/74ac317404aebb4683b99a4e5dd4f963db829575/src/NimBLEClient.cpp#L430

You can get a client instance from the server with `NimBLEServer::getClient`