h2zero/NimBLE-Arduino #845: Two kinds of passkey spelling in version 2.1.3. passkey or pass key?

asukiaaa, 2025-01-06:
Thank you for updating the library.

I found that passkey are written in two kinds spelling.
Which do you want to use?
passkey
or
passKey, pass_key, passKey,

For example, [onConfirmPasskey](https://github.com/h2zero/NimBLE-Arduino/blob/e374f507ef1a7dcbd7cb42ac3d38c52fd27e6936/src/NimBLEClient.h#L200) is passkey style.
On the other hand,  [onPassKeyEntry](https://github.com/h2zero/NimBLE-Arduino/blob/e374f507ef1a7dcbd7cb42ac3d38c52fd27e6936/src/NimBLEClient.h#L186) is passKey style.

h2zero, 2025-01-07:
Thanks, looks like I missed that one. I was intending to have them all formatted as `passkey`, oops.

thekurtovic, 2025-01-10:
Some other occurrences 
https://github.com/h2zero/NimBLE-Arduino/blob/master/src/NimBLEServer.h#L175
https://github.com/h2zero/NimBLE-Arduino/blob/master/src/NimBLEDevice.h#L166