h2zero/NimBLE-Arduino #850: Windows reconnect error - ESP32-BLE-Gamepad

lemmingDev, 2025-01-10:
Hi @h2zero

The latest NimBLE-Arduino 2.1.3 is working with [ESP32-BLE-Gamepad](https://github.com/lemmingDev/ESP32-BLE-Gamepad) after a couple of users added some patches to the library

One issue we seem to have now is that on Windows (tested on Windows 10 and 11), it works fine when you pair it the first time, however when you power the board down and it automatically reconnects, gamepad input no longer works.

The main loop seems to be working as any serial output or LEDs modified in the sketch still update.

Android and perhaps Linux seem to work fine still

If you've got the time; any advice on what might need to be changed?

There is a thread with some testing feedback here: https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/251

Thanks

h2zero, 2025-01-10:
Thanks @lemmingDev, I will try to test this soon. In the meantime maybe @afpineda has some ideas?

lemmingDev, 2025-01-10:
> Hmmm, that sounds like the cccd isn't being saved, maybe need to increase `CONFIG_BT_NIMBLE_MAX_CCCDS`

Set it to 32, but didn't fix the issue unfortunately

h2zero, 2025-01-11:
Thanks @lemmingDev @RaazP, I was able to reproduce this and have found that the error originates in the stack, some error with the storage of the subscription state when windows subscribes to a characteristic, instead of 1 or 2 as is should be, the value is 128. Please bare with me while I troubleshoot, this area of the code is not easy to navigate.

lemmingDev, 2025-01-11:
Thanks for the update!

In my own testing just now, with some sketches working on reconnect, and others not, only difference I could really see in the error log was:

In a sketch where reconnecting worked:

```
D NimBLEAdvertising: setAdvertisementData: 02 01 06 03 19 c4 03 17 09 42 4c 45 20 44 72 69 76 69 6e 67 20 43 6f 6e 74 72 6f 6c 6c 65 72
D NimBLEAdvertising: << Advertising start
[  1323][D][BleGamepad.cpp:1508] taskServer(): [BLEGamepad] Advertising started!
D NimBLEServer: >> handleGapEvent: 
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
D NimBLEServerCallbacks: onConnParamsUpdate: default
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
D NimBLEServerCallbacks: onAuthenticationComplete: default
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: subscribe event; attr_handle=8, subscribed: true
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: subscribe event; attr_handle=44, subscribed: true
D NimBLECharacteristicCallbacks: onSubscribe: default
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: subscribe event; attr_handle=39, subscribed: true
D NimBLECharacteristicCallbacks: onSubscribe: default
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: mtu update event; conn_handle=1 mtu=255
D NimBLEServerCallbacks: onMTUChange(): Default
D NimBLEServer: << handleGapEvent
``` 
and then went on to respond to button presses logging the following output for each:
```
D NimBLEServer: Gatt Read event
D NimBLEServer: >> handleGapEvent: 
D NimBLECharacteristicCallbacks: onStatus: default
D NimBLEServer: << handleGapEvent
```

But when it didn't work correctly on reconnection, it looked pretty much the same, except it seemed to be missing:
```
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: subscribe event; attr_handle=39, subscribed: true
D NimBLECharacteristicCallbacks: onSubscribe: default
D NimBLEServer: << handleGapEvent
```

and then no Gatt Read events were logged

MaksMacioszczyk, 2025-01-11:
> In case you are wondering, adding the `0x2902` descriptor does not fix this issue. I already tried.

For me, adding this descriptor kind of works. When I additionally restart Bluetooth on the Windows side after ESP32 was rebooted, I have working connection again, but it is still not the fix for me.

h2zero, 2025-01-11:
> In case you are wondering, adding the `0x2902` descriptor does not fix this issue. I already tried.
> 
> ```c++
>   /*
>      * Descriptor 2902 is required by the HID over GATT spec
>      * (input reports only), but it works without it.
>     */
> 
>     NimBLEDescriptor *p2902 = inputReportChr->createDescriptor(
>         hidReport2902DscUuid,
>         NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE | NIMBLE_PROPERTY::READ_ENC | NIMBLE_PROPERTY::WRITE_ENC,
>         2);
>     uint8_t desc2_val[] = {0b01, 0}; // Notifications enabled by default
>     p2902->setValue(desc2_val, 2);
> ```

This probably didn't work because NimBLE needs to create the descriptor, just adding the `NIMBLE_PROPERTY::NOTIFY` property to the characteristic would do it.

h2zero, 2025-01-11:
@afpineda The descriptor is created automatically in the NimBLE Host, not in this wrapper. All that is is required is to add NIMBLE_PROPERTY::NOTIFY to the characteristic

h2zero, 2025-01-11:
Original issue should be resolved in https://github.com/lemmingDev/ESP32-BLE-Gamepad/pull/257

This is an upstream oversight that I will pursue a long term solution to, the workaround in the above PR seems to be satisfactory for now.