h2zero/NimBLE-Arduino #852: Feature request: have the default NimbleCharacteristicsCallbacks print the uuid of the characteristics

jackjansen, 2025-01-10:
Not specifying your own NimbleCharacteristicsCallbacks for a characteristic you have added to your server is most likely an error.

And with the change in the API it is an error I made (because my methods that were intended didn't have the new extra argument, and I had stupidly not added the `override` specifier to my declaration, so my methods were quietly sitting there without being used, and the default implementations were used).

It took me a very long time until I found this bug, but if the default callbacks had printed something helpful it would have saved a lot of time.

(Note that I think this feature request holds only for `NimbleCharacteristicsCallbacks` set and get, because these are the only callbacks where it is probably a programmer error if you don't override them.

h2zero, 2025-01-11:
@jackjansen, sorry you had this difficulty.  I'm not sure how printing the UUID would have helped, generally speaking if  the default callbacks are printing a message that means your callback was not overriding it.

jackjansen, 2025-01-15:
Well, printing the UUID would have shown me *which* callbacks I hadn't overridden. I have a large number of services, all implemented in different places (because I'm working on a framework).

But it isn't really urgent for me anymore, I now know to add `override` always:-)