h2zero/NimBLE-Arduino #853: Not receiving value from characteristic->notify() after Reading it (Android and ESP32 Dev Module)

sanchez8096, 2025-01-12:
Hello, I'm making my Android app with MIT App Inventor that controls ESP32 via BLE, using NimBLE library.
In my program when I write a value to ESP32, I want to get it back by using characteristic->notify()
It works fine, unless I Read that value beforehand. Then ESP32 doesn't notify. 

This works:
1. Android connects to ESP32
2. Android writes value to ESP32
3. In ESP32 in onWrite() callback: value is taken with getValue(), saved to a variable, and then is notified back to Android with 
characteristic->notify()
4. Android receives that value back

This doesn't work:
1. Android connects to ESP32
2. Android reads value (by clicking a button in my example) and value is successfully read
3. Android writes value to ESP32
4. In ESP32 in onWrite() callback: value is taken with getValue(), saved to a variable, but when doing characteristic->notify(), Android doesn't receive the value back.

<details>
  <summary>The code</summary>
  
  ```
#include <NimBLEDevice.h>

uint16_t value = 55;

BLEServer *p_server = NULL;
BLEService *p_service = NULL;
BLECharacteristic *VALUE_characteristic = NULL;

#define SERVICE_UUID        "00000000-c502-4b00-ac6d-454c610489f2"
#define DEVICE_BT_NAME      "Nimble Device"
#define VALUE_UUID          "00000010-c502-4b00-ac6d-454c610489f2"

class ServerCallBacks: public NimBLEServerCallbacks {
    void onConnect(NimBLEServer* pServer, NimBLEConnInfo& connInfo){
      Serial.printf("Client address: %s\n", connInfo.getAddress().toString().c_str());
      pServer->updateConnParams(connInfo.getConnHandle(), 24, 48, 0, 180);
    }
    void onDisconnect(NimBLEServer* pServer, NimBLEConnInfo& connInfo, int reason){
      Serial.printf("Client disconnected - start advertising\n");
      NimBLEDevice::startAdvertising();
    }
} server_call_backs;

class CharacteristicCallbacks: public NimBLECharacteristicCallbacks {
  void onRead(NimBLECharacteristic* characteristic, NimBLEConnInfo& connInfo) {
    if (characteristic->getUUID().equals(BLEUUID(VALUE_UUID))){
      Serial.print("Android is reading value from ESP32: "); Serial.println(value);
      VALUE_characteristic->setValue(String(value));
    }
  }
  void onWrite(NimBLECharacteristic* characteristic, NimBLEConnInfo& connInfo){
    String string_value = characteristic->getValue();
    uint16_t int_num = static_cast<uint16_t>(string_value.toInt());
    if (characteristic->getUUID().equals(BLEUUID(VALUE_UUID))){
      Serial.print("Android is writing value to ESP32: "); Serial.println(int_num);
      value = int_num;
      VALUE_characteristic->setValue(String(value));
      VALUE_characteristic->notify();
    }
  }
} characteristic_call_backs;

void setup() {
    Serial.begin(9600);

    NimBLEDevice::init(DEVICE_BT_NAME);
    p_server = NimBLEDevice::createServer();
    p_server->setCallbacks(&server_call_backs);

    p_service = p_server->createService(SERVICE_UUID);

    VALUE_characteristic = p_service->createCharacteristic(VALUE_UUID, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE_NR | NIMBLE_PROPERTY::NOTIFY);
    VALUE_characteristic->setCallbacks(&characteristic_call_backs);
  
    p_service->start();

    NimBLEAdvertising *p_advertising = NimBLEDevice::getAdvertising();
    p_advertising->setName(DEVICE_BT_NAME);
    p_advertising->addServiceUUID(SERVICE_UUID);
    p_advertising->enableScanResponse(true);
    p_advertising->start();
}

void loop() {
}
  ```
  
</details>

<details>
  <summary>The MIT App blocks</summary>
  
[image]

  
</details>

h2zero, 2025-01-12:
Could you enable debug logs and post them here so I can see what's happening?

h2zero, 2025-01-12:
I have tested your onWrite callback locally and am unable to reproduce this issue. The value is notified to nRFConnect without issue.

sanchez8096, 2025-01-12:
Unless I'm doing something wrong, I get an error when trying to debug. I'm really not good at this.
<details>
  <summary>Error</summary>

```
Reading symbols from /home/alex/Documents/PlatformIO/Projects/ble_test/.pio/build/esp32dev/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = cmsis-dap
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Error: unable to find a matching CMSIS-DAP device

.pioinit:11: Error in sourced command file:
Remote communication error.  Target disconnected.: Connection reset by peer.
```

</details>

> I have tested your onWrite callback locally and am unable to reproduce this issue. The value is notified to nRFConnect without issue.

The issue happens when I Read characteristic first and then Notify it later. If I don't Read it before Notifying, I receive the Notified value.

h2zero, 2025-01-12:
What I meant was enabling the debug logs, build flag is  `CONFIG_NIMBLE_CPP_LOG_LEVEL=5`, no need to HW debug. I'm not familiar with the MIT App Inventor so I do not know how you would set this build flag in there, in platformio it would be 
```
build_flags = '-DCONFIG_NIMBLE_CPP_LOG_LEVEL=5'
```

h2zero, 2025-01-12:
Is it possible that your android device is not subscribed to notifications, or somehow becomes unsubscribed after reading the value?

sanchez8096, 2025-01-12:
> Is it possible that your android device is not subscribed to notifications, or somehow becomes unsubscribed after reading the value?

This turned out to be the case. The MIT App for some reason unsubscribes after Reading. I set it to RegisterForStrings again right after Reading and I started to receive Notifications again in the app. 
Not an issue with NimBLE, sorry! And thank you for helping.

closed 2025-01-12 by sanchez8096

h2zero, 2025-01-12:
Thanks for the update, glad you got it working 👍

sanchez8096, 2025-01-12:
Slight update or if anyone else finds this. After registering for values, don't Read them. Instead use Notifications. If you do want to explicitly Read values, you shouldn't Notify that value. I hope it's not too confusing.
[Read here](https://community.appinventor.mit.edu/t/ble-unable-to-write-after-read/77744)