h2zero/NimBLE-Arduino #857: NICE NANO can't run NIMBLE

FreeLarssen, 2025-01-13:
I use Nice nano to run nimble,I use these lib:n-able Arduino core 0.2.0;NimBLE-Arduino 2.2.0;adafruit_nrfutil-0.5.3.
I can compile and update code;but then mcu don't run;
[image]

h2zero, 2025-01-13:
Did you set the bootloader to adafruit before compiling?

FreeLarssen, 2025-01-13:
> Did you set the bootloader to adafruit before compiling? 

yes，I did.[image]I used Adafruit nrf52 lib,it work.but I wanna to use Nimble.

h2zero, 2025-01-13:
You probably need to set it to the other adafruit bootloader, with softdevice as that is likely what is installed already. If you have a debug tool I would suggest erasing the bootloader or installing the no-softdevice version.