h2zero/NimBLE-Arduino #861: NimBLEClient::discoverAttributes() always(?) fails

gkoh, 2025-01-15:
I updated a project from v2.1.3 to v2.2.0 and found some oddities with writing to a single remote characteristic (not convinced this is a NimBLE-Arduino issue).

Eventually, I just tried to `discoverAttributes()` and found it was always failing with:
```
D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a00
D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 0 handle: 4
D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 0 handle: 5
D NimBLERemoteCharacteristic: Descriptor Discovery >> status: 14 handle: -1
D NimBLERemoteCharacteristic: << Descriptor Discovery
D NimBLERemoteCharacteristic: << retrieveDescriptors(): found 2 descriptors.
D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a01
E NimBLERemoteCharacteristic: ble_gattc_disc_all_dscs: rc=3 
[ 41507][I][main.ino:39] loop(): [test] Failed to discover attributes
[ 41524][I][main.ino:43] loop(): [test] Disconnecting
```

I slashed everything back and it seems that regardless of target device, if the device responds to a connect and has characteristics `0x2a00` and `0x2a01` (which is most devices), `discoverAttributes()` fails in the same way where `rc=3` == `BLE_HS_EINVAL`.

I used the following Arduino code on an M5Stack Core2:
```
#include <NimBLEDevice.h>

static const char *LOG_TAG = "test";

bool found = false;
NimBLEAddress address;
NimBLEScan *scan = nullptr;

class ScanCB : public NimBLEScanCallbacks {
  void onResult(const NimBLEAdvertisedDevice *pDevice) override {
    ESP_LOGI(LOG_TAG, "Device found");
    address = pDevice->getAddress();
    found = true;
  }
};

void setup() {
  NimBLEDevice::init("ugh");
  scan = NimBLEDevice::getScan();
  scan->setScanCallbacks(new ScanCB());
  scan->setActiveScan(true);
  scan->setInterval(6553);
  scan->setWindow(6553);
}

void loop() {
  if (!found) {
    if (!scan->isScanning()) {
      scan->start(0, false);
    }
  } else {
    scan->stop();

    auto *client = NimBLEDevice::createClient();
    client->setConnectTimeout(10000);
    if (client->connect(address)) {
      ESP_LOGI(LOG_TAG, "Connected");
      if (!client->discoverAttributes()) {
        ESP_LOGI(LOG_TAG, "Failed to discover attributes");
      } else {
        ESP_LOGI(LOG_TAG, "OK discover attributes");
      }
      ESP_LOGI(LOG_TAG, "Disconnecting");
      client->disconnect();
    }
    NimBLEDevice::deleteClient(client);
    found = false;
  }
  delay(200);
}
```

and `platformio.ini`:
```
[env:default]
platform = espressif32
board = m5stack-core2
board_build.f_cpu = 80000000L
upload_protocol = esptool
framework = arduino
build_flags = -Wall -Wextra
  -DCORE_DEBUG_LEVEL=5
  -DCONFIG_BT_NIMBLE_LOG_LEVEL=0
  -DCONFIG_NIMBLE_CPP_LOG_LEVEL=4
lib_deps =
  https://github.com/h2zero/NimBLE-Arduino#2.2.0
```

And started bisecting, because I recall this used to work at some point in the recent past.
I got it down to 853241bab430fb14cc11d2117e30f5ab2e5499b5.

Following is my failed attempt to narrow the problem ...

I tried digging a bit further, but got lost in the bowels of NimBLE somewhere around `ble_gattc_disc_all_dscs_tx()`.
The call to `ble_att_clt_tx_find_info()` looks like:
```
    rc = ble_att_clt_tx_find_info(proc->conn_handle,
                                  proc->disc_all_dscs.prev_handle + 1,
                                  proc->disc_all_dscs.end_handle);
```

where `prev_handle + 1` seems to be greater than `end_handle` (if I chuck in more debug output), which perhaps trips the `EINVAL`.
So, I guess the second call to `ble_gattc_disc_all_dscs()` during discovery is passing a wrong `start_handle` parameter?

h2zero, 2025-01-15:
Thank you @gkoh, I think the issue here is that there are no descriptors for that characteristic. The end handle and the start handle are equal and therefore when +1 is added to it we get this error. What should be corrected is before we call this at all we should check the handle values.

gkoh, 2025-01-16:
Aha, I didn't even consider the characteristic had no descriptors.
I reviewed the commit I found via bisect and this check is now missing in `NimBLERemoteCharacteristic::retrieveDescriptors()`:
```
    // If this is the last handle then there are no descriptors
    if (m_handle == getRemoteService()->getEndHandle()) {
        return true;
    }
```

gkoh, 2025-01-16:
PR opened, fixes it for me.

PR #863 (2025-01-16): Fix characteristic discovery with no descriptors.
Avoid discovery of descriptors if there are no handles remaining, fixes #861.

This reinstates the check removed in 853241b.
closed 2025-01-16 by h2zero