h2zero/NimBLE-Arduino #862: Can't inherit or use Extended Advertising

AlistairMRoss, 2025-01-15:
_From Example_

class AdvertisingCallbacks : public NimBLEExtAdvertisingCallbacks {
    void onStopped(NimBLEExtAdvertising* pAdv, int reason, uint8_t instId) override {
        /* Check the reason advertising stopped, don't sleep if client is connecting */
        Serial.printf("Advertising instance %u stopped\n", instId);
        switch (reason) {
            case 0:
                Serial.printf("Client connecting\n");
                return;
            case BLE_HS_ETIMEOUT:
                Serial.printf("Time expired - sleeping for %" PRIu32 " seconds\n", sleepSeconds);
                break;
            default:
                break;
        }

Error: NimBLEExtAdvertisingCallbacks - not a class or struct name
onStopped - member function declared with 'override' does not override a base class member

h2zero, 2025-01-15:
Looks like you have not enabled extended advertising. https://github.com/h2zero/NimBLE-Arduino/blob/4464e3f8fb581eb6e30a913754cf636413a76809/src/nimconfig.h#L39

h2zero, 2025-04-23:
Closing this as stale, please re-open if still an issue.

closed 2025-04-23 by h2zero