h2zero/NimBLE-Arduino #864: Notifications from 1.4.3 to 2.2.0

thomasMEBaker, 2025-01-16:
Hi @h2zero,

I'm not receiving notifications on client devices since updating to 2.2.0. The following code was working on 1.4.3.

Is there anything I've missed or something you are aware of that could be causing this?

#include "BLEGamepad.h"

#define MOTUS_EXPLORE_SERVICE_UUID "9c9d4aa7-8050-4c8a-bc67-146a66d0443e"
#define INTENSITY_UUID "9c9d4aa7-8051-4c8a-bc67-146a66d0443e"
#define ENDURANCE_UUID "9c9d4aa7-8052-4c8a-bc67-146a66d0443e"
#define LEFT_BALANCE_UUID "9c9d4aa7-8053-4c8a-bc67-146a66d0443e"
#define RIGHT_BALANCE_UUID "9c9d4aa7-8054-4c8a-bc67-146a66d0443e"
#define MAX_EXTENSION_LEFT_UUID "9c9d4aa7-8055-4c8a-bc67-146a66d0443e"
#define MAX_EXTENSION_RIGHT_UUID "9c9d4aa7-8056-4c8a-bc67-146a66d0443e"
#define RECENTRE_UUID "9c9d4aa7-8057-4c8a-bc67-146a66d0443e"
#define CCC_DESCRIPTOR_UUID "2902"

static const char* TAG = "MotusVR";
static NimBLEHIDDevice* hid;
static NimBLECharacteristic* pInput = NULL;
static NimBLECharacteristic* pOutput = NULL;

static NimBLEServer* pServer;
static NimBLEService* pService;

static NimBLEService* pMotusService = NULL;
static NimBLECharacteristic* pIntensity = NULL;
static NimBLECharacteristic* pEndurance = NULL;
static NimBLECharacteristic* pLeftBalance = NULL;
static NimBLECharacteristic* pRightBalance = NULL;
static NimBLECharacteristic* pMaxExtenstionLeft = NULL;
static NimBLECharacteristic* pMaxExtenstionRight = NULL;
static NimBLECharacteristic* pRecentre = NULL;

uint8_t pIntensityValue = 100;
uint8_t pEnduranceValue = 75;
uint8_t pLeftBalanceValue = 50;
uint8_t pRightBalanceValue = 50;
uint8_t pMaxExtenstionLeftValue = 25;
uint8_t pMaxExtenstionRightValue = 0;
bool pRecentreValue = false;  // false represents not recentred, true recentred

uint8_t reportMapGamepad[] = {
  0x05, 0x01,  // USAGE_PAGE (Generic Desktop)
  0x09, 0x05,  // USAGE (Joystick - 0x04; Gamepad - 0x05; Multi-axis Controller - 0x08)
  0xA1, 0x01,  // COLLECTION (Application)
  0x85, 0x03,  // REPORT_ID (Default: 3)

  // Buttons (1 to 16)
  0x05, 0x09,  // USAGE_PAGE (Button)
  0x15, 0x00,  // LOGICAL_MINIMUM (0)
  0x25, 0x01,  // LOGICAL_MAXIMUM (1)
  0x75, 0x01,  // REPORT_SIZE (1)
  0x19, 0x01,  // USAGE_MINIMUM (Button 1)
  0x29, 0x10,  // USAGE_MAXIMUM (Up to 128 buttons possible)
  0x95, 0x10,  // REPORT_COUNT (# of buttons)
  0x81, 0x02,  // INPUT (Data,Var,Abs)

  // Padding to align to the next byte
  0x5, 0x01,   // REPORT_SIZE (1)
  0x09, 0x01,  // REPORT_COUNT (# of padding bits)
  0x16, 0x00,  // INPUT (Const,Var,Abs)

  // Analog Axes (X and Y)
  0x00, 0x26,  // LOGICAL_MINIMUM
  0xFF, 0x7F,  // LOGICAL_MAXIMUM - no error but not working on 0xFF
  0x75, 0x10,  // REPORT_SIZE (16) but should be 8
  0x95, 0x08,  // REPORT_COUNT
  0xA1, 0x00,  // COLLECTION (Physical)
[... 251 lines cut ...]

void BLEGamepad::updateMaxExtenstionLeftCharacteristic(uint8_t input) {
  pMaxExtenstionLeft->setValue((uint8_t*)&input, sizeof(input));
  pMaxExtenstionLeft->notify((uint8_t*)&input, sizeof(input));
}

void BLEGamepad::updateMaxExtenstionRightCharacteristic(uint8_t input) {
  pMaxExtenstionRight->setValue((uint8_t*)&input, sizeof(input));
  pMaxExtenstionRight->notify((uint8_t*)&input, sizeof(input));
}

BLEGamepad::BLEGamepad() {
  Serial.println("Initalising Gamepad BLE");
  NimBLEDevice::init("Test");
  
  NimBLEDevice::setSecurityAuth(true,false,true);

  NimBLEAddress addr = NimBLEDevice::getAddress();

  Serial.print("MAC Address: ");
  Serial.println(addr.toString().c_str());
  NimBLEDevice::setDeviceName(std::string("Test Device " + addr.toString().substr(9)));

  this->taskServer();
}

h2zero, 2025-01-16:
One thing I see is `pIntensity->notify((uint8_t*)&input, sizeof(input), true);` and `pEndurance->notify((uint8_t*)&input, sizeof(input), true);` are incorrect.
https://h2zero.github.io/NimBLE-Arduino/md_1_8x__to2_8x__migration__guide.html#characteristics

Also the function signatures in your characteristic callbacks are also incorrect.

thomasMEBaker, 2025-01-16:
Thanks - that's sorted it! Had missed that on the migration guide!

closed 2025-01-16 by thomasMEBaker