h2zero/NimBLE-Arduino #865: cannot upload extended example to esp c3

mihaigabriel620, 2025-01-17:
when i try to upload the example code for extended server it gives this error ( reinstalled everything ) ( and enabled phy in the h file )

Arduino: 1.8.19 (Windows 10), Board: "ESP32C3 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 160MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None, Disabled, Disabled"

C:\Users\mihai\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\nimble\host\src\ble_hs_periodic_sync.c: In function 'ble_hs_periodic_sync_free':

C:\Users\mihai\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\nimble\host\src\ble_hs_periodic_sync.c:58:24: error: 'struct ble_npl_event' has no member named 'event'

   58 |     if((psync->lost_ev).event != NULL)

      |                        ^

exit status 1

Error compiling for board ESP32C3 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

h2zero, 2025-01-17:
Sorry I forgot I was going to remove the option to enable periodic advertising, it's not supported yet. Just comment out that line again and it should be fine.

mihaigabriel620, 2025-01-17:
also i cant get any example code to work even nimble_server is broken (exit status 1
no matching function for call to 'NimBLEDevice::startAdvertising()'
 )

mihaigabriel620, 2025-01-17:
maybe how you say ( periodic adverising issue for this one also? )

mihaigabriel620, 2025-01-17:
and also i wanna ask if you have the time can you make a example sketch where the esp is in phy coded mode ( 125kbps one ) and you can pair with a phone and stores the mac ( using the irc,etc ) to decrypt the max and if the connection is lost it attempts to reconnect to it ( same logic like bluetooth headphones ) 

( also i'm willing to pay you ;) if you can make it )

h2zero, 2025-01-17:
>also i cant get any example code to work even nimble_server is broken (exit status 1
no matching function for call to 'NimBLEDevice::startAdvertising()'
)

You need to use the BT5 examples with extended advertising.
as long as this https://github.com/h2zero/NimBLE-Arduino/blob/3081bc1bb8db1ea6c67a1e06da63a2c4c833e4fb/src/nimconfig.h#L50 is disabled and you are using one of the examples from [here](https://github.com/h2zero/NimBLE-Arduino/tree/master/examples/Bluetooth_5) it will work.

>and also i wanna ask if you have the time can you make a example sketch where the esp is in phy coded mode ( the 125kbps one ) and you can pair with a phone and stores the mac ( using the irc,etc ) to decrypt the mac and if the connection is lost it attempts to reconnect to it ( same logic like bluetooth headphones )

This already exists to a large extent in the above examples, just enable bonding by setting the first parameter in `NimBLEDevice::setSecurityAuth` to true, such as `NimBLEDevice::setSecurityAuth(true, false, true);`, after initialization.

h2zero, 2025-04-23:
Closing this as stale, please re-open if still an issue.

closed 2025-04-23 by h2zero