h2zero/NimBLE-Arduino #868: how to send many notify messages when Characteristic's onwrite run?

ham0nd, 2025-01-19:
hi , @h2zero ,  i found a question that when Characteristic's onwrite run , it will only send once notify message , but i want to send many times notify messages(i want to use this way to send file binary value with some packets) , so i want to know is it possible to send many notify messages when Characteristic's onwrite run ? thanks 

here is my code , i test only get "end" from client

`
pDeadService->createCharacteristic(UUID,NIMBLE_PROPERTY::NOTIFY | NIMBLE_PROPERTY::WRITE);
class CharacteristicCallbacks : public NimBLECharacteristicCallbacks
{
  void sendMsg(BLECharacteristic *pCharacteristic, const char *data)
  {
    pCharacteristic->setValue(data);
    pCharacteristic->notify();
  }
  void onWrite(NimBLECharacteristic *pCharacteristic, NimBLEConnInfo &connInfo) override
  {
    sendMsg(pCharacteristic,"aaa");
    sendMsg(pCharacteristic,"bbb");
    sendMsg(pCharacteristic,"end");
  }
}

`

h2zero, 2025-01-19:
You cannot send notifications like this as it will quickly use up all the buffers. You should use the onStatus callback to know when to send the next notification

ham0nd, 2025-01-24:
if onStatus can send many times message by notify?

here is my code
```
pDeadService->createCharacteristic(UUID,NIMBLE_PROPERTY::NOTIFY | NIMBLE_PROPERTY::WRITE);
class CharacteristicCallbacks : public NimBLECharacteristicCallbacks
{
  void sendMsg(BLECharacteristic *pCharacteristic, const char *data)
  {
    pCharacteristic->setValue(data);
    pCharacteristic->notify();
  }
  void onWrite(NimBLECharacteristic *pCharacteristic, NimBLEConnInfo &connInfo) override
  {
    const uint8_t *data = pCharacteristic->getValue();
    if(data[0] == 0x01) // start reading file code
    {
      p_file = fs.open(file,'r');
      sendMsg(pCharacteristic,"p_file init success");
    }
    
  }

  void onStatus(NimBLECharacteristic *pCharacteristic, int code) override
  {
    if (code == 0) 
    {
      for(byte packet = 0;packet < fileInfo.totalPackets;packet ++)
      {
        handle_READ_FILE_POINT_READING(p_file);
        sendMsg(pCharacteristic, FileData, FileDataLength);
      }
    }
  }
}
```

h2zero, 2025-01-24:
You should only send one notification from the callback.

closed 2025-01-25 by ham0nd