h2zero/NimBLE-Arduino #870: how to see mac of device

mihaigabriel620, 2025-01-19:
Hey im new here and i want to know how to see the device mac connected to the esp ( talking about phone here ) but not the mac that changes i mean the decrypted id that uses the irk or something else that can identify the phone from another one maybe its not mac i dunno im new

h2zero, 2025-01-20:
The only way to get the static device ID address is to bond with it, then you can get the address from the onIdentity callaback.

mihaigabriel620, 2025-01-20:
can you give a code snippet example on how to implement once bonded i can see the mac on the serial monitor?

mihaigabriel620, 2025-01-20:
and once bonded will it be the true mac of the device? and not the randoom one?

h2zero, 2025-01-20:
>NimBLEDevice::setSecurityAuth(true, false, true); // first param enables bonding.

 Please see the secure server example for more info.

Then implement the callback:
https://github.com/h2zero/NimBLE-Arduino/blob/3081bc1bb8db1ea6c67a1e06da63a2c4c833e4fb/src/NimBLEServer.cpp#L1010

The identity address is in `connInfo->getIdAddress()`

mihaigabriel620, 2025-01-20:
Funny you say that i tried exactly that but the resulting mac is still the random one

h2zero, 2025-01-20:
Works for me?

[ble_client_identity.txt](https://github.com/user-attachments/files/18479800/ble_client_identity.txt)

h2zero, 2025-04-23:
Sorry for the delay @mihaigabriel620, the only way to get the identity address from a phone is to bond with it. This address will become available when the `onIdentity` callback is invoked. 

Since this not a bug and a couple months old I will close it now, please re-open if still having this issue.

closed 2025-04-23 by h2zero