h2zero/NimBLE-Arduino #872: Random issue on NimBLE v2.2.0

moiatto, 2025-01-21:
- Arduino IDE v2.3.4 (Mac)
- Arduino core for ESP32 v3.1.1
- board ESP32 Dev module

The problem appears randomly and causes the ESP32 to restart.
It happens after a few seconds or after several minutes.

```
abort() was called at PC 0x40181073 on core 0

Backtrace: 0x40082629:0x3ffd4130 0x4009632d:0x3ffd4150 0x4009c549:0x3ffd4170 0x40181073:0x3ffd41f0 0x401810a8:0x3ffd4210 0x4018189e:0x3ffd4230 0x40181209:0x3ffd4250 0x40181505:0x3ffd4270 0x40181539:0x3ffd4290 0x400e0f72:0x3ffd42b0 0x400e0fb1:0x3ffd4310 0x400d800a:0x3ffd4330 0x400e26e5:0x3ffd4390 0x400e6f1e:0x3ffd43c0 0x400ebac1:0x3ffd4430 0x400eb87f:0x3ffd4470 0x400ebd2d:0x3ffd4490 0x400ea981:0x3ffd44b0 0x4008142d:0x3ffd44d0 0x400e1eff:0x3ffd44f0 0x40096ffa:0x3ffd4510
```

Decoding stack results:
```
0x40082629: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:463
0x4009632d: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x4009c549: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:38
0x40181073: std::get_terminate() at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:97
0x401810a8: std::get_unexpected() at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:127
0x4018189e: __cxxabiv1::__cxa_free_exception(void*) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_alloc.cc:423
0x40181209:  is in operator new(unsigned int) (/Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:50).
0x40181505:  is in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned int&, unsigned int) (/Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/build/build-cc-gcc-final/xtensa-esp-elf/esp32/no-rtti/libstdc++-v3/include/bits/basic_string.tcc:140).
0x40181539:  is in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_dispose() (/Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/build/build-cc-gcc-final/xtensa-esp-elf/esp32/no-rtti/libstdc++-v3/include/bits/basic_string.h:282).
0x400e0f72:  is in NimBLEAddress::operator std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >() const (/Users/Andrea/Library/Arduino15/packages/esp32/tools/esp-x32/2405/xtensa-esp-elf/include/c++/13.2.0/bits/char_traits.h:368).
0x400e0fb1:  is in std::_Vector_base<unsigned char, std::allocator<unsigned char> >::_M_allocate(unsigned int) (/Users/Andrea/Library/Arduino15/packages/esp32/tools/esp-x32/2405/xtensa-esp-elf/include/c++/13.2.0/bits/new_allocator.h:147).
0x400d800a:  is in MyAdvertisedDeviceCallbacks::onResult(NimBLEAdvertisedDevice const*) (/Users/Andrea/Library/Arduino15/packages/esp32/tools/esp-x32/2405/xtensa-esp-elf/include/c++/13.2.0/bits/basic_string.h:222).
0x400e26e5:  is in NimBLEScan::handleGapEvent(ble_gap_event*, void*) (/Users/Andrea/Documents/SpeedoX/Arduino/libraries/NimBLE-Arduino/src/NimBLEScan.cpp:139).
0x400e6f1e: ble_gap_rx_conn_complete at /Users/Andrea/Documents/SpeedoX/Arduino/libraries/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:2147
0x400ebac1: ble_hs_hci_evt_disconn_complete at /Users/Andrea/Documents/SpeedoX/Arduino/libraries/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_evt.c:218
0x400ebd2d: ble_hs_hci_evt_acl_process at /Users/Andrea/Documents/SpeedoX/Arduino/libraries/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_evt.c:1134
0x400ea981: ble_hs_timer_reset at /Users/Andrea/Documents/SpeedoX/Arduino/libraries/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:454
0x4008142d: nimble_port_run at /Users/Andrea/Documents/SpeedoX/Arduino/libraries/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:530
0x400e1eff: NimBLEDevice::onSync() at /Users/Andrea/Documents/SpeedoX/Arduino/libraries/NimBLE-Arduino/src/NimBLEDevice.cpp:791
0x40096ffa: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

h2zero, 2025-01-21:
Can you share the code in your scan callback?

moiatto, 2025-01-21:
Hi @h2zero,

yes, this:

```
class MyAdvertisedDeviceCallbacks: public NimBLEScanCallbacks 
{
    /** Initial discovery, advertisement data only. */
    void onDiscovered(const NimBLEAdvertisedDevice* advertisedDevice) override 
    {
        // printf("Discovered Device: %s\n", advertisedDevice->toString().c_str());
    }

    /**
     *  If active scanning the result here will have the scan response data.
     *  If not active scanning then this will be the same as onDiscovered.
     */
    void onResult(const NimBLEAdvertisedDevice* advertisedDevice) override
    {  
        BLEAddress *pDeviceAddress = new NimBLEAddress(advertisedDevice->getAddress());

        String deviceAddress = String(pDeviceAddress->toString().c_str());
        deviceAddress.toUpperCase();

        if (deviceAddress.equals(tpms[0].address)) 
          { 
            TPMS_SENSOR_MASK_DATA tpmsMaskData = advertisedDevice->getManufacturerData<TPMS_SENSOR_MASK_DATA>(true);
            decodeTpmsManufacturerData(0, tpmsMaskData);
          }

        if (deviceAddress.equals(tpms[1].address)) 
          { 
            TPMS_SENSOR_MASK_DATA tpmsMaskData = advertisedDevice->getManufacturerData<TPMS_SENSOR_MASK_DATA>(true);
            decodeTpmsManufacturerData(1, tpmsMaskData);
          }
    }

    void onScanEnd(const NimBLEScanResults& results, int reason) override
    {
        // printf("Scan ended reason = %d; restarting scan\n", reason);
        vTaskDelay(10 / portTICK_PERIOD_MS);
        NimBLEDevice::getScan()->start(BleScanTimeMs, false, true);
    }
};
```

moiatto, 2025-01-25:
Hi @h2zero 
can i try to change something? suggestions?

thekurtovic, 2025-01-25:
@moiatto I tried partially testing your code but couldn't reproduce it.
Have you tried testing with a simplified `onResult` callback, does this crash?
```
void onResult(const NimBLEAdvertisedDevice* advertisedDevice) override
{  
    String deviceAddress = String(advertisedDevice->getAddress().toString().c_str());
    deviceAddress.toUpperCase();
    Serial.printf("found %s\n", advertisedDevice->getAddress().toString().c_str());
}
```

moiatto, 2025-02-01:
modified as you suggested...seems to work 👍

Thanks

h2zero, 2025-02-01:
Great 👍, please close the issue if it is resolved.

closed 2025-02-01 by moiatto