h2zero/NimBLE-Arduino #873: weird issue with blegamepad and esp32wiimote

5ghzx, 2025-01-21:
have this code here
```
#include <Arduino.h>
#include <BleGamepad.h>
#include "ESP32Wiimote.h"
#include "NimBLEDevice.h"

ESP32Wiimote wiimote;
BleGamepad Gamepad;

#define numOfButtons 13
#define numOfHatSwitches 0
#define enableX false
#define enableY false
#define enableZ false
#define enableRX false
#define enableRY false
#define enableRZ false
#define enableSlider1 false
#define enableSlider2 false
#define enableRudder false
#define enableThrottle false
#define enableAccelerator false
#define enableBrake false
#define enableSteering false

static bool logging = true;
static long last_ms = 0;
static int num_run = 0, num_updates = 0;

void setup()
{
    Serial.begin(115200);
    Serial.println("ESP32Wiimote");

    BleGamepadConfiguration BleGamepadConfig;
    BleGamepadConfig.setAutoReport(false);
    BleGamepadConfig.setButtonCount(numOfButtons);
    Gamepad.begin();

    wiimote.init();
    if (! logging)
        wiimote.addFilter(ACTION_IGNORE, FILTER_ACCEL); // optional
    
    Serial.println("Started");
    last_ms = millis();

}

void loop()
{
    wiimote.task();
    num_run++;

    if (wiimote.available() > 0) 
    {
        ButtonState  button  = wiimote.getButtonState();
        AccelState   accel   = wiimote.getAccelState();
        NunchukState nunchuk = wiimote.getNunchukState();

        num_updates++;
        if (logging)
        {
            bool isA     = (button & BUTTON_A)     != 0;
            bool isB     = (button & BUTTON_B)     != 0;
            bool isC     = (button & BUTTON_C)     != 0;
            bool isZ     = (button & BUTTON_Z)     != 0;
            bool isOne   = (button & BUTTON_ONE)   != 0;
            bool isTwo   = (button & BUTTON_TWO)   != 0;
            bool isMinus = (button & BUTTON_MINUS) != 0;
            bool isPlus  = (button & BUTTON_PLUS)  != 0;
            bool isHome  = (button & BUTTON_HOME)  != 0;
            bool isLeft  = (button & BUTTON_LEFT)  != 0;
            bool isRight = (button & BUTTON_RIGHT) != 0;
            bool isUp    = (button & BUTTON_UP)    != 0;
            bool isDown  = (button & BUTTON_DOWN)  != 0;

            char ca     = (button & BUTTON_A)     ? 'A' : '.';
            char cb     = (button & BUTTON_B)     ? 'B' : '.';
            char cc     = (button & BUTTON_C)     ? 'C' : '.';
            char cz     = (button & BUTTON_Z)     ? 'Z' : '.';
            char c1     = (button & BUTTON_ONE)   ? '1' : '.';
            char c2     = (button & BUTTON_TWO)   ? '2' : '.';
            char cminus = (button & BUTTON_MINUS) ? '-' : '.';
            char cplus  = (button & BUTTON_PLUS)  ? '+' : '.';
            char chome  = (button & BUTTON_HOME)  ? 'H' : '.';
            char cleft  = (button & BUTTON_LEFT)  ? '<' : '.';
            char cright = (button & BUTTON_RIGHT) ? '>' : '.';
            char cup    = (button & BUTTON_UP)    ? '^' : '.';
            char cdown  = (button & BUTTON_DOWN)  ? 'v' : '.';

            Serial.printf("button: %05x = ", (int)button);
            Serial.print(ca);
            Serial.print(cb);
            Serial.print(cc);
            Serial.print(cz);
            Serial.print(c1);
            Serial.print(c2);
            Serial.print(cminus);
            Serial.print(chome);
            Serial.print(cplus);
            Serial.print(cleft);
            Serial.print(cright);
            Serial.print(cup);
            Serial.print(cdown);
            Serial.printf(", wiimote.axis: %3d/%3d/%3d", accel.xAxis, accel.yAxis, accel.zAxis);
            Serial.printf(", nunchuk.axis: %3d/%3d/%3d", nunchuk.xAxis, nunchuk.yAxis, nunchuk.zAxis);
            Serial.printf(", nunchuk.stick: %3d/%3d\n", nunchuk.xStick, nunchuk.yStick);

            if (Gamepad.isConnected())
            {
              if (isA) { Gamepad.press(1); } else { Gamepad.release(1); }
              if (isB) { Gamepad.press(2); } else { Gamepad.release(2); }
              if (isC) { Gamepad.press(3); } else { Gamepad.release(3); }
              if (isZ) { Gamepad.press(4); } else { Gamepad.release(4); }
              if (isOne) { Gamepad.press(5); } else { Gamepad.release(5); }
              if (isTwo) { Gamepad.press(6); } else { Gamepad.release(6); }
              if (isMinus) { Gamepad.press(7); } else { Gamepad.release(7); }
              if (isPlus) { Gamepad.press(8); } else { Gamepad.release(8); }
              if (isHome) { Gamepad.press(9); } else { Gamepad.release(9); }
              if (isUp) { Gamepad.press(10); } else { Gamepad.release(10); }
              if (isDown) { Gamepad.press(11); } else { Gamepad.release(11); }
              if (isLeft) { Gamepad.press(12); } else { Gamepad.release(12); }
              if (isRight) { Gamepad.press(13); } else { Gamepad.release(13); }
              Gamepad.sendReport();
            }
            delay(10);
        }
    }

    if (! logging)
    {
        long ms = millis();
        if (ms - last_ms >= 1000)
        {
            Serial.printf("Run %d times per second with %d updates\n", num_run, num_updates);
            num_run = num_updates = 0;
            last_ms += 1000;
        }
    }
}
```
an esp32-wroom-32d, this esp32 to be exact [https://community.microcenter.com/kb/articles/652-inland-esp32-core-board-black-and-eco-friendly](url), im getting this error
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4916
load:0x40078000,len:16436
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3524
entry 0x400805b8
ESP32Wiimote
E (21) BTDM_INIT: esp_bt_controller_init -3

btStart() failed
Started

assert failed: npl_freertos_mutex_pend C:\Users\hadia\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\porting\npl\freertos\src\npl_os_freertos.c:1619 (mu->handle)

Backtrace: 0x400824b5:0x3ffcf370 0x40093719:0x3ffcf390 0x4009930a:0x3ffcf3b0 0x400e4655:0x3ffcf4e0 0x400dc964:0x3ffcf500 0x400dc97b:0x3ffcf520 0x400dc733:0x3ffcf540 0x400d42ff:0x3ffcf560 0x400d21c8:0x3ffcf580 0x4009413e:0x3ffcf5a0

ELF file SHA256: 164c9fc9f

Rebooting...

5ghzx, 2025-01-21:
latest version of nimble, latest version of esp32 blegamepad by lemmingdev. esp32 wiimote works perfectly when i try it without this code.

gopro2027, 2025-01-25:
Just realized this seems very related to the issue I just wrote up https://github.com/h2zero/NimBLE-Arduino/issues/876

5ghzx, 2025-01-25:
oh ya, it is. maybe its something about max connections?

Zanacks1, 2025-02-08:
mine wont even run, still in setup..
{
    Serial.begin(115200);
    Serial.println("Starting BLE work!");
    bleGamepad.begin();
    // The default bleGamepad.begin() above enables 16 buttons, all axes, one hat, and no simulation controls or special buttons
}

Something silly on my part? ESP32-WROOM-DA

Bluepad32 (C) 2016-2024 Ricardo Quesada and contributors.
Version: v4.1.0
BTstack: Copyright (C) 2017 BlueKitchen GmbH.
BTstack up and running at 24:0A:C4:0F:73:96
Starting BLE work!

assert failed: npl_freertos_mutex_pend c:\Users\xxxxx\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\porting\npl\freertos\src\npl_os_freertos.c:1619 (mu->handle)

Backtrace: 0x40083755:0x3ffe9250 0x40093be5:0x3ffe9270 0x40098fb5:0x3ffe9290 0x400e5f11:0x3ffe93c0 0x400de0a4:0x3ffe93e0 0x400de0bb:0x3ffe9400 0x400dde77:0x3ffe9420 0x400d54f7:0x3ffe9440 0x400d40a4:0x3ffe9460

ELF file SHA256: 9c3d779a54126c82

Decoding stack results
0x40083755: panic_abort at /home/riq/bin/esp/esp-idf-v4.4/components/esp_system\panic.c:408
0x40093be5: esp_system_abort at /home/riq/bin/esp/esp-idf-v4.4/components/esp_system\esp_system.c:137
0x40098fb5: __assert_func at /home/riq/bin/esp/esp-idf-v4.4/components/newlib\assert.c:85
0x400e5f11: npl_freertos_mutex_pend at c:\Users\xxxxx\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\porting\npl\freertos\src\npl_os_freertos.c:1623
0x400de0a4: ble_hs_lock_nested at c:\Users\xxxxx\Documents\Arduino\libraries\NimBLE-Arduino\src/nimble/porting/npl/freertos/include/nimble\nimble_npl_os.h:587
0x400de0bb: ble_hs_lock at c:\Users\xxxxx\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\nimble\host\src\ble_hs.c:225
0x400dde77: ble_gatts_reset at c:\Users\xxxxx\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\nimble\host\src\ble_gatts.c:3208
0x400d54f7: NimBLEDevice::createServer() at c:\Users\xxxxx\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:125
0x400d40a4: BleGamepad::taskServer(void*) at c:\Users\xxxxx\Documents\Arduino\libraries\ESP32-BLE-Gamepad\BleGamepad.cpp:1917

h2zero, 2025-02-08:
Something in your code either has not initialized the NimBLE stack or has deinitialized it.