h2zero/NimBLE-Arduino #875: Are HIDTypes.h listed correctly???

RabByte-0, 2025-01-24:
`#define USAGE_PAGE(size)        (0x04 | size)`

shouldn't it be 0x05?

```
/* Local items */
#define USAGE(size)                 (0x08 | size)
```

shouldn't that be 0x09?

I'm very new to all this and i'm worried this is just user error to when i put the input report into [USB Descriptor](https://eleccelerator.com/usbdescreqparser/)  it donsn't compile correctly. adding 1 to many of the sections seems to fix...  the report

h2zero, 2025-01-26:
Great question, I have no idea sadly, not my expertise but @afpineda might be able to answer?

RabByte-0, 2025-01-27:
I really appreciate your reply! Thanks for setting me straight

h2zero, 2025-01-27:
Thanks again @afpineda, I had no idea about this, appreciate your help very much.
@RabByte-0 If the resolves your issue could you please close this?

RabByte-0, 2025-01-27:
Absolutely! This is literally my second comment in GitHub and I'm still unfamiliar with etiquette.

closed 2025-01-27 by RabByte-0

h2zero, 2025-01-27:
All good, thanks and welcome!