h2zero/NimBLE-Arduino #876: Conflict with jvpernis/PS3 Controller Host

gopro2027, 2025-01-25:
The only reason i raise this issue is that the built in espressif BLE has no issue running with it, but NimBLE does. 
Brief description of jvpernis/PS3 Controller Host: It is a bluetooth library to connect the ps3 controller which works on some obscure bluetooth protocol. idk the specifics, but it's different enough to not interfere with and work along side both built in espressif libraries, legacy serial bluetooth and ble, without issue.
However when using NimBLE, nimble will fail very soon on `NimBLEDevice::createServer();` 
Disabling my ps3 controller code causes the nimble code to boot up and work just fine.
Also i tried waiting until after Nimble was initialized and it seems to cause some issues there too (prevents connection, no crash though). 
Luckily this is not a pressing issue for me, i can revert back to the built in ble for this portion of the project, but surely this would be worth fixing if not at least investigating as to why it is causing issues.
Crash log:
```
assert failed: npl_freertos_mutex_pend npl_os_freertos.c:1619 (mu->handle)

Backtrace: 0x40083b75:0x3ffbbfa0 0x40094521:0x3ffbbfc0 0x40099d25:0x3ffbbfe0 0x400e7219:0x3ffbc110 0x400df1a8:0x3ffbc130 0x400df1bf:0x3ffbc150 0x400def7b:0x3ffbc170 0x400d7e6f:0x3ffbc190 0x400d35ef:0x3ffbc1b0 0x400d505e:0x3ffbc1f0
  #0  0x40083b75 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
  #1  0x40094521 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
  #2  0x40099d25 in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85
  #3  0x400e7219 in npl_freertos_mutex_pend at .pio/libdeps/upesy_wroom/NimBLE-Arduino/src/nimble/porting/npl/freertos/src/npl_os_freertos.c:1623
  #4  0x400df1a8 in ble_npl_mutex_pend at .pio/libdeps/upesy_wroom/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:587
      (inlined by) ble_hs_lock_nested at .pio/libdeps/upesy_wroom/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:178
  #5  0x400df1bf in ble_hs_lock at .pio/libdeps/upesy_wroom/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:225
  #6  0x400def7b in ble_gatts_reset at .pio/libdeps/upesy_wroom/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gatts.c:3208
  #7  0x400d7e6f in NimBLEDevice::createServer() at .pio/libdeps/upesy_wroom/NimBLE-Arduino/src/NimBLEDevice.cpp:125
  #8  0x400d35ef in ble_setup() at src/bluetooth/ble.cpp:71
  #9  0x400d505e in task_bluetooth(void*) at src/tasks/tasks.cpp:27
```

5ghzx, 2025-01-25:
Say, might this issue be caused by the numbers of bt devices that an esp32 can be connected to? i google beforehand and it says 4, but if that isn't true, this might be happening.

h2zero, 2025-01-26:
The backtrace suggests that NimBLE has not be initialized, asserted on mutex pend which was not initialized.

h2zero, 2025-04-23:
Closing this as stale as there has not been any further communication.

closed 2025-04-23 by h2zero