h2zero/NimBLE-Arduino #879: ESP32-C3 not receiving data with ESP-NOW after initialising NimBLE

Gabriele-Cano, 2025-01-27:
Hi, 

I'm using two ESP32-C3, the first one to turn on is the one who becomes the master after 3 seconds. The master should initialise NimBLE and then broadcast every second or so a packet called ACK for other ESP32s with esp now. The problem is that when the slave device receives the packet it tries to reply back to the master but the attempt fails. I know for sure that this kind of communication works because I tried it without initialising NimBLE. 

(Note: after initialising NimBLE the master ESP is correctly seen by the phone)
(I already tried to ask on Arduino-ESP32 core but didn't replied and I need to finish this code before friday)

I can't figure out what's the problem. 

This is the code 

```cpp
#include <NimBLEDevice.h>
#include "ESP32_NOW.h"
#include "WiFi.h"
#include "structs.cpp"

#include <esp_mac.h>  // For the MAC2STR and MACSTR macros

/* Definitions */

#define ESPNOW_WIFI_CHANNEL 6
#define BATTERY_PIN 3

/* Global Variables */

uint32_t msg_count = 0;
bool inGroup = false;
bool isMaster = false;
long startupTime, lastBatteryCheckSent, lastACKSent, lastAliveCheck;
bool connectedBLE = false;

/* Functions */ 

int getCurrentBatterySOT() {
  int batteria = map(analogReadMilliVolts(BATTERY_PIN) * 2, 3150, 4150, 0, 100);
  batteria = max(1, min(batteria, 100));
  return batteria;
}

/* Classes */

// Creating a new class that inherits from the ESP_NOW_Peer class is required.
class ESP_NOW_Broadcast_Peer : public ESP_NOW_Peer {
public:
  // Constructor of the class using the broadcast address
  ESP_NOW_Broadcast_Peer(uint8_t channel, wifi_interface_t iface, const uint8_t *lmk) : ESP_NOW_Peer(ESP_NOW.BROADCAST_ADDR, channel, iface, lmk) {}

  // Destructor of the class
  ~ESP_NOW_Broadcast_Peer() {
    remove();
  }

  // Function to properly initialize the ESP-NOW and register the broadcast peer
  bool begin() {
    if (!ESP_NOW.begin() || !add()) {
      log_e("Failed to initialize ESP-NOW or register the broadcast peer");
      return false;
    }
    return true;
  }

  // Function to send a message to all devices within the network
  bool send_message(const uint8_t *data, size_t len) {
    if (!send(data, len)) {
      log_e("Failed to broadcast message");
      return false;
    }
    return true;
  }
[... 227 lines cut ...]
    broadcast_peer.send_message((uint8_t *)&pack, sizeof(ACK));
    lastACKSent = millis();
  }

  if (isMaster && lastBatteryCheckSent + 2000 < millis()) {
    BATTERY pack = BATTERY(255);
    for (int i = devices.size() - 1; i >= 0; i--) {
      devices[i].send_message((uint8_t *)&pack, sizeof(BATTERY));
    }
    lastBatteryCheckSent = millis();
  }

  if (isMaster && lastAliveCheck + 1000 < millis()) {
    for (int i = devices.size() - 1; i >= 0; i--) {
      if (!devices[i].isAlive) {
        devices[i].deinitialise();
        devices.erase(devices.begin() + i);
      }
    }
    lastAliveCheck = millis();
  }

  delay(100);
}
```

h2zero, 2025-01-27:
Try initializing BLE at startup, but not advertising until you need to, for some reason it seem to help sometimes.

Other than that I would need to see a debug log to start with.

Gabriele-Cano, 2025-01-27:
Thank you for your answer

Ok, I put initBLE in the setup function just after Serial.begin, then moved     NimBLEDevice::startAdvertising(); inside the loop right after IM THE MASTER println. But still there is the same problem.

These are the logs: 
# Master 
```
============ Before Setup End ============
[   484][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 18 already has type USB_DM (38) with bus 0x3fc98ed0
[   485][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 19 already has type USB_DP (39) with bus 0x3fc98ed0
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
D NimBLEService: >> start(): Starting service: UUID: 0x180f, handle: 0x0000
D NimBLEService: Adding 1 characteristics for service UUID: 0x180f, handle: 0x0000
D NimBLEService: << start()
Finished BLE init
[   577][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 101 - WIFI_READY
[   581][V][STA.cpp:186] _onStaEvent(): STA Started
[   581][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 110 - STA_START
[   582][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 110 - STA_START
[   584][V][ESP32_NOW.cpp:18] _esp_now_add_peer(): ff:ff:ff:ff:ff:ff
[   585][V][ESP32_NOW.cpp:311] add(): Peer added - ff:ff:ff:ff:ff:ff
=========== After Setup Start ============

IM THE MASTER
D NimBLEAdvertising: >> Advertising start: duration=0, dirAddr=NULL
primary service
           uuid 0x1800
         handle 1
     end_handle 5
characteristic
           uuid 0x2a00
     def_handle 2
     val_handle 3
   min_key_size 0
          flags [READ]
characteristic
           uuid 0x2a01
     def_handle 4
     val_handle 5
   min_key_size 0
          flags [READ]
primary service
           uuid 0x1801
         handle 6
     end_handle 13
characteristic
           uuid 0x2a05
     def_handle 7
     val_handle 8
   min_key_size 0
          flags [INDICATE]
ccc descriptor
           uuid 0x2902
         handle 9
   min_key_size 0
          flags [READ|WRITE]
characteristic
           uuid 0x2b3a
     def_handle 10
     val_handle 11
   min_key_size 0
          flags [READ]
characteristic
           uuid 0x2b29
     def_handle 12
     val_handle 13
   min_key_size 0
          flags [READ|WRITE]
primary service
[... 60 lines cut ...]
00:00:27.756 ============ After Setup End =============
00:00:28.247 [  2850][V][ESP32_NOW.cpp:112] _esp_now_rx_cb(): Broadcast from d8:3b:da:e3:1c:cc, data length : 1
00:00:28.248 0x00, // .
00:00:28.248 [  2850][V][ESP32_NOW.cpp:115] _esp_now_rx_cb(): Calling new_cb, peer not found.
00:00:28.248 Unknown peer d8:3b:da:e3:1c:cc sent a broadcast message
00:00:28.248 Registering the peer as a master
00:00:28.248 [  2853][V][ESP32_NOW.cpp:18] _esp_now_add_peer(): d8:3b:da:e3:1c:cc
00:00:28.248 [  2856][V][ESP32_NOW.cpp:311] add(): Peer added - d8:3b:da:e3:1c:cc
00:00:28.248 [  2856][V][ESP32_NOW.cpp:384] send(): d8:3b:da:e3:1c:cc, data length 1
00:00:28.288 [  2894][V][ESP32_NOW.cpp:133] _esp_now_tx_cb(): d8:3b:da:e3:1c:cc : FAILED
00:00:28.289 [  2895][I][ESP32_NOW.h:78] onSent(): Message transmission to peer d8:3b:da:e3:1c:cc failed
00:00:29.245 [  3850][V][ESP32_NOW.cpp:112] _esp_now_rx_cb(): Broadcast from d8:3b:da:e3:1c:cc, data length : 1
00:00:29.245 0x00, // .
00:00:29.245 [  3851][V][ESP32_NOW.cpp:122] _esp_now_rx_cb(): Checking peer ff:ff:ff:ff:ff:ff
00:00:29.246 [  3851][V][ESP32_NOW.cpp:122] _esp_now_rx_cb(): Checking peer d8:3b:da:e3:1c:cc
00:00:29.246 [  3851][V][ESP32_NOW.cpp:125] _esp_now_rx_cb(): Calling onReceive
00:00:29.246 The peer received a packet
00:00:30.345 [  4950][V][ESP32_NOW.cpp:112] _esp_now_rx_cb(): Broadcast from d8:3b:da:e3:1c:cc, data length : 1
00:00:30.346 0x00, // .
00:00:30.346 [  4951][V][ESP32_NOW.cpp:122] _esp_now_rx_cb(): Checking peer ff:ff:ff:ff:ff:ff
00:00:30.346 [  4951][V][ESP32_NOW.cpp:122] _esp_now_rx_cb(): Checking peer d8:3b:da:e3:1c:cc
00:00:30.346 [  4952][V][ESP32_NOW.cpp:125] _esp_now_rx_cb(): Calling onReceive
00:00:30.346 The peer received a packet

```

h2zero, 2025-01-27:
I can see that it's successfully advertising, your slave device doesn't scan for BLE advertisements though but it is receiving the esp now messages. What am I missing here?

Gabriele-Cano, 2025-01-28:
Yes what you described works, the problem js that the master device doesn't receive back the message from the slave after pairing.

You can see in the slave log I copied here 

```
00:00:28.248 [  2850][V][ESP32_NOW.cpp:115] _esp_now_rx_cb(): Calling new_cb, peer not found.
00:00:28.248 Unknown peer d8:3b:da:e3:1c:cc sent a broadcast message
00:00:28.248 Registering the peer as a master
00:00:28.248 [  2853][V][ESP32_NOW.cpp:18] _esp_now_add_peer(): d8:3b:da:e3:1c:cc
00:00:28.248 [  2856][V][ESP32_NOW.cpp:311] add(): Peer added - d8:3b:da:e3:1c:cc
00:00:28.248 [  2856][V][ESP32_NOW.cpp:384] send(): d8:3b:da:e3:1c:cc, data length 1
00:00:28.288 [  2894][V][ESP32_NOW.cpp:133] _esp_now_tx_cb(): d8:3b:da:e3:1c:cc : FAILED
00:00:28.289 [  2895][I][ESP32_NOW.h:78] onSent(): Message transmission to peer d8:3b:da:e3:1c:cc failed
00
```

h2zero, 2025-01-28:
Ahh okay, in that case the only thing I can think of is since the master device is advertising BLE the antenna is in use and the coexist functions may not work well with esp now, hard to say for sure, never tried this.

Gabriele-Cano, 2025-01-28:
It seems it should be working according to this page https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/coexist.html

I don't know what to try next

h2zero, 2025-01-28:
Does it work if you do not start advertising?

h2zero, 2025-01-28:
Some info here : https://esp32.com/viewtopic.php?t=23406

AlexLaramx, 2025-02-05:
check this example.

`#include <NimBLEDevice.h>
#include <esp_sleep.h>
#include <WiFi.h>
#include "ESP32_NOW.h"  // We use the new ESP32_NOW library
#include <esp_mac.h>
// =============================================
//                  DEFINES
// =============================================
#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"
#define BLE_WAIT_TIME       5000  // 5 seconds for BLE connection
#define SLEEP_TIME_SEC      5     // Sleep time in seconds
#define ESPNOW_WIFI_CHANNEL 6     // Wi-Fi channel for ESP-NOW
// Structure for ESP-NOW data
typedef struct struct_message {
    uint32_t counter;  // We use the same format as BLE
} struct_message;
// =============================================
//              GLOBAL VARIABLES
// =============================================
RTC_DATA_ATTR uint32_t count = 0;  // Persistent counter in RTC memory
struct_message myData;
// BLE objects
NimBLEServer* pServer = nullptr;
NimBLECharacteristic* pCharacteristic = nullptr;
bool deviceConnected = false;
// Class to handle ESP-NOW broadcast
class ESP_NOW_Broadcast_Peer : public ESP_NOW_Peer {
public:
    // Constructor of the class using the broadcast address
    ESP_NOW_Broadcast_Peer(uint8_t channel, wifi_interface_t iface, const uint8_t *lmk)
        : ESP_NOW_Peer(ESP_NOW.BROADCAST_ADDR, channel, iface, lmk) {}
    // Destructor of the class
    ~ESP_NOW_Broadcast_Peer() {
        remove();
    }
    // Function to properly initialize ESP-NOW and register the broadcast peer
    bool begin() {
        if (!ESP_NOW.begin() || !add()) {
            log_e("Failed to initialize ESP-NOW or register the broadcast peer");
            return false;
        }
        return true;
    }
    // Function to send a message to all devices on the network
    bool send_message(const uint8_t *data, size_t len) {
        if (!send(data, len)) {
            log_e("Failed to broadcast message");
            return false;
        }
        return true;
    }
};
// Instance of the broadcast peer
ESP_NOW_Broadcast_Peer broadcast_peer(ESPNOW_WIFI_CHANNEL, WIFI_IF_STA, NULL);
// =============================================
//             CALLBACKS & CLASSES
// =============================================
// BLE server callbacks
class ServerCallbacks: public NimBLEServerCallbacks {
  void onConnect(NimBLEServer* pServer, NimBLEConnInfo& connInfo) override {
    Serial.printf("Client address: %s\n", connInfo.getAddress().toString().c_str());
    deviceConnected = true;
    /**
      *  We can use the connection handle here to ask for different connection parameters.
      *  Args: connection handle, min connection interval, max connection interval
      *  latency, supervision timeout.
      *  Units; Min/Max Intervals: 1.25 millisecond increments.
[... 84 lines cut ...]
        WiFi.setChannel(ESPNOW_WIFI_CHANNEL);
        while (!WiFi.STA.started()) {
            delay(100);
        }
        // Register the broadcast peer
        if (!broadcast_peer.begin()) {
            Serial.println("Failed to initialize broadcast peer");
            Serial.println("Rebooting in 5 seconds...");
            delay(5000);
            ESP.restart();
        }
        
        sendDataEspNow();
        goToSleep(SLEEP_TIME_SEC);
    }
}
void loop() {
    if (deviceConnected) {
        sendDataBle();
        count++;  // Increment after sending
        delay(1000);  // Wait 1 second between sends
    } else {
        goToSleep(SLEEP_TIME_SEC);
    }
}`

AlexLaramx, 2025-02-05:
check this example.

```cpp
#include <NimBLEDevice.h>
#include <esp_sleep.h>
#include <WiFi.h>
#include "ESP32_NOW.h"  // We use the new ESP32_NOW library
#include <esp_mac.h>

// =============================================
//                  DEFINES
// =============================================
#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"
#define BLE_WAIT_TIME       5000  // 5 seconds for BLE connection
#define SLEEP_TIME_SEC      5     // Sleep time in seconds
#define ESPNOW_WIFI_CHANNEL 6     // Wi-Fi channel for ESP-NOW
// Structure for ESP-NOW data
typedef struct struct_message {
    uint32_t counter;  // We use the same format as BLE
} struct_message;

// =============================================
//              GLOBAL VARIABLES
// =============================================
RTC_DATA_ATTR uint32_t count = 0;  // Persistent counter in RTC memory
struct_message myData;
// BLE objects
NimBLEServer* pServer = nullptr;
NimBLECharacteristic* pCharacteristic = nullptr;
bool deviceConnected = false;
// Class to handle ESP-NOW broadcast
class ESP_NOW_Broadcast_Peer : public ESP_NOW_Peer {
public:
    // Constructor of the class using the broadcast address
    ESP_NOW_Broadcast_Peer(uint8_t channel, wifi_interface_t iface, const uint8_t *lmk)
        : ESP_NOW_Peer(ESP_NOW.BROADCAST_ADDR, channel, iface, lmk) {}
    // Destructor of the class
    ~ESP_NOW_Broadcast_Peer() {
        remove();
    }
    // Function to properly initialize ESP-NOW and register the broadcast peer
    bool begin() {
        if (!ESP_NOW.begin() || !add()) {
            log_e("Failed to initialize ESP-NOW or register the broadcast peer");
            return false;
        }
        return true;
    }
    // Function to send a message to all devices on the network
    bool send_message(const uint8_t *data, size_t len) {
        if (!send(data, len)) {
            log_e("Failed to broadcast message");
            return false;
        }
        return true;
    }
};
// Instance of the broadcast peer
ESP_NOW_Broadcast_Peer broadcast_peer(ESPNOW_WIFI_CHANNEL, WIFI_IF_STA, NULL);

// =============================================
//             CALLBACKS & CLASSES
// =============================================
// BLE server callbacks
class ServerCallbacks: public NimBLEServerCallbacks {
  void onConnect(NimBLEServer* pServer, NimBLEConnInfo& connInfo) override {
    Serial.printf("Client address: %s\n", connInfo.getAddress().toString().c_str());
    deviceConnected = true;
    /**
[... 92 lines cut ...]
            delay(100);
        }
        // Register the broadcast peer
        if (!broadcast_peer.begin()) {
            Serial.println("Failed to initialize broadcast peer");
            Serial.println("Rebooting in 5 seconds...");
            delay(5000);
            ESP.restart();
        }
        
        sendDataEspNow();
        goToSleep(SLEEP_TIME_SEC);
    }
}

void loop() {
    if (deviceConnected) {
        sendDataBle();
        count++;  // Increment after sending
        delay(1000);  // Wait 1 second between sends
    } else {
        goToSleep(SLEEP_TIME_SEC);
    }
}
```