h2zero/NimBLE-Arduino #883: Cannot get services from peripheral

NxSoftware, 2025-01-31:
# Scenario

I'm trying to discover services on BLE devices but I've not had much success at all, my current situation is that the `NimBLEClient::getServices(true)` method is hanging and not returning services for a connected client.

I've attached logs with NimBLE debug enabled and all of the relevant code below.

# Logs

```
Scanning for BLE devices...
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
D NimBLEScan: >> start: duration=30000
D NimBLEScan: Scan started
D NimBLEScan: << start()
I NimBLEScan: New advertiser: e1:a1:05:cc:e6:c9
ScanCallbacks::onDiscovered: Found device: Name: , Address: e1:a1:05:cc:e6:c9, manufacturer data: a705060012202500ca00001000000000000000, serviceUUID: 0xfe07, txPower: 0 (1070185792)
Is advertising: 1
Found interesting device!
Creating new client
Attempting to connect
D NimBLEClient: >> connect(e1:a1:05:cc:e6:c9)
D NimBLEScan: >> stop()
D NimBLEScan: << stop()
D NimBLEClient: >> handleGapEvent 
ClientCallbacks::onConnect to: e1:a1:05:cc:e6:c9
Get Services...
```

# Code

`main.cpp`
```cpp
void setup() {
  Serial.begin(115200);
  while (!Serial) {
    continue;
  }

  Serial.println("Scanning for BLE devices...");

  NimBLEDevice::init("");
  scan = NimBLEDevice::getScan();
  scan->setScanCallbacks(&scanCallbacks, false);
  scan->setActiveScan(true);
  scan->setMaxResults(0);
  scan->setInterval(100);
  scan->setWindow(100);
  scan->setFilterPolicy(0);
  scan->start(bleScanTimeMs, false, true);
}
```

`ScanCallbacks.h`
```cpp
#pragma once
#include <NimBLEScan.h>

class ClientCallbacks;

class ScanCallbacks : public NimBLEScanCallbacks {
public:
    ScanCallbacks(ClientCallbacks *clientCallbacks) {
        this->_clientCallbacks = clientCallbacks;
    };

    void onDiscovered(const NimBLEAdvertisedDevice *advertisedDevice) override;
    void onScanEnd(const NimBLEScanResults &results, int reason) override;

private:
    ClientCallbacks *_clientCallbacks;
};
```

`ScanCallbacks.cpp`
```cpp

ClientCallbacks clientCallbacks;
ScanCallbacks scanCallbacks(&clientCallbacks);

NimBLEUUID interestingUUID("FE07");

/** Initial discovery, advertisement data only. */
void ScanCallbacks::onDiscovered(const NimBLEAdvertisedDevice* advertisedDevice) {
    Serial.printf(
        "ScanCallbacks::onDiscovered: Found device: %s (%d)\n", 
        advertisedDevice->toString().c_str());

    bool isAdvertising = advertisedDevice->isAdvertisingService(interestingUUID);
    Serial.printf("Is advertising: %d\n", isAdvertising);
    if (!isAdvertising) {
        return;
    }
    
    Serial.println("Found interesting device!");

    NimBLEClient *client = NimBLEDevice::getDisconnectedClient();
    if (!client) {
        Serial.println("Creating new client");
        client = NimBLEDevice::createClient();
        if (!client) {
            Serial.println("Failed to create BLE client");
            return;
        }
    } else {
        Serial.println("Using existing disconnected client");
    }

    Serial.println("Attempting to connect");
    client->setClientCallbacks(this->_clientCallbacks, false);
    if (!client->connect(advertisedDevice, false, true)) {
        NimBLEDevice::deleteClient(client);
        Serial.println("Failed to connect");
        return;
    }
}

void ScanCallbacks::onScanEnd(const NimBLEScanResults& results, int reason) {
    printf("ScanCallbacks::onScanEnd: reason = %d\n", reason);
}
```

`ClientCallbacks.h`
```cpp
#pragma once
#include <NimBLEClient.h>

class ClientCallbacks : public NimBLEClientCallbacks {
    void onConnect(NimBLEClient* pClient) override;
    void onDisconnect(NimBLEClient* pClient, int reason) override;
};
```

`ClientCallbacks.cpp`
```cpp
#include <Arduino.h>
#include <BLE.h>
#include <ClientCallbacks.h>
#include <NimBLEDevice.h>

void ClientCallbacks::onConnect(NimBLEClient* pClient) {
    Serial.printf("ClientCallbacks::onConnect to: %s\n", pClient->getPeerAddress().toString().c_str());

    Serial.println("Get Services...");
    auto services = pClient->getServices(true); // this never seems to return
    Serial.printf("Found %d services\n", services.size());
}

void ClientCallbacks::onDisconnect(NimBLEClient* pClient, int reason) {
    Serial.printf("ClientCallbacks::onDisconnect: %s disconnected, reason = %d\n", pClient->getPeerAddress().toString().c_str(), reason);
}
```

thekurtovic, 2025-01-31:
Might not be a good idea to do so within a callback. Have you tried moving the code to the main loop?

NxSoftware, 2025-02-01:
Can confirm moving it out of the callback worked. I've got it in my own task as I've got LVGL running on the main loop and don't need BLE getting polled that frequently :) Thanks for your suggestion.

closed 2025-02-01 by NxSoftware