h2zero/NimBLE-Arduino #884: Weird crash with discoverAttributes() in 2.2.1

gkoh, 2025-02-04:
I was updating to the latest and greatest 2.2.1 to verify both #861 and #867 when I got a consistent crash.
Using the same code as per #861:
https://github.com/h2zero/NimBLE-Arduino/issues/861#issue-2788589893

and trying to connect to whatever happens to allow it in the area, things seem to go horribly wrong:
```
D NimBLERemoteService: >> retrieveCharacteristics()
I (2733) NimBLE: GATT procedure initiated: discover all characteristics; 
I (2740) NimBLE: start_handle=1 end_handle=5

D NimBLERemoteService: Characteristic Discovery >> status: 0 handle: 2
D NimBLERemoteService: Characteristic Discovery >> status: 0 handle: 4
D NimBLERemoteService: Characteristic Discovery >> status: 14 handle: -1
D NimBLERemoteService: << Characteristic Discovery
D NimBLERemoteService: << retrieveCharacteristics()
D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a00
I (2942) NimBLE: GATT procedure initiated: discover all descriptors; 
I (2949) NimBLE: chr_val_handle=3 end_handle=5

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d454a  PS      : 0x00060430  A0      : 0x800d3242  A1      : 0x3ffc6b70  
A2      : 0x3ffc9f20  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000003  
A6      : 0x3ffc6ba0  A7      : 0x0000000c  A8      : 0x800d452c  A9      : 0x3ffc6b40  
A10     : 0x00000000  A11     : 0x7669bc1f  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffc2370  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400d4547:0x3ffc6b70 0x400d323f:0x3ffc6bc0 0x400d1a59:0x3ffc6bf0 0x400f4aa3:0x3ffc6c20 0x40096ed2:0x3ffc6c50
```

The backtrace decodes to this:
```
x/.pio/libdeps/default/NimBLE-Arduino/src/NimBLERemoteCharacteristic.cpp:128
x/.pio/libdeps/default/NimBLE-Arduino/src/NimBLEClient.cpp:714 (discriminator 2)
x/src/main.ino:38
```

which is the closing brace of `NimBLERemoteCharacteristic::retrieveDescriptors()` ... OK ...

I jammed in esp-idf and enabled full heap poisoning and stack smash protection, but behaviour is still the same, no magical reveals.
Thanks to the beautifully clean commit history (really, thank you for this) I was again able to quickly run a bisect, which identifies 1f2676b8f32fe2a42f35b073651eb9bdd7c3de11 as the point of failure.

Running a backtrace on that commit crash yields:
```
x/.pio/libdeps/default/NimBLE-Arduino/src/NimBLERemoteCharacter.stic.cpp:118
x/.pio/libdeps/default/NimBLE-Arduino/src/NimBLEClient.cpp:714
x/src/main.ino:38 (discriminator 8)
```

That line of code is in `NimBLERemoteCharacteristic::retrieveDescriptors()`:
```
NimBLEUtils::taskWait(filter->taskData, BLE_NPL_TIME_FOREVER);
```

Let me know if more information is needed, happy to help.

thekurtovic, 2025-02-04:
It's a nullptr from `retrieveDescriptors` being called with default parameter after a recent change.
Try the PR once it passes

gkoh, 2025-02-04:
> It's a nullptr from `retrieveDescriptors` being called with default parameter after a recent change. Try the PR once it passes

Indeed, the PR fixes it. Thank you @thekurtovic!

thekurtovic, 2025-02-04:
Glad to hear, sorry about the oversight.

h2zero, 2025-02-04:
Thanks for the detailed report @gkoh and the fix @thekurtovic

h2zero, 2025-02-07:
@gkoh, I have created a new PR #888 please let me know if it fixes the issue.

h2zero, 2025-02-07:
Thanks, missed that for sure! Will update shortly.

h2zero, 2025-02-07:
Updated.

closed 2025-02-08 by h2zero

gkoh, 2025-02-09:
Can confirm that #888 fixes this.
Thanks @h2zero and @thekurtovic.

PR #888 (2025-02-08): [Bugfix]Provide default task data when retrieving all descriptors.
* Update the descriptor filter when trying again with different UUID sizes.

Fixes #884