h2zero/NimBLE-Arduino #886: Missing calls to onDisconnect() callback

benjie-git, 2025-02-06:
I'm using NimBLE 2.2.1 on an ESP32-S3.

I'm running a BLE server that allows up to 4 devices to connect.  I keep advertising unless I have 4 active connections.

I've noticed that when I have 2 connected clients, I can disconnect one (initiated from the client side), and I see an onDisconnect() call, and the client will quickly reconnect.  I can then disconnect from the other client, and I'll see an onDisconnect() call, and the client will reconnect.  I can continue alternating, and this keeps working.

But if I disconnect and reconnect from the same client 2 times in a row, the onDisconnect() call is not called the 2nd time, and the server->getConnectedCount() fails to decrement.  After disconnecting a 3rd time, advertising seems to stop, and my clients can no longer connect, although any remaining connected clients remain connected and continue working.  At that point I need to reboot the ESP32 to get advertising to work again.

h2zero, 2025-02-06:
How quickly are you repeating the disconnect? Example code?

benjie-git, 2025-02-06:
I get the same issue whether I disconnect within 1 sec of connecting, or wait 1 minute after connecting before disconnecting.  I'll test whether this happens in a trivial example, and show code.

benjie-git, 2025-02-06:
Here is a test case.  It acts as an XBox game controller.  When I disconnect quickly after connection a few times in a row, NimBLE seems to keep a connection that the mac thinks it has already closed.  After doing this a couple of times, it runs out of connection slots (4), and stops advertising/allowing new connections.

But I added workaround code into onConnect() (currently commented out), to disconnect any lingering connections from the same address as the current incoming connection.  If you can first reproduce the problem, then you can uncomment and try the workaround.

```c++
#include <Arduino.h>
#include <NimBLEDevice.h>
#include <NimBLEHIDDevice.h>

const char *XB_NAME = "Xbox Wireless Controller";
const char *XB_MANUFACTURER = "Microsoft";

#define SERVICE_UUID_DEVICE_INFORMATION        "180A"      // Service - Device information
#define CHARACTERISTIC_UUID_SYSTEM_ID          "2A23"      // Characteristic - System ID 0x2A23
#define CHARACTERISTIC_UUID_MODEL_NUMBER       "2A24"      // Characteristic - Model Number String - 0x2A24
#define CHARACTERISTIC_UUID_SERIAL_NUMBER      "2A25"      // Characteristic - Serial Number String - 0x2A25
#define CHARACTERISTIC_UUID_FIRMWARE_REVISION  "2A26"      // Characteristic - Firmware Revision String - 0x2A26

// Product: latest Xbox series X wireless controller
#define VENDOR_USB_SOURCE 0x02
#define XBOX_VENDOR_ID 0x045E

#define XBOX_INPUT_REPORT_ID 0x01
#define XBOX_EXTRA_INPUT_REPORT_ID 0x02
#define XBOX_OUTPUT_REPORT_ID 0x03
#define XBOX_EXTRA_OUTPUT_REPORT_ID 0x04

#define XBOX_PRODUCT_ID 0x0B13
#define XBOX_BCD_DEVICE_ID 0x0509
#define XBOX_SERIAL "09711282953030"
#define XBOX_MODEL "1914"
#define XBOX_FW_VER "5.9.2709.0"

static const uint8_t Xbox_HIDDescriptor[] {
    0x05, 0x01,                 //(GLOBAL) USAGE_PAGE         0x0001 Generic Desktop Page 
    0x09, 0x05,                 //(LOCAL)  USAGE              0x00010005 Game Pad (Application Collection)  
    0xA1, 0x01,                 //(MAIN)   COLLECTION         0x01 Application (Usage=0x00010005: Page=Generic Desktop Page, Usage=Game Pad, Type=Application Collection)
    0x85, XBOX_INPUT_REPORT_ID, //  (GLOBAL) REPORT_ID          0x01 (1)  
    0x09, 0x01,                 //  (LOCAL)  USAGE              0x00010001 Pointer (Physical Collection)  
    0xA1, 0x00,                 //  (MAIN)   COLLECTION         0x00 Physical (Usage=0x00010001: Page=Generic Desktop Page, Usage=Pointer, Type=Physical Collection)
    0x09, 0x30,                 //    (LOCAL)  USAGE              0x00010030 X (Dynamic Value)  
    0x09, 0x31,                 //    (LOCAL)  USAGE              0x00010031 Y (Dynamic Value)  
    0x15, 0x00,                 //    (GLOBAL) LOGICAL_MINIMUM    0x00 (0)  <-- Info: Consider replacing 15 00 with 14
    0x27, 0xFF, 0xFF, 0x00, 0x00,//    (GLOBAL) LOGICAL_MAXIMUM    0x0000FFFF (65535)  
    0x95, 0x02,                 //    (GLOBAL) REPORT_COUNT       0x02 (2) Number of fields  
    0x75, 0x10,                 //    (GLOBAL) REPORT_SIZE        0x10 (16) Number of bits per field  
    0x81, 0x02,                 //    (MAIN)   INPUT              0x00000002 (2 fields x 16 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
    0xC0,                       //  (MAIN)   END_COLLECTION     Physical 
    0x09, 0x01,                 //  (LOCAL)  USAGE              0x00010001 Pointer (Physical Collection)  
    0xA1, 0x00,                 //  (MAIN)   COLLECTION         0x00 Physical (Usage=0x00010001: Page=Generic Desktop Page, Usage=Pointer, Type=Physical Collection)
    0x09, 0x32,                 //    (LOCAL)  USAGE              0x00010032 Z (Dynamic Value)  
    0x09, 0x35,                 //    (LOCAL)  USAGE              0x00010035 Rz (Dynamic Value)  
    0x15, 0x00,                 //    (GLOBAL) LOGICAL_MINIMUM    0x00 (0) <-- Redundant: LOGICAL_MINIMUM is already 0 <-- Info: Consider replacing 15 00 with 14
    0x27, 0xFF, 0xFF, 0x00, 0x00,//    (GLOBAL) LOGICAL_MAXIMUM    0x0000FFFF (65535) <-- Redundant: LOGICAL_MAXIMUM is already 65535 
    0x95, 0x02,                 //    (GLOBAL) REPORT_COUNT       0x02 (2) Number of fields <-- Redundant: REPORT_COUNT is already 2 
    0x75, 0x10,                 //    (GLOBAL) REPORT_SIZE        0x10 (16) Number of bits per field <-- Redundant: REPORT_SIZE is already 16 
    0x81, 0x02,                 //    (MAIN)   INPUT              0x00000002 (2 fields x 16 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
    0xC0,                       //  (MAIN)   END_COLLECTION     Physical 
    0x05, 0x02,                 //  (GLOBAL) USAGE_PAGE         0x0002 Simulation Controls Page 
    0x09, 0xC5,                 //  (LOCAL)  USAGE              0x000200C5 Brake (Dynamic Value)  
    0x15, 0x00,                 //  (GLOBAL) LOGICAL_MINIMUM    0x00 (0) <-- Redundant: LOGICAL_MINIMUM is already 0 <-- Info: Consider replacing 15 00 with 14
    0x26, 0xFF, 0x03,           //  (GLOBAL) LOGICAL_MAXIMUM    0x03FF (1023)  
    0x95, 0x01,                 //  (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields  
    0x75, 0x0A,                 //  (GLOBAL) REPORT_SIZE        0x0A (10) Number of bits per field  
    0x81, 0x02,                 //  (MAIN)   INPUT              0x00000002 (1 field x 10 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
[... cut ...]
    hid->startServices();

    // Start BLE advertisement
    pServer->getAdvertising()->setAppearance(HID_GAMEPAD);
    pServer->getAdvertising()->addServiceUUID(hid->getHidService()->getUUID());
    pServer->getAdvertising()->enableScanResponse(true);
    pServer->getAdvertising()->start();
    printf("Advertising Started\n");
}

void loop() {
    inputReport.x += 200;
    inputReport.rz += 200;
    inputChar->setValue((uint8_t*)&inputReport, sizeof(inputReport));
    inputChar->notify();

    vTaskDelay(20);
}
```

h2zero, 2025-02-07:
Try changing this `pServer->updateConnParams(connInfo.getConnHandle(), 6, 12, 0, 180);` to have a lower supervision timeout, say 30.

benjie-git, 2025-02-07:
Thanks!  I just tested with `pServer->updateConnParams(connInfo.getConnHandle(), 6, 12, 0, 30);`  Unfortunately, this change doesn't completely solve my issue.  This issue is intermittent, and takes a few tries to trigger, but I think it is taking more disconnect/reconnect attempts to cause the issue now, with this change.  But it feels like it's masking some other problem.

h2zero, 2025-02-07:
It takes time for the disconnect to happen and if the peer is reconnecting faster than you disconnect from them then you will have ghost connections. reducing the timeout reduced the amount of time for the disconnect to register, it was testing, not fixing. 

In your code you are continuously advertising so the client is able to reconnect faster than the disconnect event can complete. The fact you saw a change indicates that, so, as an HID device you should only advertise when you do not have a peer connected, removing `NimBLEDevice::startAdvertising();` from the onConnect callback should fix this.

benjie-git, 2025-02-07:
Thanks.  My use case is using this as a controller for multiple clients at once, which I know is not standard HID operation, so I believe that to do this I need to continue advertising after each connection, so that the remaining clients can connect.

h2zero, 2025-02-07:
In that case I would suggest you stop advertising when disconnecting a client and then start it when the onDisconnect callback is called.

benjie-git, 2025-02-07:
Thanks.  The disconnects are being initiated by the clients.  I tried disabling advertising in `onDisconnect()` and enabling it again after a couple of seconds, but the same client will try to auto-reconnect immediately after disconnecting, and succeeds, maybe because advertising was still going, up until milliseconds earlier.  I additionally tried rejecting connections in `onConnect()` within this couple second time window after `onDisconnect()` is called, but I still ended up with ghost connections.

That said, my workaround included in the above code is working well enough for me for now.

doudar, 2025-02-18:
I'm getting the same issue, where pClient.isConnected() = false after a disconnect, but onDisconnect() is never called.

h2zero, 2025-02-18:
@doudar Are you referring to the client `onConnect` or server? Please check your callback signatures, add the `override` specifier to be sure.

doudar, 2025-02-18:
> [@doudar](https://github.com/doudar) Are you referring to the client `onConnect` or server? Please check your callback signatures, add the `override` specifier to be sure.

My issue is with the client - When a remote server disconnects, the onDisconnect() callback isn't being invoked. onConnect() is called correctly. 

```class MyClientCallback : public NimBLEClientCallbacks {
 public:
  void onConnect(NimBLEClient* pClient) override;
  void onDisconnect(NimBLEClient* pClient, int reason) override;
  void onPassKeyEntry(NimBLEConnInfo& connInfo) override;
  void onConfirmPasskey(NimBLEConnInfo& connInfo, uint32_t pass_key) override;
  void onAuthenticationComplete(NimBLEConnInfo& connInfo) override;
};

h2zero, 2025-02-18:
Weird, this should not be possible. The isconndcted function explicitly checks for a valid connection handle and the only time it is set to invalid is in the disconnect event, which calls the callback.

h2zero, 2025-02-18:
Please enable debug logs and look for this to be printed: https://github.com/h2zero/NimBLE-Arduino/blob/351f7c33dc5c971af1c487c6c16f3e5959d87f62/src/NimBLEClient.cpp#L935

h2zero, 2025-02-26:
@doudar could you please try removing the `if` block around this: https://github.com/h2zero/NimBLE-Arduino/blob/83030f6590184d1c750c96f10d5f3e6a07cc18b4/src/NimBLEClient.cpp#L957

h2zero, 2025-02-26:
Wondering if this yet again a unreliable disconnect event from the controller, need to find a way to reproduce this.

benjie-git, 2025-02-26:
@h2zero I confirmed that for my use case (ESP32 running as a server, allowing multiple clients) removing this if guard around the `onDisconnect()` call does not fix the issue.  I still intermittently see skipped `onDisconnect()` calls.  It only seems to happen sometimes, when I'm connecting and disconnecting very quickly - under 1 sec.  I manually disconnect from the client side, as soon as I see the client show that it has connected.  When this error happens, I can check in my own `onConnect()` code to see if I'm already connected to the incoming address, and if so, disconnect the old connection, and this seems to consistently work around the problem.

So maybe the client is optimistically showing that it has connected before the connection is complete, and I'm initiating a disconnect before the connection is fully established?

> [@doudar](https://github.com/doudar) could you please try removing the `if` block around this:
> 
> [NimBLE-Arduino/src/NimBLEClient.cpp](https://github.com/h2zero/NimBLE-Arduino/blob/83030f6590184d1c750c96f10d5f3e6a07cc18b4/src/NimBLEClient.cpp#L957)
> 
> Line 957 in [83030f6](/h2zero/NimBLE-Arduino/commit/83030f6590184d1c750c96f10d5f3e6a07cc18b4)
> 
>  if (rc != (BLE_HS_ERR_HCI_BASE + BLE_ERR_CONN_ESTABLISHMENT) || pClient->m_config.asyncConnect) {

h2zero, 2025-02-26:
Thank you @benjie-git, that suggest was mainly for @doudar as his application is seeing the issue on the client whereas yours is on the server side. There is no such `if` block in the server event handler so this was more seeking confirmation that the issue is the same on the client and therefore an upstream bug.

h2zero, 2025-02-26:
One thing that both of you may want to try is the NimBLE 1.6 core PR #891 as the issue may have been resolved there.

benjie-git, 2025-02-26:
Hah, of course.  Well, I tried using the nimble-core-1.6 branch, but my clients are not able to connect at all.  The server sees the clients connection attempt, but the client never sees the connection succeed.  I erased my flash, and cleared the client-side bonds as well, but no dice.

h2zero, 2025-02-26:
Thanks for trying, that's very strange, any debug logs? It's still a WIP so I haven't done any testing yet.

benjie-git, 2025-02-26:
I get the following debug output:
```
D NimBLEServer: >> handleGapEvent: 
Client connected 1                        <=== From my code's onConnect()
E NimBLEServer: Update params error: 6, 
D NimBLEAdvertising: >> Advertising start: duration=0, dirAddr=NULL
D NimBLEAdvertising: << Advertising start
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
D NimBLEServer: << handleGapEvent
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: mtu update event; conn_handle=1 mtu=255
D NimBLEServerCallbacks: onMTUChange(): Default
D NimBLEServer: << handleGapEvent
```

My code's onConnect() is called, but not my onConnParamsUpdate() nor my onAuthenticationComplete().

h2zero, 2025-03-21:
There has been quite a few reports of this issue and I am wondering if it is chip/arduino core version specific. It seems to revolve around the supervision timeout not generating the disconnect event. I've tested with a regular esp32, with arduino core versions 3.0.1 and 3.1.3 and am not able to repro this error, tested by walking out of range.

h2zero, 2025-04-23:
@benjie-git Please test with the latest nimble core update that was just merged, [8f32a23](https://github.com/h2zero/NimBLE-Arduino/commit/8f32a23e2c8a0071b98345915388eb0146b6bfdc)

h2zero, 2025-04-25:
@benjie-git can we close this?

closed 2025-04-25 by benjie-git