h2zero/NimBLE-Arduino #899: ESP32_BLE OTA UPDATE USING WIFI

Nandhini-271, 2025-02-24:
I am using the NimBLE library to retrieve the firmware URL from a control device via BLE. After successfully retrieving and reconstructing the URL, the OTA process begins. The firmware is downloaded using Wi-Fi and WiFiManager, and the update is applied.

However, when switching from BLE to Wi-Fi, I am encountering issues, including the following error messages:      

15:24:23.814 -> *wm:StartAP with SSID:  ESPCGMS
15:24:24.315 -> *wm:AP IP address: 192.168.4.1
15:24:24.315 -> *wm:Starting Web Portal 
15:24:37.124 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
15:24:37.230 -> Debug exception reason: Stack canary watchpoint triggered (nimble_host) 
15:24:37.230 -> Core  0 register dump:
15:24:37.230 -> PC      : 0x400999ae  PS      : 0x00060636  A0      : 0x80098bb7  A1      : 0x3ffd2910  
15:24:37.230 -> A2      : 0x3ffbdf2c  A3      : 0xffffffff  A4      : 0x00060623  A5      : 0x3f412d90  
15:24:37.230 -> A6      : 0x3ffd2acc  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00000001  
15:24:37.230 -> A10     : 0x007bdf2c  A11     : 0x3ffbdf2c  A12     : 0x3ffc6ab4  A13     : 0x00020000  
15:24:37.230 -> A14     : 0x00000031  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000001  
15:24:37.230 -> EXCVADDR: 0x00000000  LBEG    : 0x40092444  LEND    : 0x4009244f  LCOUNT  : 0x00000000  
15:24:37.230 -> 
15:24:37.230 -> 
15:24:37.230 -> Backtrace: 0x400999ab:0x3ffd2910 0x40098bb4:0x3ffd2940 0x40096d01:0x3ffd2960 0x40097152:0x3ffd2980 0x4013ba7a:0x3ffd29c0 0x4013bdb2:0x3ffd29e0 0x40128105:0x3ffd2a00 0x401290c3:0x3ffd2a20 0x401299cf:0x3ffd2a80 0x4012675a:0x3ffd2b20 0x400ee751:0x3ffd2b90 0x401d3d4f:0x3ffd2bf0 0x400f39d5:0x3ffd2c10 0x400f3a57:0x3ffd2c30 0x400f4649:0x3ffd2c50 0x400f467f:0x3ffd2c90 0x400e0b15:0x3ffd2cb0 0x400e7976:0x3ffd2cf0 0x401d3a22:0x3ffd2d60 0x400ea6b1:0x3ffd2d80 0x400e49b3:0x3ffd2da0 0x400f476b:0x3ffd2dc0 0x400f4797:0x3ffd2de0 0x400f5039:0x3ffd2e40 0x400eb05e:0x3ffd2ea0 0x400eb4cb:0x3ffd2ef0 0x400eb92f:0x3ffd2f60 0x400d2ca5:0x3ffd2fc0 0x400d354b:0x3ffd3300 0x400d35c4:0x3ffd3340 0x400e02b5:0x3ffd3360 0x400fc27f:0x3ffd3400 0x400fccae:0x3ffd3420 0x40105a95:0x3ffd3690 0x40105b52:0x3ffd36c0 0x401004c9:0x3ffd3700 0x40100e03:0x3ffd3730 0x401018ad:0x3ffd3750 0x400ff9b9:0x3ffd3780 0x40107ee7:0x3ffd37b0 0x4010695d:0x3ffd37e0 0x4010696f:0x3ffd3800 0x4008154d:0x3ffd3820 0x400fc63a:0x3ffd3840 0x400998ae:0x3ffd3860
15:24:37.329 -> 
15:24:37.329 -> 
15:24:37.329 -> 
15:24:37.329 -> 
15:24:37.329 -> ELF file SHA256: f078c470e7d0a211                                                      
15:24:37.693 -> Rebooting...                                              

It seems that when switching from BLE to Wi-Fi, the stack canary watchpoint is being triggered, leading to a Guru Meditation error.

Has anyone encountered a similar issue when switching between BLE and Wi-Fi? Any suggestions on how to properly transition between these two modes without causing a crash would be greatly appreciated.

h2zero, 2025-02-24:
Looks like you overflowed a buffer somewhere. There is no need to switch between BLE and WiFi, the work fine at the same time.

Nandhini-271, 2025-02-25:
Could you please provide more clarity on which buffer is overflowing and how to prevent it?

h2zero, 2025-04-23:
Closing this as there has been no follow up. Please re-open if there is still a problem.

closed 2025-04-23 by h2zero