h2zero/NimBLE-Arduino #900: Crash when ther is on onMTUchange callback

Hilbert70, 2025-02-24:
In our project we handle lego PU hubs and remotes, in some cases when the remote is not connected and on a reconnect I get the following "Guru meditation crash:
```
sp32-backtrace ./.pio/build/esp32/firmware.elf /tmp/backtrace 
PC: 0x400e2d0a is in NimBLEClient::handleGapEvent(ble_gap_event*, void*) (.pio/libdeps/esp32/NimBLE-Arduino/src/NimBLEClient.cpp:1163).
1163	            pClient->m_pClientCallbacks->onMTUChange(pClient, event->mtu.value);
BT-0: 0x400e2d07 is in NimBLEClient::handleGapEvent(ble_gap_event*, void*) (.pio/libdeps/esp32/NimBLE-Arduino/src/NimBLEClient.cpp:1163).
1163	            pClient->m_pClientCallbacks->onMTUChange(pClient, event->mtu.value);
BT-1: 0x400e734d is in ble_gap_call_event_cb (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:839).
839	        rc = cb(event, cb_arg);
BT-2: 0x400e7397 is in ble_gap_call_conn_event_cb (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:865).
865	    rc = ble_gap_call_event_cb(event, cb, cb_arg);
BT-3: 0x400e898f is in ble_gap_mtu_event (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:7222).
7222	    ble_gap_call_conn_event_cb(&event, conn_handle);
BT-4: 0x400e5435 is in ble_att_clt_rx_mtu (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_att_clt.c:140).
140	            ble_gap_mtu_event(conn_handle, BLE_L2CAP_CID_ATT, mtu);
BT-5: 0x400e5269 is in ble_att_rx (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_att.c:519).
519	    rc = entry->bde_fn(conn_handle, om);
BT-6: 0x400eca8b is in ble_hs_hci_evt_acl_process (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_evt.c:1165).
1165	        rc = rx_cb(conn->bhc_rx_chan);
BT-7: 0x400eb87c is in ble_hs_process_rx_data_queue (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:249).
249	        ble_hs_hci_evt_acl_process(om);
BT-8: 0x400eb887 is in ble_hs_event_rx_data (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:546).
546	    ble_hs_process_rx_data_queue();
BT-9: 0x400813bd is in nimble_port_run (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:530).
530	    ev->fn(ev);
BT-10: 0x400e2eb3 is in NimBLEDevice::host_task(void*) (.pio/libdeps/esp32/NimBLE-Arduino/src/NimBLEDevice.cpp:832).
832	    nimble_port_run(); // This function will return only when nimble_port_stop() is executed
```

Shouldn't here
```
            NIMBLE_LOGI(LOG_TAG, "mtu update: mtu=%d", event->mtu.value);
            pClient->m_pClientCallbacks->onMTUChange(pClient, event->mtu.value);
            rc = 0;
```

be a check if that function is declared?

I fixed it on my side by adding a dummy function "onMTUChange"

h2zero, 2025-02-24:
It is not possible for the callback to not exist, it will always call the default one if you have not provided an override. I would need to see your code but I suspect the client instance has been deleted when that event was triggered which is the only way I could see this causing a crash.

Hilbert70, 2025-02-25:
The code can be found at [MTC4BT](https://github.com/Mattzobricks/MattzoControllers/tree/develop/src/MTC4BT) probably too big for you to find the error in my coding ;-)

Most stuff is happening in `BLEHub.cpp`

Maybe related:
```
esp32-backtrace ./.pio/build/esp32/firmware.elf /tmp/backtrace 
BT-1: 0x400e73b5 is in ble_gap_call_event_cb (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:839).
839	        rc = cb(event, cb_arg);
BT-2: 0x400e73ff is in ble_gap_call_conn_event_cb (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:865).
865	    rc = ble_gap_call_event_cb(event, cb, cb_arg);
BT-3: 0x400e89f7 is in ble_gap_mtu_event (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:7222).
7222	    ble_gap_call_conn_event_cb(&event, conn_handle);
BT-4: 0x400e549d is in ble_att_clt_rx_mtu (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_att_clt.c:140).
140	            ble_gap_mtu_event(conn_handle, BLE_L2CAP_CID_ATT, mtu);
BT-5: 0x400e52d1 is in ble_att_rx (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_att.c:519).
519	    rc = entry->bde_fn(conn_handle, om);
BT-6: 0x400ecaf3 is in ble_hs_hci_evt_acl_process (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_evt.c:1165).
1165	        rc = rx_cb(conn->bhc_rx_chan);
BT-7: 0x400eb8e4 is in ble_hs_process_rx_data_queue (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:249).
249	        ble_hs_hci_evt_acl_process(om);
BT-8: 0x400eb8ef is in ble_hs_event_rx_data (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:546).
546	    ble_hs_process_rx_data_queue();
BT-9: 0x400813bd is in nimble_port_run (.pio/libdeps/esp32/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:530).
530	    ev->fn(ev);
BT-10: 0x400e2f1b is in NimBLEDevice::host_task(void*) (.pio/libdeps/esp32/NimBLE-Arduino/src/NimBLEDevice.cpp:832).
832	    nimble_port_run(); // This function will return only when nimble_port_stop() is executed
```

The serial log:
```
[0052] [1] [INFO] Try mqtt reconnect... 
[0053] [1] [INFO] MQTT: Connected MTC4BT
[0054] [1] [INFO] Reconnected!
[0055] [1] [DEBUG] SetE-Brake: 0
[0056] [1] [DEBUG] SetE-Brake: 0
[0057] [0] [DEBUG] BLE : mtuChange from hub '90:84:2b:c0:52:74' value: 158.
[0058] [0] [INFO] BLE : Tying to connecto to hub '90:84:2b:c0:52:74'.
[0059] [0] [INFO] BLE : Connected to hub '90:84:2b:c0:52:74'.
[0060] [0] [DEBUG] hub connected with address 90:84:2b:c0:52:74, has callback 1
[0061] [0] [INFO] BLE : Watchdog timeout not set for PU hubs
[0062] [0] [INFO] BLE : Connecting to hub '90:84:2b:23:af:0e'...
[0063] [0] [DEBUG] hub created: 1073664180 (90:84:2b:23:af:0e)
[0064] [0] [INFO] PU  : Found integrated RGB LED at port 32
[0065] [0] [DEBUG] BLE : mtuChange from hub '90:84:2b:23:af:0e' value: 158.
[0066] [0] [INFO] BLE : Tying to connecto to hub '90:84:2b:23:af:0e'.
[0067] [0] [INFO] BLE : Connected to hub '90:84:2b:23:af:0e'.
[0068] [0] [DEBUG] hub connected with address 90:84:2b:23:af:0e, has callback 1
[0069] [0] [INFO] BLE : Watchdog timeout not set for PU hubs
[0070] [0] [INFO] PU  : Found integrated RGB LED at port 32
[0071] [0] [INFO] Loop: Connected to all hubs of loco 'BR146'.
[0072] [0] [INFO] BLE : Connecting to hub 'e4:e1:12:9b:42:14'...
[0073] [0] [DEBUG] hub created: 1073665032 (e4:e1:12:9b:42:14)
[0074] [0] [WARN] BLE : Failed to connect to hub 'e4:e1:12:9b:42:14', deleted client.
[0075] [0] [WARN] Loop: Connect failed. Will retry...
[0076] [0] [INFO] BLE : Scanning for 1 hub(s)...
[0077] [0] [INFO] BLE : Scanning for 1 hub(s) aborted.
[0078] [0] [INFO] BLE : Discovered hub:  (e4:e1:12:9b:42:14).
[0079] [0] [INFO] BLE : Connecting to hub 'e4:e1:12:9b:42:14'...
[0080] [0] [DEBUG] hub created: 1073625472 (e4:e1:12:9b:42:14)
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x02889a85  PS      : 0x00060930  A0      : 0x800e73b8  A1      : 0x3ffeadd0  
A2      : 0x3ffeae8c  A3      : 0x3ffe3980  A4      : 0x02889a85  A5      : 0x3ffebf10  
A6      : 0x00000002  A7      : 0x00000001  A8      : 0x800e2d78  A9      : 0x3ffeae00  
A10     : 0x3ffed1d4  A11     : 0x3ffe3980  A12     : 0x00000017  A13     : 0x00000000  
A14     : 0x3ffdfbe0  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x00000014  
EXCVADDR: 0x02889a84  LBEG    : 0x40091e0c  LEND    : 0x40091e17  LCOUNT  : 0x00000000  

[... 39 lines cut ...]
			log4MC::warn("BLE : Max clients reached - no more connections available.");
			_isDiscovered = false;
			return false;
		}

		_hub = NimBLEDevice::createClient();
		log4MC::vlogf(LOG_DEBUG,"hub created: %d (%s)",_hub,_config->DeviceAddress->toString().c_str());
		if (!_hub) {
			log4MC::vlogf(LOG_ERR,"BLE: Could not create hub for %s!",_config->DeviceAddress->toString().c_str());
			_isDiscovered = false;
			return false;
		}
		if (_clientCallback == nullptr) {
			_clientCallback = new BLEClientCallback(this);
		}
		if (!_clientCallback) {
			log4MC::vlogf(LOG_ERR,"BLE: Could not create client callback for %s!",_config->DeviceAddress->toString().c_str());
			NimBLEDevice::deleteClient(_hub);
			_hub= NULL;
			_isDiscovered = false;
			return false;
		}
		_hub->setClientCallbacks( _clientCallback, true);

```

h2zero, 2025-02-25:
I see the problem, if the connection fails and you delete the client then the callback that was created with `new` will also be deleted but the pointer that it was assigned to will not be set to `nullptr` and therefore not recreated the next time the connection is attempted and the client callbacks will be set to the previously allocated memory which is now invalid.

Hilbert70, 2025-02-26:
Oh thanks a lot, I would not have found that one without your help.

And thank you a lot for making the project!!!

closed 2025-02-26 by Hilbert70