h2zero/NimBLE-Arduino #905: Suspected regression in release 2.2.2 when connecting to device found in scanning

mdxs, 2025-02-26:
After upgrading to release 2.2.2 it appears that connecting to a device (using `connect` with a NimBLEAdvertisedDevice) found in scanning fails.

This code is working fine in release 2.2.1, so it appears there is a regression in 2.2.2 as released on 25 February 2025 and found in the Arduino IDE as an update to apply.

mdxs, 2025-02-26:
Looking at the commits, it appears that the following:
[Partially revert Commit](https://github.com/h2zero/NimBLE-Arduino/commit/d6013fd3342ca1b33f9a3a9ade2858087aee8617) https://github.com/h2zero/NimBLE-Arduino/commit/9397a497f4390f52ccd1fd49872f8ee99977cfa2 [to restore client connect with device parameter.](https://github.com/h2zero/NimBLE-Arduino/commit/d6013fd3342ca1b33f9a3a9ade2858087aee8617)

was done after:
[Release 2.2.2](https://github.com/h2zero/NimBLE-Arduino/commit/57db938804415830846be73903fad4479fe1aeab)

Which appears (to me) to be relevant to the observed regression.

h2zero, 2025-02-26:
Thanks, yes this has been fixed, new release later today to address it.

Berg0162, 2025-02-27:
Dear h2zero,
I am sorry to report that with version 2.2.2 NimBLEClient::connect() is no longer working like before... 
I could not find any direct reference in the changelog to changes in this member.
My code works perfectly with version 2.2.1 (and earlier 2.x versions), however after installing 2.2.2 pClient->connect(trainerDevice, true) ALWAYS returns false!
Nor onConnect() is called, neither onConnectfail() is called, what I would expect in this case, to expose the reason of failure...

```C++
bool ClientSide::clientConnectServer(void) {
    // Handle first time connect AND a reconnect. One fixed Peripheral (trainer) to account for!
    if(pClient == nullptr) { // First time -> create new pClient and service database!
      pClient = NimBLEDevice::createClient(); 
      pClient->setClientCallbacks(new client_Connection_Callbacks(this));
      // First Time Connect to the FTMS Trainer (Server/Peripheral)
      hasConnectPassed = pClient->connect(trainerDevice, true);   // Delete attribute objects and Create service database
    } else if(pClient == NimBLEDevice::getDisconnectedClient()) { // Allow for a streamlined reconnect
        // Reconnect to the disconnected FTMS Trainer (Server/Peripheral)
        hasConnectPassed = pClient->connect(trainerDevice, false);  // Just refresh the service database 
      } 

    if(!hasConnectPassed) 
      return hasConnectPassed; // Connect failed!
.
.
```

h2zero, 2025-02-27:
The fix is the the master branch, haven't had time to release it yet.

mdxs, 2025-03-06:
Confirmed fixed in released v2.2.3

closed 2025-03-06 by mdxs