h2zero/NimBLE-Arduino #906: ESP32-P4 Compilation Fails Due to Unconditional esp_bt.h Includes

landfishllc, 2025-02-28:
Hi `NimBLE-Arduino` maintainers!!!!🐟 👋 🤗 Thanks for your hard wizarding work ❤

I'm working with an [ESP32P4 dev board](https://www.aliexpress.com/i/3256808141773824.html), which includes a peripheral ESP32C6, pre-configured for [ESP-Hosted](https://github.com/espressif/esp-hosted)

When I compile a minimal source file containing a `#include <BleGamepad.h>` statement, the build fails at `libraries/NimBLE-Arduino/src/nimble/esp_port/esp-hci/esp_nimble_hci.c:18` due to an unconditional `#include <esp_bt.h>` (P4 has no onboard radio) even though [commit 081bb7f](https://github.com/h2zero/NimBLE-Arduino/commit/081bb7f) added ESP32P4 include guards elsewhere/attempted to implement Hosted compatibility.

(My `sdkconfig` is correctly set up, I have been editing the template `sdkconfig` used by arduino-cli)

### Steps to Reproduce:

    Board: ESP32-P4 DevKit (Arduino IDE 2.3.2)
    ESP32 Core: 3.2.0-RC1
    Library: NimBLE-Arduino 2.2.2
    Error:

1) Create an Arduino project
2) Install and include NimBLE-dependent library (such as ESP32-BLE-Gamepad)
3) Compile

```
src/nimble/esp_port/esp-hci/src/esp_nimble_hci.c:18:10: fatal error: esp_bt.h: No such file or directory  
   18 | #include <esp_bt.h>  
```

### Expected Behavior:

Successful compilation when CONFIG_IDF_TARGET_ESP32P4 is defined.

### Workaround Attempt:

Wrapping the include with #if !defined(CONFIG_IDF_TARGET_ESP32P4) in affected files resolves the immediate error, but causes subsequent errors (`undefined reference`, etc.) that get worse as I include more wraps

### Question:

Is ESP32P4 supported yet?

Environment:

-:: Linux x64

    ESP-IDF: v5.2.1
    Arduino Core: esp32:esp32@3.2.0-RC1

Thank you for your time! Let me know if more details would help.

h2zero, 2025-02-28:
Hello @landfishllc, the CPP code in this repo mirrors [esp-nimble-cpp](https://github.com/h2zero/esp-nimble-cpp) which does support the P4. This repo cannot support it due to dependencies and conflicts in the Arduino core. You will need to use esp-nimble-cpp and esp-idf to support this board.

landfishllc, 2025-02-28:
h2zero, you are my h2hero, thanks for getting back to me so fast.

Unfortunately, my product parameters require the use of Arduino IDE (target developers will be children and teens). 

I will attempt to incorporate the esp-nimble-cpp IDF component into the local esp-idf managed by Arduino.  If this does not work out, I will write custom C6 firmware and comm with AT commands.

I am doing my best Googling, but as of now, do you know of any BLE library for Arduino IDE supporting the Hosted architecture?

h2zero, 2025-02-28:
That's unfortunate but understandable. In that can you can make a custom arduino core and enable Nimble and add esp-nimble-cpp, Tasmota has a similar configuration. You'll need to add the hosted library as well of course.

I haven't seen anyone attempt this with Arduino so you're probably the first, good luck!

lyusupov, 2025-06-21:
is there any progress on that ?
Recent [**Release 2.3.1**](https://github.com/h2zero/NimBLE-Arduino/releases/tag/2.3.1) still gives build failure with ESP32-P4 and Arduino Core 3.x...

closed 2026-04-10 by h2zero