h2zero/NimBLE-Arduino #907: Structural problems to subscribe remote characteristics with version 2.2.3 installed!

Berg0162, 2025-03-01:
Dear h2zero,
Yes, my code can pClient->connect(...) again with version 2.2.3 installed, thank you for that!
However, since installation of 2.2.3 I have structural problems to subscribe remote characteristics like:
```C++
pRemote_CP_ControlPoint_Chr->subscribe(indications, Static_CP_ControlPoint_Indicate_Callback);
```
In the 2.2.1 version, connecting and subscribing worked perfectly.... My code has 6 remote characteristics to subscribe to and 2 out of 6 are failing with 2.2.3 installed! The problems remind me to issue (https://github.com/h2zero/NimBLE-Arduino/issues/867)... Is it back again?

h2zero, 2025-03-01:
Oof, this should be working fine, I'll have a look, thanks!

h2zero, 2025-03-01:
@Berg0162 I cannot reproduce this issue, can you share any code snippets that may help?

Berg0162, 2025-03-02:
Dear h2zero,
My application is a BLE MITM for indoor cycling that extracts virtual road information transmitted by Zwift to the bike trainer. The info is used to lift the front wheel of the bike to simulate changing road incline during the virtual Zwift ride.
Below are code snippets (for 1 out of 6 characteristics) that handle the client's (un)subscribing action when connections are (dis)established at both sides.
When these (un)subscribe calls fail something is really wrong like recently with issue (https://github.com/h2zero/NimBLE-Arduino/issues/867). Yesterday that happened suddenly after installing the new NimBLE-Arduino version, when I was in a process of testing the robustness of code. However, today cause and effect is not clear to me anymore, since I cannot reproduce the error state... Probably it was an unexpected server/trainer problem rather than a MITM client problem. I feel ashamed that I have bothered you! I'm sorry I took up your valuable time!

```C++

void HRM::Static_client_HR_Measurement_Notify_Callback(NimBLERemoteCharacteristic* pBLERemoteCharacteristic, uint8_t* pData, size_t length, bool isNotify) {
    if (HRM::instance) {
        HRM::instance->client_HR_Measurement_Notify_Callback(pBLERemoteCharacteristic, pData, length, isNotify);
    }
}

void HRM::client_HR_Subscribe(void) {
  if ( pRemote_HR_Measurement_Chr != nullptr )
    if( !pRemote_HR_Measurement_Chr->subscribe(notifications, Static_client_HR_Measurement_Notify_Callback) ) 
      LOG(">>> ERROR remote HRM Measure Subscribe failed!");                
}

void HRM::client_HR_Unsubscribe(void) {
  if ( pRemote_HR_Measurement_Chr != nullptr )
    if( !pRemote_HR_Measurement_Chr->unsubscribe(false) )
      LOG(">>> ERROR remote HRM Measure Unsubscribe failed!");
}

```

closed 2025-03-02 by Berg0162

h2zero, 2025-03-02:
Thanks for getting back to me and I hope the issue is resolved. Please reopen this if you encounter it any further.