h2zero/NimBLE-Arduino #909: Connection Negotiation fails from iOS Device

arturdev, 2025-03-05:
I'm having an issue connecting the iOS Device (central) to the ESP32-s3-mini (peripheral). 
Sometimes, the iOS device fails to connect to the peripheral, and an Apple engineer claims that it is solely a firmware issue.
More is here: https://developer.apple.com/forums/thread/775811?page=1#827665022 

More specifically, the Apple Engineer stated:

> iOS requires a specific set of connection parameters, and will reject anything that does not comply.
> 
> These rules are described in [Accessory Design Guidelines for Apple Products](https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf)
> Connection parameter requests may be rejected if they do not meet the guidelines in this section. General connection parameter request guidelines:
> 
> Peripheral Latency ≤ 30 connection intervals.
> Supervision Timeout from 6 seconds to 18 seconds.
> Interval Min ≥ 15 ms.
> Interval Min ≤ 2 seconds.
> Interval Min is a multiple of 15 ms.
> One of the following:
> Interval Max at least 15 ms greater than Interval Min.
> Interval Max and Interval Min are both 15 ms.
> Interval Max * (Peripheral Latency + 1) of 6 seconds or less.
> Supervision Timeout greater than Interval Max * (Peripheral Latency + 1) * 3.
> 

Any clue? Is there anything I can do from my cpp code to make it work? Or is it a limitation in NimBLE stack? Or maybe something else?

arturdev, 2025-03-07:
@h2zero 
That's the thing that on NimBLE side I don't see any errors. 
I have enabled debug logs, and when the iOS device (when in locked state) initiates a connection, in the SerialMonitor I don't see any connection requests/attempts at all. I even added some logs in ble_gap.c to check if something is happening in lower level... but no results. nothing is being called. 
However, on iOS side, the system delegate events of a successful connection, and right after a disconnection, reasoning that the peripheral initiated the disconnection..

h2zero, 2025-03-12:
Do you have bonding enabled? This can happen if the bond is deleted.

arturdev, 2025-03-13:
No, I intentionally disabled the bonding because otherwise, mobile devices would have remembered the peripheral and it would have appeared in the Bluetooth list.. and the users would have ended up with hundreds of Bluetooth devices in their list, haha