h2zero/NimBLE-Arduino #910: ExtAdvertisement on ESP32 - data length issue

Jeronimo59, 2025-03-07:
Hello,

Thank you for this great work.

When trying to send/receive extended advertisements (using NimBLEExt.setManufacturerdata()) with more than 28 bytes, I get the following message on the receiver's logs:
`E (22095) NimBLE: Received HCI data length at host (74) exceeds maximum configured HCI event buffer size (70).` 

After some investigation, I found the log originates from file esp_nimble_hci.c (line 198):
`        if (totlen >  {
            ESP_LOGE(LOG_TAG, "Received HCI data length at host (%d) exceeds maximum configured HCI event buffer size (%d).",
                     totlen, MYNEWT_VAL(BLE_TRANSPORT_EVT_SIZE));
            ble_hs_sched_reset(BLE_HS_ECONTROLLER);
            return 0;
        }`

The message size appears to be limited by MYNEWT_VAL(BLE_TRANSPORT_EVT_SIZE)) that is defined (for ESP platform) in esp_nimble_cfg.h (line 1592) with a value of 70, in line with the log message:

`#ifndef MYNEWT_VAL_BLE_TRANSPORT_EVT_SIZE
# define MYNEWT_VAL_BLE_TRANSPORT_EVT_SIZE (70)
#endif
`
Changing the value to 200 seems to solve the problem, as I managed to receive the data properly.

I guess this definition of MYNEWT_VAL_BLE_TRANSPORT_EVT_SIZE should be somehow increased to over 200 when Extended Advertisement is enabled (CONFIG_BT_NIMBLE_EXT_ADV defined).

Any help on how to do this properly (not sure what should be the value) will be welcome.

davidk88, 2025-03-12:
Hello,

I'm getting similiar error:

E (7254378) NimBLE: Received HCI data length at host (165) exceeds maximum configured HCI event buffer size (70).

davidk88, 2025-03-12:
I added -DDMYNEWT_VAL_BLE_TRANSPORT_EVT_SIZE=256 to build_flags (platformIO)

Jeronimo59, 2025-03-12:
Hello

I did the same, and got it to broadcast (100ms interval) up to 220 bytes. Works perfectly with 2 receiving devices , but I guess it should work with any number of them.

I am still interested by h2zero's feedback on this fix, as there may be side effects, and it should be better applied in the original code.

> Hello,
> 
> I'm getting similiar error on ESP32-C3:
> 
> E (7254378) NimBLE: Received HCI data length at host (165) exceeds maximum configured HCI event buffer size (70).

h2zero, 2025-03-12:
Thank you for reporting this. There was a change in the available config settings and the default for this is supposed to be 257 when extended advertising is enabled. I will make this correction for the next release, in the mean time you should set this value yourself either in `nimconfig.h` or on the command line.

Jeronimo59, 2025-03-18:
> Thank you for reporting this. There was a change in the available config settings and the default for this is supposed to be 257 when extended advertising is enabled. I will make this correction for [...]

Thank you for that.

FYI, all is fine with messages up to 223 bytes (+6 byte-header). With 224 bytes (=230 bytes incl. header), logs show that packet is received (NimBLEScanCallbacks: Discovered), but NimBLEAdvertisedDevice::haveManufacturerData() returns false.

Maybe this is expected behavior, or another little fix with config settings could give us a few more bytes... 

But this is already a huge benefit vs 31 bytes, it makes my project (real-time control of a dozen trains) possible !

In case this helps someone, I found that restraining the scan to a single PHY (hence avoiding channel change) on the receiver side ensures virtually 100% reception of packets when broadcasting every 100ms. I need to figure out how to use other PHYs to establish occasional connections while keeping the bradcasting...

Thank you again.

Jeronimo59, 2025-03-18:
> I added -DDMYNEWT_VAL_BLE_TRANSPORT_EVT_SIZE=256 to build_flags (platformIO)

I guess you mean -DMYNEWT_VAL_BLE_TRANSPORT_EVT_SIZE=256  ;-)

h2zero, 2025-04-23:
Closing as resolved in [8f32a23](https://github.com/h2zero/NimBLE-Arduino/commit/8f32a23e2c8a0071b98345915388eb0146b6bfdc)

closed 2025-04-23 by h2zero