h2zero/NimBLE-Arduino #912: characteristic larger than 253 bytes freezes the onRead callback

jingoro2112, 2025-03-13:
from standard characteristic read example:
```
void onRead( NimBLECharacteristic* characteristic, NimBLEConnInfo& connInfo ) override
{
	const uint8_t buf[256] = {0};
	characteristic->setValue( buf, 253 ); // for values of 253 or less, it works as expected, getting a callback for each read request
	characteristic->setValue( buf, 254 ); // onRead() is never called again, despite the client making and receiving a read request
}
```

Just stumbled across this, It is possible I'm doing something wrong? but there seems to be something magical happening at the 253-4 byte boundary, I suspect an int8 being used with some kind of two-byte header? I have not looked at the source yet, spent too many hours already zeroing in on this problem.

jingoro2112, 2025-03-13:
this is on an esp32 S3 I should have mentioned. I tried with two separate GATT handlers (on the linux/raspberry pi side) and same result.

I swapped out NimBLE for BLEDevice (ArduinoBLE) and the problem went away so it seems certainly related to NimBLE

As a side note I get about 50% more throughput with the ArduinoBLE implementation.

jingoro2112, 2025-03-14:
> Thanks for reporting this. The issue appears to be that the client is
> using a normal read rather than a long read. There is an unfortunate
> situation with the NimBLE stack itself where the onRead callback would be
> called successively for the same read operation when a characteristic value
> is longer than the MTU -3, I have tried to handle this here via detecting a
> long read PDU so that the application callback is only called once, if the
> client instead send normal read commands and the characteristic value is
> more than the MTU - 3 then the callback will not be called. Stuck between a
> rock and a hard place with this one sadly, my suggestion is to increase the
> MTU or if possible, use long read commands.

Hmm.. using gattlib (which drives Bluez) I am using
gattlib_read_char_by_uuid(...) which does.. whatever it does. I'll see if
maybe I can set a flag on the connection object. Thx for taking a look!

jingoro2112, 2025-03-14:
follow-up: upping the MTU ```NimBLEDevice::setMTU( 512 );``` does fix both the throughput and read issues, thx again for the attention.

I will leave it to you to close/defer this issue since there seems to be an underlying tension.

h2zero, 2025-03-14:
Thanks for confirming, I have opened #913 to resolve this properly.

PR #913 (2025-04-13): Bugfix - Characteristic onRead callback not called.
Fixes detection of the original read request vs a followup read for characteristic values greater than MTU - 3.

Fixes #912
closed 2025-04-13 by h2zero