h2zero/NimBLE-Arduino #916: Question about NimBLECharacteristicCallbacks::onStatus()

sylque, 2025-03-20:
First, thanks for the amazing work on this library!

Suppose I am a server with two connected clients, A and B. Suppose I notify (or indicate) both A and B. Will I receive two `NimBLECharacteristicCallbacks::onStatus()` events? If I do, how do I know which is which?

Thanks.

h2zero, 2025-03-21:
There should only be one callback event, are you seeing more than that?

sylque, 2025-03-21:
My current application notifies remote clients blindly, without any check. I want to change that and use indications, in order to detect when a client has been turned off unexpectedly. When this happens, I would like to know which client has a problem, hence my question.

I haven't tried yet, but looking at [this code](https://github.com/h2zero/NimBLE-Arduino/blob/master/src/NimBLECharacteristic.cpp#L272-L294) in `NimBLECharacteristic.cpp`, I got the feeling that the multiple calls to `ble_gattc_indicate_custom()` (in the loop over clients) would translate into multiple `BLE_GAP_EVENT_NOTIFY_TX` events, and hence into multiple calls to `onStatus()`.

h2zero, 2025-03-21:
That only applies if sending a value via notify/indicate instead of a call without a value which uses the characteristic value instead.

sylque, 2025-03-21:
Yes, I call `indicate()` with a value. I figured it would be more efficient this way, skipping the `NimBLEAttValue` class altogether.

From what you've said, I understand that using `indicate()` without a value would get me a single `onStatus()` event. But how would I know which client has failed?

h2zero, 2025-03-21:
The only way an indication or notification would fail that I have seen is that the client has disconnected, which would generate a disconnect event. At the moment that would be the only way, in the future I will look to add the connection information to this callback.

sylque, 2025-03-23:
Checking the disconnect event sounds like a good solution, thank you.

closed 2025-03-23 by sylque

sylque, 2025-07-18:
> From what you've said, I understand that using indicate() without a value would get me a single onStatus() event

For those landing here, notice that the above assumption is wrong:  calling `indicate()` with or without a value gets you one `onStatus()` call per client.