h2zero/NimBLE-Arduino #917: Ondisconnect function not working when device disconnecting

kushalpatelkushal, 2025-03-21:
Hi,

I am using NimBLE v1.4.2. when i am connecting to device, it connects to device and i get onconnect callback. when i disconnect, i got ondisconnect callback. Issue happens when device is sending data like 128 data in 1 second, when device disconnect, i dont get ondisconnect callback. I dont know what to do in this case? I also tried numconnected device function to find if device is connected or not.

h2zero, 2025-03-21:
I believe this to be an issue with the controller, which requires a fix/update with the Arduino core and/or esp-idf.

h2zero, 2025-03-21:
I just tested this with arduino core 3.0.1 and it works fine, this must be something in the new core versions.

h2zero, 2025-03-21:
Appears to work fine win arduino core 3.1.3 as well.

kushalpatelkushal, 2025-03-24:
i am using esp32 controller. any test code to test on this issue with esp32?

h2zero, 2025-04-23:
@kushalpatelkushal Please test with the latest nimble core update that was just merged [8f32a23](https://github.com/h2zero/NimBLE-Arduino/commit/8f32a23e2c8a0071b98345915388eb0146b6bfdc)

h2zero, 2025-04-25:
It's been reported that the issue has be resolved in the latest Arduino core. Please update that to version 3.2 and test.

h2zero, 2025-09-02:
Closing as there has been no update, please reopen if it's still an issue.

closed 2025-09-02 by h2zero