h2zero/NimBLE-Arduino #921: Unexpected onSubscribe behavior in 2.x

Berg0162, 2025-03-27:
Dear H2zero,
During the process of updating some code from NimBLE-Arduino version 1.x to 2.x I came across the following (consistent and reproducible) behavior, that should get your attention because it is possibly a unexpected side-effect of an underlying problem.

In 1.x the following snippet handling of onSubscribe was working perfectly: Notify enabled (subValue==1) and Indicate enabled (subValue==2) 
were handled as one would expect, as well as subValue==0 for Notify/Indicate Disabled
```C++
/** Handler class for characteristic actions */
class CharacteristicCallbacks: public NimBLECharacteristicCallbacks {
    void onSubscribe(NimBLECharacteristic* pCharacteristic, ble_gap_conn_desc* desc, uint16_t subValue) {
        String str = "Central Updated CCCD -->";
        NimBLEUUID pCharUUID = pCharacteristic->getUUID();
        if(subValue == 0) {
            str += " Notify/Indicate Disabled for Char:"; 
            if(pCharUUID == STEERING_ANGLE_CHAR_UUID) pServer_Notify_Enabled = false; // Set Notify pServer_SteeringDEVCharAngle (30)
            if(pCharUUID == STEERING_TX_CHAR_UUID) pServer_Indicate_Enabled = false;  // Set Indicate pServer_SteeringDEVCharTx (32)
        } else if(subValue == 1) {
            str += " Notify Enabled for Char:";
            if(pCharUUID == STEERING_ANGLE_CHAR_UUID) pServer_Notify_Enabled = true;  // Set Notify pServer_SteeringDEVCharAngle (30)
            if(pCharUUID == UUID16_CHR_BATTERY_LEVEL)  {                              // Notify pServer_Battery_Level_Chr first time
                pServer_Battery_Level_Chr->notify(&batteryPercentage, 1);            // uint8_t stored in array of 1 byte
                //DEBUG_PRINTF("Notify Battery Level: %3d%%\n", batteryPercentage);   
            }
        } else if(subValue == 2) {
            str += " Indicate Enabled for Char:";
            if(pCharUUID == STEERING_TX_CHAR_UUID) pServer_Indicate_Enabled = true;   // Set Indicate pServer_SteeringDEVCharTx (32)
        } else if(subValue == 3) {
            str += " Notify & Indicate Enabled for Char:";
        }
        DEBUG_PRINTF("%s", str.c_str());
        str = std::string(pCharUUID).c_str();
        DEBUG_PRINTF(" [%s]\n", str.c_str());
    };
};
```

In 2.x the following "same" snippet shows a different behavior:

```C++
class CharacteristicCallbacks: public NimBLECharacteristicCallbacks {
    void onSubscribe(NimBLECharacteristic* pCharacteristic, NimBLEConnInfo& connInfo, uint16_t subValue) {
        String str = "Central Updated CCCD -->";
        NimBLEUUID pCharUUID = pCharacteristic->getUUID();
        if(subValue == 0) {
            str += " Notify/Indicate Disabled for Char:"; 
            if(pCharUUID == STEERING_ANGLE_CHAR_UUID) pServer_Notify_Enabled = false; // Set Notify pServer_SteeringDEVCharAngle (30)
            if(pCharUUID == STEERING_TX_CHAR_UUID) pServer_Indicate_Enabled = false;  // Set Indicate pServer_SteeringDEVCharTx (32)
        } else if(subValue == 1) {
            str += " Notify Enabled for Char:";
            if(pCharUUID == STEERING_ANGLE_CHAR_UUID) pServer_Notify_Enabled = true;  // Set Notify pServer_SteeringDEVCharAngle (30)
            if(pCharUUID == UUID16_CHR_BATTERY_LEVEL)  {                              // Notify pServer_Battery_Level_Chr first time
                pServer_Battery_Level_Chr->notify(&batteryPercentage, 1);            // uint8_t stored in array of 1 byte
                //DEBUG_PRINTF("Notify Battery Level: %3d%%\n", batteryPercentage);   
            }
        } else if(subValue == 2) {
            str += " Indicate Enabled for Char:";
            if(pCharUUID == STEERING_TX_CHAR_UUID) pServer_Indicate_Enabled = true;   // Set Indicate pServer_SteeringDEVCharTx (32)
        } else if(subValue == 3) {
            str += " Notify & Indicate Enabled for Char:";
        }
        DEBUG_PRINTF("%s", str.c_str());
        str = std::string(pCharUUID).c_str();
        DEBUG_PRINTF(" [%s]\n", str.c_str());
    };
};
```
When the client Notify/Indicate disables a server characteristic onSubscibe is called with subValue==0,
when the client <b>Notify</b> enables a server characteristic onSubscibe is called with subValue==1,
However, when the client <b>Indicate</b> enables a server characteristic onSubscibe is called with <b>subValue==1</b> ????
onSubscribe with subValue==2 is not happening anymore!

I could not find in the (migration or code) documentation that subValue has changed this way in version 2.x 
```C++
 * @param [in] subValue The subscription status:
 * * 0 = Un-Subscribed
 * * 1 = Notifications
 * * 2 = Indications
 * * 3 = Notifications and Indications
```

The workaround is simple, but it is possibly a side-effect of an underlying problem. 
Thanks in advance!

h2zero, 2025-03-27:
Thanks @Berg0162 I will look to rectify this shortly!

h2zero, 2025-04-23:
Closing as fixed in https://github.com/h2zero/NimBLE-Arduino/commit/8d1b5373109c9826baf441969ff9b22d198d1bfb

closed 2025-04-23 by h2zero