h2zero/NimBLE-Arduino #922: indicate() [4/5] not working in 2.x

Berg0162, 2025-03-27:
Dear H2zero,
During the process of updating some code from NimBLE-Arduino version 1.x to 2.x I came across the following (consistent and reproducible) behavior, that should get your attention because of the underlying problem.

In 1.x the following snippet is working perfectly and the client confirms the "indicate", and the check never returns false!

```C++
.
uint8_t Server_authResponse[6] = {0x03, 0x12, 0x00, 0x00, 0x00, 0x00};
.
.
// Server replies with a packet: 03 12 ## ## ## ## containing (4 byte --> 32 bit unsigned int) Response (TxChar 32)
if(!pServer_SteeringDEVCharTx->indicate(Server_authResponse, 6)) {  // Array of 6 bytes
    DEBUG_PRINTLN("Indicate authResponse Failed!");
}
```

However, in 2.x the very same code snippet is no longer working!
No compiler complains (like undefined use of indicate(...) )
1) the check never returns false
2) the client does not get the Indicate and as a consequence never confirms "indicate"

As if <b>pServer_SteeringDEVCharTx->indicate(Server_authResponse, 6)</b> is never executed!

The following workaround is working perfectly in 2.x:

```C++
// Server replies with a packet: 03 12 ## ## ## ## containing (4 byte --> 32 bit unsigned int) Response (TxChar 32)
pServer_SteeringDEVCharTx->setValue(Server_authResponse, 6); // Array of 6 bytes
if(!pServer_SteeringDEVCharTx->indicate()) { 
    DEBUG_PRINTLN("Indicate authResponse Failed!");
}
```

I could not find in the (migration or code) documentation that indicate() has changed this way in version 2.x 
```C++
indicate() [4/5]
bool NimBLECharacteristic::indicate(const uint8_t * value, size_t length, uint16_t connHandle = BLE_HS_CONN_HANDLE_NONE) const
Send an indication.

Parameters
[in]	value	A pointer to the data to send.
[in]	length	The length of the data to send.
[in]	connHandle	Connection handle to send an individual indication, or BLE_HS_CONN_HANDLE_NONE to send the indication to all subscribed clients.
Returns
True if the indication was sent successfully, false otherwise.
```

The workaround is simple again, but it has possibly a serious underlying cause. 
Thanks in advance!

h2zero, 2025-03-27:
@Berg0162 Thanks, I am currently looking into this and some similar issues that I hope to resolve shortly.

h2zero, 2025-04-23:
@Berg0162 Please try with the merging of nimble core 1.6, [8f32a23](https://github.com/h2zero/NimBLE-Arduino/commit/8f32a23e2c8a0071b98345915388eb0146b6bfdc)

PR #943 (2025-04-24): [Bugfix] notify/indicate incorrectly returning success with custom value
When sending an array of data with a length value the wrong overload/template was being used and the length value was being interpreted as the connection handle.

This updates the template to disable it when an array is passed and will now also report the error.

Fixes #922
closed 2025-04-24 by h2zero