h2zero/NimBLE-Arduino #923: NimBLE_Extended_Server.ini example not working?

djdehaan, 2025-03-27:
I try to use bluetooth 5 with NimBLE, but connecting will not work. 

I apply the exact sample code (NimBLE_Extended_Server.ini), either with or without changing both primary and secundairyPhy to BLE_HCI_LE_PHY_1M, compile and upload to nRF52840 using Jlink. I use this platform.ini:

[env:Nim-nrf52840server]
platform = https://github.com/h2zero/platform-n-able.git@^1.0.0
board = nRF52840_WHX
framework = arduino
monitor_speed = 115200
;upload_protocol = nrfutil
upload_protocol = jlink
lib_deps = 
	h2zero/NimBLE-Arduino@^2.1.0
upload_port = COM9
extra_scripts = 
	pre:build_versioning.py
board_bootloader = adafruit
build_flags = 
	-DNRF52_SERIES
	-DNRF52840_XXAA
	-DCONFIG_NIMBLE_CPP_LOG_LEVEL=0
	-DCORE_DEBUG_LEVEL=0
	-DCONFIG_BT_NIMBLE_EXT_ADV=1

The serial monitor show 'Started advertising', like it should. When I look in BLE Scanner app on my Samsung S24 I do see the device, I can click on Connect but it wont connect: in the app status is set to Connecting, but then goes back to Disconnected. Any ideas?

h2zero, 2025-03-27:
Do you have legacy advertising enabled? Most phones don't seem to support extended only advertising.

djdehaan, 2025-03-30:
Thanks h2zero, that could be the problem. I now tried on 2 nRF52840 devices the Server and Client Bluetoot_5 examples (because in the end I need a Long Range connection between 2 nRF52840 devices) and try to set up a connectiong between them. And I get exact the same errors:
-they see each other, I see the address coming by
-the filter on the server does not work, actually both advertisedDevice->getServiceUUIDCount() and advertisedDevice->getServiceDataCount() are zero
-when I filter on the address, and allow to connect when the right address is scanned, it does connect but then error stating it cannot find the service - disconnect reason 534.

To follow your tip, I also tried to set setLegacyAdvertising(true), but that is not allowed for extAdv, I guess because extAdv is using extended Advertising and I need extended Advertising to be able to set the coding to LE_CODED. Is that correct thinking?

djdehaan, 2025-03-30:
The server seems to actually 

Starting NimBLE Extended Server
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
D NimBLEService: >> start(): Starting service: UUID: 0xabcd, handle: 0x0000
D NimBLEService: Adding 1 characteristics for service UUID: 0xabcd, handle: 0x0000
D NimBLEService: << start()
primary service
           uuid 0x1800
         handle 1
     end_handle 5
characteristic
           uuid 0x2a00
     def_handle 2
     val_handle 3
   min_key_size 0
          flags [READ]
characteristic
           uuid 0x2a01
     def_handle 4
     val_handle 5
   min_key_size 0
          flags [READ]
primary service
           uuid 0x1801
         handle 6
     end_handle 13
characteristic
           uuid 0x2a05
     def_handle 7
     val_handle 8
   min_key_size 0
          flags [INDICATE]
ccc descriptor
           uuid 0x2902
         handle 9
   min_key_size 0
          flags [READ|WRITE]
characteristic
           uuid 0x2b3a
     def_handle 10
     val_handle 11
   min_key_size 0
          flags [READ]
characteristic
           uuid 0x2b29
     def_handle 12
     val_handle 13
   min_key_size 0
          flags [READ|WRITE]
primary service
           uuid 0xabcd
         handle 14
     end_handle 17
characteristic
           uuid 0x1234
     def_handle 15
     val_handle 16
   min_key_size 0
          flags [READ|WRITE|NOTIFY]
ccc descriptor
           uuid 0x2902
         handle 17
   min_key_size 0
          flags [READ|WRITE]
Started advertising

djdehaan, 2025-03-30:
The client now sees the service in the advertisement, but connection still goes wrong. It seems to be connected but then cannot find the service and times out / disconnects. The loggin while scanning:
```
Service Data:
UUID: 0xabcd, Data: Scan me. Services: 1/1  Service: 0xabcd/0xabcd
Found Our Service !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Connected
Connected to: da:78:25:7c:d7:32 RSSI: -62 Phy: 1/1
da:78:25:7c:d7:32 Disconnected, reason = 534 - Starting scan
ABCD service not found.
Done with this device!
```

Anyone has any tips on how to make them connect using Extended Advertising on a nRF52840? I used both bluetooth_5 samples for Server and Client, and that simply does not work.

h2zero, 2025-04-06:
No idea what the issue is, I will try to reproduce this.

djdehaan, 2025-04-23:
With the solution for https://github.com/h2zero/NimBLE-Arduino/issues/929 this issue is solved.
Thanks h2zero!

closed 2025-04-23 by djdehaan