h2zero/NimBLE-Arduino #924: ESP32 Reboots on BLEDevice::init called

vicatcu, 2025-03-28:
I just went through migrating my NimBLE code from 1.4.2 to 2.2.3 and it compiles without error. It was pretty straightforward as my firmware only acts as a BLE Server (no BLE Client behaviors implemented). Unfortunately, runtime hit a wall really quickly.

Definitively, the call to `BLEDevice::init(productid);` consistently causes the ESP32 to reboot without any kind of core-dump / exception indicated. Silent reboot. Right before the call to `BLEDevice::init(productid)` I put `Serial.printf("Free Memory before BLEDevice::init: %d\r\n", esp_get_free_heap_size());` and it reports (with some variation) that `"Free Memory before BLEDevice::init: 178104"` which seems like an ample amount of Free RAM?

What advise can you offer on how I can diagnose further?

vicatcu, 2025-03-28:
For what it's worth, as best as I can tell, the exact same thing seems to happen when I run the packaged NimBLE_Server example without modifications. Well, at any rate, if I add a delay after `Serial.printf("Starting NimBLE Server\n");` it prints that message and then immediately reboots.

vicatcu, 2025-03-28:
Also, for what it's worth, I'm compiling with v3.0.7 of the esp32-arduino boards package, using ESP32 Dev Module configured like so: 

[image]

and using the following custom partitions table:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1E0000,
app1,     app,  ota_1,   0x1F0000,0x1E0000,
eeprom,   data, 0x99,    0x3F0000,0x1000,
spiffs,   data, spiffs,  0x3F1000,0xF000,
```

vicatcu, 2025-03-28:
Even this basic sketch will cause the reset behavior:
```

#include <Arduino.h>
#include <NimBLEDevice.h>

void setup(void) {
    Serial.begin(115200);
    Serial.printf("Starting NimBLE Server\n");
    delay(1000);
    /** Initialize NimBLE and set the device name */
    NimBLEDevice::init("NimBLE");

}

void loop() {
    /** Loop here and send notifications to connected peers */
    delay(2000);
}
```

Also fwiw, I cloned this repo directly into the ~/Arduino/libraries folder

h2zero, 2025-03-28:
Try another board or cable, this looks like a power issue.

vicatcu, 2025-03-28:
Yea, that's good point, I'll give it a try and post back next week.

vicatcu, 2025-03-28:
Yea, it didn't do the rebooting thing in another jig... but it doesn't show up in Bluetooth scans despite going through initialization...

vicatcu, 2025-03-28:
The NimBLE_Server example sketch does advertise though, so I think we can safely close this issue. Thanks for the heads straightening :)

closed 2025-03-28 by vicatcu