h2zero/NimBLE-Arduino #926: ESP32: Stack canary watchpoint triggered (nimble_host)

vicatcu, 2025-03-31:
I'm randomly getting this shortly after writing a characteristic value:
```
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (nimble_host)
```

My free RAM after Bluetooth initialization is complete and advertising is under way, using `esp_get_free_heap_size()` indicates there are 102672 bytes free... 

At a bit of a loss as to what's happening. Any ideas on diagnostics would be appreciated.

h2zero, 2025-03-31:
I suspect you are doing too much in your callbacks. You can either increase the nimble host task stack size or move your code out of the callbacks and just set a flag or trigger a task (preferred option).

h2zero, 2025-04-24:
@vicatcu Any progress on this?

lazd, 2025-05-25:
> I suspect you are doing too much in your callbacks. You can either increase the nimble host task stack size or move your code out of the callbacks and just set a flag or trigger a task (preferred option).

I added the following platform.io build flags and it seemed to fix my issue, but I will report back if it happens again:

```
build_flags =
  -D ARDUINO_NIMBLE
  -D CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE=6144
```