h2zero/NimBLE-Arduino #928: Callbacks is not triggered

dinesh-slx, 2025-04-03:
Even after setting up the callback, it is not being called during data transfer.

```
#include <NimBLEDevice.h>
#include <NimBLEExtAdvertising.h>
#include "led.h"

// Server callback using the new NimBLEConnInfo reference signature
class MyServerCallbacks : public NimBLEServerCallbacks {
  public:
    void onConnect(NimBLEServer *pServer, NimBLEConnInfo &connInfo) override {
        Serial.println("Client connected");

        // Update connection parameters for the connection handle.
        // (Parameters: min interval, max interval, latency, supervision timeout)
        pServer->updateConnParams(connInfo.getConnHandle(), 24, 40, 0, 500);

        // Obtain the client instance from the server.
        NimBLEClient *pClient = pServer->getClient(connInfo);
        if (pClient) {
            // Request 2M PHY upgrade
            bool phyUpdated = pServer->updatePhy(connInfo.getConnHandle(), BLE_HCI_LE_PHY_2M_PREF_MASK, BLE_HCI_LE_PHY_2M_PREF_MASK, 0);
            Serial.println(phyUpdated ? "success" : "Fail");
        } else {
            Serial.println("Unable to get client instance");
        }
    }

    void onDisconnect(NimBLEServer *pServer, NimBLEConnInfo &connInfo, int reason) override {
        Serial.printf("Client disconnected, reason: %d\n", reason);
        // Restart advertising after a disconnect.
        NimBLEDevice::startAdvertising(0);
    }
};

// Characteristic callback class for handling read/write operations
class CharacteristicCallbacks: public NimBLECharacteristicCallbacks {
    void onRead(NimBLECharacteristic* pCharacteristic){
        Serial.println("----> onRead callback entered");
        NimBLEUUID uuid = pCharacteristic->getUUID();
        Serial.print("Characteristic UUID: ");
        Serial.println(uuid.toString().c_str());
        
        if (uuid.equals(NimBLEUUID("abcd1234-ab12-cd34-ef56-abcdef123456"))) {
            Serial.println("Matching UUID found for read characteristic");
            // Get current color and convert to hex string
            uint32_t color = GetCurrentColor();
            char colorStr[7];
            snprintf(colorStr, sizeof(colorStr), "%06X", color);
            
            pCharacteristic->setValue(colorStr);
            Serial.print("Read current color value: ");
            Serial.println(colorStr);
        } else {
            Serial.println("UUID did not match read characteristic");
        }
    }
    
    void onWrite(NimBLECharacteristic* pCharacteristic){
        Serial.println("----> onWrite callback entered");
        NimBLEUUID uuid = pCharacteristic->getUUID();
        Serial.print("Characteristic UUID: ");
        Serial.println(uuid.toString().c_str());
        
        std::string value = pCharacteristic->getValue();
        Serial.print("Received color value: ");
        Serial.println(value.c_str());
        
        // Convert the hex string to uint32_t
        uint32_t color = strtoul(value.c_str(), NULL, 16);
        Serial.print("Converted color value: 0x");
        Serial.println(color, HEX);
        
        // Set the light color
        cl_err_t result = SetLightColor(color);
        
        // Check if the operation was successful
        if (result == CL_OK) {
            Serial.println("Successfully set light color");
        } else {
            Serial.println("Failed to set light color");
        }
    }
};

static CharacteristicCallbacks chrCallbacks;

void setup() {
    Serial.begin(115200);
    Serial.println("Starting NimBLE Server with Extended Advertising");

    // Initialize the NimBLE device with a device name.
    NimBLEDevice::init("ESP32-ExtAdv");
    NimBLEDevice::setDeviceName("ESP32-ExtAdv");
    NimBLEDevice::setPower(ESP_PWR_LVL_P9); // Set maximum TX power
    InitNeoPixels();
    // Create the BLE server and set its callbacks.
    NimBLEServer *pServer = NimBLEDevice::createServer();
    pServer->setCallbacks(new MyServerCallbacks());

    // Create a service with a 128-bit UUID.
    NimBLEService *pService = pServer->createService("12345678-1234-1234-1234-1234567890ab");
    // Create read characteristic
    NimBLECharacteristic *pReadCharacteristic = pService->createCharacteristic(
        "abcd1234-ab12-cd34-ef56-abcdef123456",
        READ,3);
    pReadCharacteristic->setValue("Hello from ESP32");
    pReadCharacteristic->setCallbacks(&chrCallbacks);

    // Create write characteristic
    NimBLECharacteristic *pWriteCharacteristic = pService->createCharacteristic(
        "abcd1234-ab12-cd34-ef56-abcdef123457",
        WRITE,3
    );
    pWriteCharacteristic->setCallbacks(&chrCallbacks);    
    pService->start();

    // Set up extended advertising.
    NimBLEExtAdvertisement extAdv;
    extAdv.setName("ExtAdv");
    extAdv.setConnectable(true);
    extAdv.addTxPower();
    extAdv.setFlags(0x06);

    // extAdv.setLegacy(true); // Use extended advertising
    // Add the service UUID to the advertisement payload.
    extAdv.addServiceUUID(pService->getUUID());
    // Optionally set advertising intervals (in units of 0.625 ms).
    extAdv.setMinInterval(0x20);
    extAdv.setMaxInterval(0x40);
    extAdv.setAppearance(0x0340); // Example appearance value for a generic heart rate sensor

 
    // Get the extended advertising instance.
    NimBLEExtAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
    // Pass the configured advertisement to instance 0.
    pAdvertising->setInstanceData(0, extAdv);
    // Start advertising using instance ID 0.
    pAdvertising->start(0);
 
    Serial.println("Extended advertising started");

}

void loop() {
    // BLE events are handled by callbacks—no need to add code here.
}
```

h2zero, 2025-04-03:
Which callbacks aren't being called? Please enable debug logging and share the output so I can help debug this.

dinesh-slx, 2025-04-08:
```
[20:11:37:172] D NimBLEServerCallbacks: onConnect(): Default␍␊
[20:11:37:172] D NimBLEServer: << handleGapEvent␍␊
[20:11:37:181] D NimBLEServer: >> handleGapEvent: ␍␊
[20:11:37:181] D NimBLEExtAdvertisingCallbacks: onStopped: Default␍␊
[20:11:37:297] D NimBLEServer: >> handleGapEvent: ␍␊
[20:11:37:297] D NimBLEServer: << handleGapEvent␍␊
[20:11:37:302] D NimBLEServer: >> handleGapEvent: ␍␊
[20:11:37:302] I NimBLEServer: mtu update event; conn_handle=1 mtu=255␍␊
[20:11:37:308] D NimBLEServerCallbacks: onMTUChange(): Default␍␊
[20:11:37:308] D NimBLEServer: << handleGapEvent␍␊
[20:11:37:927] D NimBLEServer: >> handleGapEvent: ␍␊
[20:11:37:927] I NimBLEServer: subscribe event; attr_handle=8, subscribed: true␍␊
[20:11:37:933] D NimBLEServer: << handleGapEvent␍␊
[20:11:38:017] D NimBLEServer: Gatt Write event␍␊
[20:11:38:024] D NimBLECharacteristicCallbacks: onWrite: default␍␊
[20:11:38:107] D NimBLEServer: Gatt Write event␍␊
[20:11:38:107] D NimBLECharacteristicCallbacks: onWrite: default␍␊
[20:11:38:197] D NimBLEServer: Gatt Write event␍␊
[20:11:38:197] D NimBLECharacteristicCallbacks: onWrite: default␍␊
[20:11:38:288] D NimBLEServer: Gatt Write event␍␊
```

It is going to the default characteristic callbacks instead of the custom characteristic callback that I created.

dinesh-slx, 2025-04-08:
I apologize; I realize I’m not using the correct function signatures for onRead and onWrite.

closed 2025-04-08 by dinesh-slx

h2zero, 2025-04-08:
Glad you got it sorted out 😄