h2zero/NimBLE-Arduino #929: Example NimBLE_multi_advertiser.ino not working on nRF52840

djdehaan, 2025-04-03:
I burn the exact script of the multi-adviser BT5 example with one corrected compile error: 
`
     legacyConnectable.setAddress(NimBLEAddress("DE:AD:BE:EF:BA:AD", 1));
`
And then: 2 devices appear on a phone. The legacy-one shows "connectable" but is not connectable. After short trying, it keeps not connected. What goes wrong? Who has tried this example on a nRF52840?

Thanks for any help,
Dirk Jan

h2zero, 2025-04-06:
Hello @djdehaan, do you see anything in the logs, phone and nrf?

djdehaan, 2025-04-06:
Hi h2zero, 

The **platform.ini** is:
```
[env:Nim-nrf52840server]
platform = n-able
board = nRF52840_WHX
framework = arduino
monitor_speed = 115200
;upload_protocol = nrfutil
upload_protocol = jlink
lib_deps = 
	h2zero/NimBLE-Arduino@^2.2.3
upload_port = COM9
extra_scripts = 
	pre:build_versioning.py
board_bootloader = adafruit
build_flags = 
	-DUSE_LFRC
	-DNRF52_SERIES
	-DNRF52840_XXAA
  -DCONFIG_NIMBLE_CPP_LOG_LEVEL=5
	-DCORE_DEBUG_LEVEL=5
	-DCONFIG_BT_NIMBLE_EXT_ADV=1
	-DCONFIG_NIMBLE_CPP_ENABLE_RETURN_CODE_TEXT
	-DCONFIG_NIMBLE_CPP_ENABLE_GAP_EVENT_CODE_TEXT
	-DCONFIG_NIMBLE_CPP_ENABLE_ADVERTISEMENT_TYPE_TEXT
 	-DCONFIG_BT_NIMBLE_ROLE_CENTRAL_DISABLED
  -DCONFIG_BT_NIMBLE_ROLE_OBSERVER_DISABLED
```

The code I use is like this: **main.h**
```
#pragma once
#include <Arduino.h>
#include <NimBLEDevice.h>

class _ServerCallbacks : public NimBLEServerCallbacks 
{
  public:
    void onConnect(NimBLEServer* pServer, NimBLEConnInfo& connInfo) override;
    void onDisconnect(NimBLEServer* pServer, NimBLEConnInfo& connInfo, int reason) override;
    void onIdentity	(	NimBLEConnInfo & 	connInfo	)	override;

};

/** Callback class to handle advertising events */
class _AdvCallbacks : public NimBLEExtAdvertisingCallbacks 
{
  public:
    bool m_updatedSR = false;
    void onStopped(NimBLEExtAdvertising* pAdv, int reason, uint8_t instId) override ;
    void onScanRequest(NimBLEExtAdvertising* pAdv, uint8_t instId, NimBLEAddress addr) override;
};
```
And **main.cpp**
```
/**
 *  NimBLE Multi Advertiser Demo:
 *
 *  Demonstrates the Bluetooth 5.x extended advertising capabilities.
 *
 *  This demo will advertise 2 advertisements, and extended scannable instance
 *  and a connectable legacy instance. They will advertise for 5 seconds then
 *  sleep for 20 seconds. The extended scannable instance will use the scan
 *  request callback to update it's data when a scan response is requested.
 *
 *  Created: on April 9 2022
 *      Author: H2zero
 */
[... 232 lines cut ...]
     def_handle 12
     val_handle 13
   min_key_size 0
          flags [READ|WRITE]
primary service
           uuid 0xabcd
         handle 14
     end_handle 17
characteristic
           uuid 0x1234
     def_handle 15
     val_handle 16
   min_key_size 0
          flags [READ|WRITE|NOTIFY]
ccc descriptor
           uuid 0x2902
         handle 17
   min_key_size 0
          flags [READ|WRITE]
Started advertising
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
```
The zero's at the end are repeating until the device reboots as designed in the demo code - showing that it is alive an that there is not a call to the onConnect or anything else. I don't know how to see logging on my phone. I use BLE Scanner, I see the device, I can click CONNECT and the it shows **Connected, discovering services...**, but after some seconds the status is **Disconnected** again. When I upload this code to a ESP32S3 it will connect and I can look into the characteristics. 

Do you have any idea??

juricabi, 2025-04-06:
Tried the same, it gives 

[image], with legacy advertising everything works, thank you for your efforts.

h2zero, 2025-04-07:
Thanks to both of you for the info, I will have a look to see what could be causing this, I suspect an issue in the stack which is not always easy to find.

djdehaan, 2025-04-07:
Maybe relevant: connecting with my phone to the extended advertiser results in exact the same errors and messages.

djdehaan, 2025-04-13:
@h2zero Any luck finding this issue? Would appreciate a fix very much!

h2zero, 2025-04-13:
@djdehaan I've been able to reproduce it but I've been too busy with other (paid) work to find the cause yet. Hopefully this week.

h2zero, 2025-04-18:
@djdehaan The fix is in the nimble core update PR #891 , you can use this for now but beware that it will change, also the n-able core will be updated soon to better work with this.

djdehaan, 2025-04-20:
@h2zero Great! It indeed solves the problem. I found a way to use this fixed version and now indeed I can connect to the device using extended advertising! For who want to know: i downloaded this version as a lib in the project folder /lib manually 
```
git clone https://github.com/h2zero/NimBLE-Arduino.git
git fetch origin pull/891/head:pr-891
git checkout pr-891
```

Will that mean that in the next release I can revert back to the platformio lib_deps parameter?
```
lib_deps = 
	;h2zero/NimBLE-Arduino@^2.2.3
```

h2zero, 2025-04-23:
Closing this as fixed in #891

You are able to reference this in platformio via the commit sha:
```
lib_deps = https://github.com/h2zero/NimBLE-Arduino#8f32a23
```

closed 2025-04-23 by h2zero