h2zero/NimBLE-Arduino #933: configMAX_PRIORITIES seems to be incorrect value for freeRTOS task priority

pwilkowski, 2025-04-08:
According to freertos docs , task should have maximum priority of configMAX_PRIORITIES - 1
https://www.freertos.org/Documentation/02-Kernel/02-Kernel-features/01-Tasks-and-co-routines/03-Task-priorities

Which seems to be bigger here
```cpp
ll_task_h = xTaskCreateStatic(nimble_port_ll_task_func, "ll", NIMBLE_LL_STACK_SIZE,
                                  NULL, configMAX_PRIORITIES, ll_xStack, &ll_xTaskBuffer);
```
https://github.com/h2zero/NimBLE-Arduino/blob/62180ab2bc9c9a8b41f9fb800ea7256e89628a3a/src/nimble/porting/npl/freertos/src/nimble_port_freertos.c#L114

h2zero, 2025-04-08:
Thanks! not sure how it was missed and also apparently working for so long...

pwilkowski, 2025-04-08:
after enabling debugging for tasks in free rots i see this:

[image]

i came to conclusions
1. let Ble tasks to have configurable priorities
2. by default imo it should be `configMAX_PRIORITIES -1` and `configMAX_PRIORITIES - 2` so they dont collide with built in Tmr task

h2zero, 2025-04-23:
Closing this as fixed in #891

closed 2025-04-23 by h2zero