h2zero/NimBLE-Arduino #936: Assistance on creating a HID Braille display

jwlilly, 2025-04-18:
Hello, 

I am working on creating an ESP-32 braille display using the HID standards. I forked my repo from a project using NimBLE. The braille display seems to be working fine on iOS, macOS, and Windows. They are recognizing the device as a braille display and sending the data to the device using the output report. 

The issue I'm running into is with Android. For some reason, it has some very specific requirements around braille displays. For starters, it seems to filter specific devices based on the advertised name. I've gotten past that part, but it also seems to do a second check for a specific UUID. If that UUID is not present, it will not send braille data back to the device. 

The specific UUID that it is looking for is `00001124-0000-1000-8000-00805f9b34fb`. I am new to bluetooth and HID specifications, but in my research, the`0x1124` seems to be a Service Class and Profile for the Human Interface Device Profile (HID). The problem that I'm running into is that I don't know how to define the UUID with NimBLE. The only place I can see to set a UUID is when creating a service. I tried that (most likely doing it wrong) and the nRF Connect lists it as an unknown service (and the Android device refuses to connect to the braille display). I also tried adding the UUID as a characteristic, but that didn't change anything. 

If you're unfamiliar, the accessibility service called TalkBack handles the logic for accepting a braille display. This is the file that is looking for the specific UUID: https://github.com/google/talkback/blob/26a27dc009d5b3605e744222541f045a3c24e038/braille/brailledisplay/src/phone/java/com/google/android/accessibility/braille/brailledisplay/platform/connect/bt/BtConnectManager.java#L57C43-L57C79 

This is the repo that I am working from. It does not contain any of my recent experiments with the UUID: https://github.com/jwlilly/ESP32-BLE-BrailleHID 

I appreciate any assistance! The HID braille display standard is not documented very well so we've been trying to figure things out in t[his discussion](https://github.com/nvaccess/nvda/discussions/14829) if you're interested in some history.

h2zero, 2025-04-18:
Are you advertising that service uuid (just use 0x1124, not the 128bit version)? You will probably also need to add it as a service and the mandatory characteristics if any.

jwlilly, 2025-04-19:
Thanks for the reply. I am advertising 0x1124, and I have added the uuid as a service as well (using both 0x1124 and the 128bit version, not both at the same time though). I tried adding the same characteristics as the other hid service, but that didn't seem to solve the issue. I'm not sure what characteristics are mandatory for that uuid or even where to find that information.

jwlilly, 2025-04-28:
Thank you for the information! I believe the issue is that they are only checking for classic bluetooth. I kept seeing the HID service for 0x1124 and the HID service for 0x1812 (for GATT) and was wondering what the difference was. But now it makes sense. I guess I need to create and issue on TalkBack for some clarification.

closed 2025-04-28 by jwlilly

discapacidad5, 2025-04-28:
Hola amigo veo que cerraste como completado pudiste resolver el problema