h2zero/NimBLE-Arduino #938: It looks like connections are not being closed

rovale, 2025-04-20:
Good morning! Thank you so much for this library. It is getting better every time and I really appreciate all the improvements in v2. The support you provide is so helpful, too! :-)

I am using this library to read advertisement data from sensors and to allow my device to be controlled with a mobile app. What I notice, and it is quite hard to reproduce, is that connections are not always closed. 

This graph shows the result of  `NimBLEDevice::getServer()->getConnectedCount()`:
[image]

At one point the device runs into the configured limit of 3 of `CONFIG_BT_NIMBLE_MAX_CONNECTIONS` and as a result the device would not allow new connections.

Did you ever run into this before? Maybe some things for me to check?

Thanks in advance!

h2zero, 2025-04-20:
Hello @rovale this is an issue that has been brought up quite a bit lately, unfortunately I have not found a solution yet. Please see these threads:

#917 
https://github.com/h2zero/esp-nimble-cpp/issues/320

rovale, 2025-04-20:
Thank you! I didn't realize they are related. I will keep a close eye on it.

closed 2025-04-20 by rovale